autowarefoundation/autoware #2398: `map` to `base_link` frame is not publishing

GPrathap, 2022-01-11:
## Bug report

### Required information:

- Operating system and version:
   Ubuntu 18.04
- Autoware installation type:
   From the source, Docker 
- Autoware version or commit hash
  1.14.0
- ROS distribution and version:
 Melodic
- ROS installation type:
   Docker
- Package or library, if applicable:
  - ndt_matching 

### Description of the bug
I was trying to run the example provided here https://www.svlsimulator.com/docs/system-under-test/autoware-instructions/ 
However, all the time `map` to `base_link` frame is not defined. 
When I check the codebase I notice this, I guess that is the issue. 

Here  https://github.com/Autoware-AI/core_perception/blob/1.14.0/lidar_localizer/nodes/ndt_matching/ndt_matching.cpp#L928

`if (map_loaded == 1 && init_pos_set == 1) ` only this `map` to `base_link` is published. Yet, map_loaded is never be true. 

When I set the initial position `init_pos_set` becomes 1. I guess by looking into bit deeper I noticed these  
https://github.com/Autoware-AI/core_perception/blob/1.14.0/lidar_localizer/nodes/ndt_matching/ndt_matching.cpp#L1670
 https://github.com/Autoware-AI/core_perception/blob/1.14.0/lidar_localizer/nodes/ndt_matching/ndt_matching.cpp#L536
 
I have no idea how to solve this issue, let me know what should I do
### Steps to reproduce the bug

Try to follow the instructions up to the localization node https://www.svlsimulator.com/docs/system-under-test/autoware-instructions/  

### Expected behaviour
`map` to `base_link` frame should be published yet it is not 

### Actual behavior
, all the time `map` to `base_link` frame is not defined.

siyouluo, 2022-08-28:
你好, 我也遇到了这个问题. 结果发现是因为点云地图没有成功导入. 在 autoware-1.14.0 中无法通过 launch 文件的形式导入点云地图, 包括在 Runtime Manager -> Quick Start -> Map 中启动 my_map.launch 也无法导入点云地图. 你可以查看点云地图话题下的消息, 大概率会得到如下输出
```
$ rostopic echo /points_map
WARNING: no messages received and simulated time is active.
Is /clock being published?
```

解决办法有两个
1. 在 Runtime Manager -> Map -> Point Cloud 中载入 pcd 点云文件来加载点云地图, 并确认进度条加载到 100%, 且右边显示 OK 字样. 如果进度条没加载完成就失败, 你可以继续重试, 一般重试几次即可;
2. 换其他版本的 autoware, 比如 autoware-1.13.0 就没有这个问题;

以上办法并没有彻底解决该问题, 因为我还不知道这个 bug 出现的真正原因是什么, 如果有人知道, 欢迎交流讨论.

siyouluo, 2022-08-29:
感谢 @reflector-li 同学发现了真正的问题所在, 出现这个bug的真正原因在于autoware-1.14.0 更换了点云地图的加载方式，却没有相应地修改 `my_map.launch`.   
如果你按照如下方式修改, 应该能够成功加载点云地图, 从而实现 ndt_matching 定位.

```diff
<!-- Point Cloud -->
-  <node pkg="map_file" type="points_map_loader" name="points_map_loader" args="noupdate $(env HOME)/.autoware/data/map/pointcloud_map/bin_Laser-00167_-00864.pcd $(env HOME)/.autoware/data/map/pointcloud_map/bin_Laser-00153_-00852.pcd"/>

+  <include file="$(find map_file)/launch/points_map_loader.launch">
+     <arg name="path_pcd" value='$(env HOME)/.autoware/data/map/pointcloud_map/bin_Laser-00167_-00864.pcd, $(env HOME)/.autoware/data/map/pointcloud_map/bin_Laser-00153_-00852.pcd'/>
+  </include>
```