autowarefoundation/autoware #2400: Not able to run YOLO

wirin123, 2022-02-16:
I am unable to get the boundary boxes during the live image streaming

TheGermanKid, 2022-02-24:
Please provide some more information: which node are you running, have you chosen the correct weight and model in the settings (in case you are using the GUI) or launch file and which topic are you displaying as your image? Additionally you should provide the output of the terminal to look at errors.

wirin123, 2022-03-16:
Hi Jonas, 
There was glitch in the gui so i installed the autoware.ai through docker.
My issue is while connecting my camera.I have a point grey Blackfly camera FLIR Camera.When I try launch it from the runtime manager i get a error.i have attached the screenshot
[image]

JonasLos, 2022-03-16:
Unfortunately, I am not familiar with the docker installation. As the error suggest there seems to be an issue with the spinnaker driver not being installed or not being found. I would suggest that you try to run the autoware_pointgrey_drivers from terminal  in the docker to verify if it doesnt work at all. ( roslaunch autoware_pointgrey_drivers spinnaker.launch). Alternatively you should probably ask this question again on answers.ros.org/ with autoware.ai tag as there may be more support available.

wirin123, 2022-03-17:
Hello Jonas,
Thank you for the response