autowarefoundation/autoware #2413: Fix ARM build of scenario_simulator

kenji-miyake, 2022-06-26:
`libembree-dev` can't be installed on ARM.
https://github.com/autowarefoundation/autoware/pull/8#issuecomment-1014333684

- Build `embree` manually
- Run `rosdep install` with `--skip-keys=embree`

oguzkaganozt, 2023-09-27:
Hi, any updates on this issue ?

@hakuturu583 Can we use `embree_vendor` package on arm CPU ?

hakuturu583, 2023-10-02:
> @hakuturu583 Can we use embree_vendor package on arm CPU ?
Yes, I tested on qemu.

oguzkaganozt, 2023-10-04:
Hi @hakuturu583,
 I've tried with Scenario Simulator **master** branch and [embree_vendor](https://github.com/OUXT-Polaris/embree_vendor). It uses embree version 4.2.0 and when I try to do:
 `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-up-to simple_sensor_simulator`
 
I got following error on compilation:
`error: ‘RTCIntersectContext’ has not been declared`

I also checked with the original Embree version 4.0 changelog and it says:
`RTCIntersectContext is renamed to RTCRayQuery context and most members moved to
    new RTCIntersectArguments and RTCOccludedArguments structures.`
   
So it appears that embree version 4.x is not fit for Scenario simulator. How should I proceed do you have any suggestions ? 

Thanks in advance.

mitsudome-r, 2023-10-05:
I just checked installed embree apt package in my amd64 platform. It looks like I have version 3.12.2 installed.
May be you could try changing [this line](https://github.com/OUXT-Polaris/embree_vendor/blob/7cd96dd0a938e93a4bdc4f09089ef070f0a407ad/CMakeLists.txt#L32C13-L32C19) to ``v3.12.2`.

oguzkaganozt, 2023-10-05:
I tried building with `v3.12.2` and it requires libTBB version `2021.1.1` and is incompatible with `2021.5.0` that we have on Ubuntu22/arm64. Also tried version `v3.13.5` and it failed with ` error: unrecognized command-line option ‘-msse2’` seems like the flag incompatibility with arm64.

mitsudome-r, 2023-10-18:
I was able to build on arm with the following workaround:
1. Use Embree 3.13.5 and cherry-pick https://github.com/embree/embree/commit/82ca6b5ccb7abe0403a658a0e079926478f04cb1 as mentioned in [this issue](https://github.com/embree/embree/issues/383#issuecomment-1453490760). I've created a fork of embree_vendor with the fix so you can use the following:
```
cd autoware/src/universe/external
git clone -b v3.13.5_arm https://github.com/mitsudome-r/embree_vendor
```
2. Modify simple_sensor_simulator's dependency in package.xml to use embree_vendor instead of embree.
3. Even with the above fix, I still had errors from Eigen in openscenario_visualization package related to this [issue](https://gitlab.com/libeigen/eigen/-/issues/2326). I added this warning as exception in the compile option as a quick fix.

For 2 and 3, you can checkout my fork:
```
cd autoware/src/simulator/scenario_simulator
git remote add https://github.com/mitsudome-r/scenario_simulator_v2
git fetch mitsudome-r
git checkout arm
```

mitsudome-r, 2023-10-19:
Related Issue: https://github.com/tier4/scenario_simulator_v2/issues/1094

oguzkaganozt, 2023-10-19:
> I was able to build on arm with the following workaround:
> 
> 1. Use Embree 3.13.5 and cherry-pick [embree/embree@82ca6b5](https://github.com/embree/embree/commit/82ca6b5ccb7abe0403a658a0e079926478f04cb1) as mentioned in [this issue](https://github.com/embree/embree/issues/383#issuecomment-1453490760). I've created a fork of embree_vendor with the fix so you can use the following:
> 
> ```
> cd autoware/src/universe/external
> git clone -b v3.13.5_arm https://github.com/mitsudome-r/embree_vendor
> ```
> 
> 2. Modify simple_sensor_simulator's dependency in package.xml to use embree_vendor instead of embree.
> 3. Even with the above fix, I still had errors from Eigen in openscenario_visualization package related to this [issue](https://gitlab.com/libeigen/eigen/-/issues/2326). I added this warning as exception in the compile option as a quick fix.
> 
> For 2 and 3, you can checkout my fork:
> 
> ```
> cd autoware/src/simulator/scenario_simulator
> git remote add https://github.com/mitsudome-r/scenario_simulator_v2
> git fetch mitsudome-r
> git checkout arm
> ```

Thanks @mitsudome-r that solved the issue.

closed 2023-10-30 by idorobotics