autowarefoundation/autoware #2446: Use a common base image for CUDA libraries

kenji-miyake, 2022-06-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, we install CUDA using `setup-dev-env.sh`, but it creates different layers and enforces people to pull a large part of the image every time.

To avoid that, it's better to have a common base image.

### Purpose

To minimize the differences between Autoware's Docker Images.

### Possible approaches

- Wait for NVIDIA to release a new image and deb packages, and use them. (Recommended)
  - We might have to add the `--no-cuda` option to `setup-dev-env.sh`.
- Create a common Autoware base image.

### Definition of done

Autoware's Docker Images are based on a common base image.

kenji-miyake, 2022-07-04:
For Galactic, it will be resolved by https://github.com/autowarefoundation/autoware/pull/2703.

xmfcx, 2023-11-15:
For both arm64 and amd64, we can use [nvidia/cuda:12.3.0-base-ubuntu22.04](https://hub.docker.com/layers/nvidia/cuda/12.3.0-base-ubuntu22.04/images/sha256-2e73f975a775308d37bfb40cb0b13d42730f52c53cc9918cc3be05ef0cbc0452?context=explore) since now we have upgraded to CUDA 12.3 with:
- https://github.com/autowarefoundation/autoware/issues/3943

I will create a small PR to build and test.

xmfcx, 2023-11-16:
Closing this since it will be handled in more detail in:
- https://github.com/autowarefoundation/autoware/issues/3999

closed 2023-11-16 by xmfcx