autowarefoundation/autoware #2452: CUDA environment is broken when I run a docker container with rocker

yukke42, 2022-06-26:


angry-crab, 2022-06-28:
I've also met the same problem. I'm using `rocker --nvidia --x11 --user --volume $HOME/autoware -- ghcr.io/autowarefoundation/autoware-universe:latest`.

VRichardJP, 2022-06-28:
I have the problem with `rocker --nvidia --x11 --user --volume $HOME/autoware -- ghcr.io/autowarefoundation/autoware-universe:galactic-latest` but not with `rocker --nvidia --x11 --user --volume $HOME/autoware -- ghcr.io/autowarefoundation/autoware-universe:latest`

kenji-miyake, 2022-06-28:
> but not with rocker --nvidia --x11 --user --volume $HOME/autoware -- ghcr.io/autowarefoundation/autoware-universe:latest

@VRichardJP Could you tell me the hash of your images like this? I guess one is old (and based on CUDA image).

```sh-session
$ docker images | grep autoware-universe | grep " latest "
ghcr.io/autowarefoundation/autoware-universe                                          latest                                                                       5fdadcabfe9c   5 weeks ago     5.73GB
```

VRichardJP, 2022-06-29:
Yes indeed, my `latest` is older than `galactic-latest`:
```
```

angry-crab, 2022-06-29:
`docker run --rm -it --gpus all -e DISPLAY -e TERM -e QT_X11_NO_MITSHM=1 -v /tmp/.X11-unix:/tmp/.X11-unix -v /etc/localtime:/etc/localtime:ro -v $HOME/adehome:/home/adehome [ghcr.io/autowarefoundation/autoware-universe:humble-latest](http://ghcr.io/autowarefoundation/autoware-universe:humble-latest)`
Using `docker` instead of `rocker` does not solve the problem. CUDA, TensorRT, and cuDNN could not be found.

kenji-miyake, 2022-06-29:
@angry-crab `humble-latest` doesn't contain CUDA now. Please use `humble-latest-cuda` instead.
https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/26944787?tag=humble-latest-cuda

kenji-miyake, 2022-07-04:
Sent a PR: https://github.com/osrf/rocker/pull/182

If it's not accepted, I'll add the following block in the Dockerfile.

```dockerfile

## Set env for nvidia-container-runtime
ENV NVIDIA_VISIBLE_DEVICES all
ENV NVIDIA_DRIVER_CAPABILITIES compute,utility
```

kenji-miyake, 2022-07-08:
@yukke42 My PR to `rocker` is merged. Do you think we can close this issue?
We can set `NVIDIA_DRIVER_CAPABILITIES` manually until the new `rocker` is released.

yukke42, 2022-07-12:
@kenji-miyake 
I'm sorry for the late replay. AND thank you for identifying the cause of this issue!
I will close this issue after creating a PR to do your suggestion.

yukke42, 2022-07-12:
> @.kenji-miyake I'm sorry for the late replay. AND thank you for identifying the cause of this issue! I will close this issue after creating a PR to add your suggestion into the document.

I will do after [this related issue](https://github.com/autowarefoundation/autoware.universe/issues/1311#issuecomment-1181318486) is closed.

kenji-miyake, 2022-07-13:
Resolved by #2732

closed 2022-07-13 by kenji-miyake