autowarefoundation/autoware #2454: Create separate containers for testing Open AD Kit

xmfcx, 2022-06-26:


LalithVipulananthan, 2022-07-06:
@xmfcx Once the containers have been created, I am happy to help out with the testing part of this issue.

xmfcx, 2022-07-06:
# Required nodes

In order to list the required nodes for the containers, I am trying to use @kenji-miyake -san's tool from https://github.com/orgs/autowarefoundation/discussions/2581

The tool is private for now but I'm working in cooperation with him to resolve some issues on the package.

With the tool, I will hopefully have a list of all the nodes that are launched with the main launch files.

# Building the docker image manually

Right now I am trying to use this [build.sh](https://github.com/autowarefoundation/autoware/blob/main/docker/build.sh) to build the docker image but every time I tried, I got some connectivity issues and every time I restarted, it failed at some point and takes a very long time to retry. I will investigate this further.

I will try to build the container without buildx thingy, maybe add more layers to the image, also will consult to @kenji-miyake on this.

LalithVipulananthan, 2022-07-19:
@xmfcx Hi Fatih. Would it be possible to update this issue with your current progress please? :)

LalithVipulananthan, 2022-08-03:
@xmfcx Hi Fatih. Based on the conversation last week, it seems as if this issue is no longer considered to beneficial towards developing a microservices architecture for Autoware and the CI/CD infrastructure changes that would need to be made to support that architecture. If so, could you please add a short summary note and close the issue please.

xmfcx, 2022-08-04:
Hello @LalithVipulananthan,

Yeah, like you've said, this issue wouldn't achieve much. What we really need is to have a solid package release process based on [bloom](https://docs.ros.org/en/galactic/How-To-Guides/Releasing/Releasing-a-Package.html).

closed 2022-08-04 by xmfcx

doganulus, 2022-08-04:
I wonder what prevents to build containers as specified in this discussion.

Honestly speaking, it is hard to argue that ROS tooling is modern by current standards. I am afraid of relying more on that would bring more headache. Containers seemed like a good solution to me. What went wrong exactly?

xmfcx, 2022-08-04:
@doganulus Autoware Universe currently provides several docker containers in https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe and you can follow the [instructions from the documentation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/) to run Autoware using them.

These containers are built as artifacts of the [Autoware CI workflows](https://github.com/autowarefoundation/autoware/tree/main/.github/workflows),   [twice in a month](https://github.com/autowarefoundation/autoware/blob/main/.github/workflows/docker-build-and-push-humble-self-hosted.yaml#L8).

Also the plan is create our own debian repository for Autoware ROS packages' releases. This way if we want to create containers from select packages, we can install them into the container using `apt`. Any other ROS package is released/installed that way, including many dependencies of the Autoware.