autowarefoundation/autoware #2819: sending output via udp

urbansound8K, 2022-08-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

i have managed to run autoware in jetson AGX and everything seems to work well!
the only thing I am looking for is that how to send the output of the lidar via UDP!
what I mean is that I used autoware to detect and track objects using lidar only
is there any way to make the output send thro udp
like xyz

### Purpose

to run with a new contral unit

### Possible approaches

using autoware with different robot not only a car

### Definition of done

autwoare is installed

mitsudome-r, 2022-09-15:
As mentioned in the [support guideline](https://autowarefoundation.github.io/autoware-documentation/main/support/support-guidelines/#github-issues), we are using Github issue as a place to track bugs and tasks. If you have question, GitHub Discussions is a better place. 
https://github.com/autowarefoundation/autoware/discussions/categories/q-a

As a short answer to your question, if you want to see the LiDAR data, ROS2 natively supports communication between multiple machines by setting the same [ROS_DOMAIN_ID](https://docs.ros.org/en/foxy/Concepts/About-Domain-ID.html)

urbansound8K, 2022-09-15:
I opened it the same day I posted this issue!
https://github.com/autowarefoundation/autoware/discussions/2818

could that happned with ROS1?

mitsudome-r, 2022-09-20:
Checkout http://wiki.ros.org/ROS/Tutorials/MultipleMachines

closed 2022-10-17 by urbansound8K