autowarefoundation/autoware #2840: Reduce docker image size for runtime usages

Sharrrrk, 2022-09-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description
Originated from discussion  posted by @kasperornmeck in https://github.com/autowarefoundation/autoware/discussions/2780_

## Purpose

Reduce docker images size for runtime environment where unnecessary software could be removed from development environment and reduce the disk space used.

## Current status

Details of the container used in the analysis:
> REPOSITORY: ghcr.io/autowarefoundation/autoware-universe
> TAG: galactic-20220728-prebuilt-cuda-arm64
> IMAGE ID: 8b7fbc56f6bf
> CREATED: 8 days ago
> SIZE: 19.3GB

### Definition of done
- [ ] runtime dependencies analysis
- [ ] reduce sized docker image for runtime

Sharrrrk, 2022-09-10:
With docker image built from nvidia/cuda:11.4.0-runtime-ubuntu20.04, and runtime dependencies installed for autoware, the disk usage of `/usr/local/cuda-11.4` folder could be reduced by 2.2GB.

Original:   `4.0G    /usr/local/cuda-11.4`
New:        `1.8G    /usr/local/cuda-11.4`

Sharrrrk, 2022-09-10:
Currently I'm thinking of two approaches of this work:

- Reduce size of `prebuilt` image
  Considering the prebuilt images is for users to try out Autoware quickly, dev environment could be remove from it. 

- Keep current `prebuilt` image and build another `runtime` image
  This allows users to rebuild partial of Autoware, not the whole project using prebuilt image, then runtime image may be provided just for execution. 

The main concept of size reducing:
- Only use runtime libraries for CUDA/TensorRT (This could save ~2.2GB)
- Only copy the `install` folder or maybe also keep `src` folder (This could save up to ~3.3GB)

@kenji-miyake, what's your opinion about it ?

kenji-miyake, 2022-09-10:
@Sharrrrk Thank you for investigating this issue!
I totally agree with you, and either option seems okay. The first option would be better as a first step because it's simpler.

kasperornmeck, 2022-09-11:
@Sharrrrk thanks for taking this on!

For inspiration, you could take a look at the images layers which ROS and NVIDIA have used to create their Docker images. See
* https://github.com/osrf/docker_images/blob/3f4fbca923d80f834f3a89b5960bad5582652519/ros/galactic/ubuntu/focal/ros-core/Dockerfile
* https://hub.docker.com/layers/nvidia/cuda/11.4.2-runtime-ubuntu18.04/images/sha256-9fd4770341bd8f2cd754b70d552dfd0cc5e6e3ed1406449f00b1c5bb944133b0?context=explore

I'd recommend building up a new container and not removing from an existing one.

Sharrrrk, 2022-09-15:
Hi @kasperornmeck, I looked into the docker file of Nvidia and ROS, Nvidia used a script from outside to install all the packages, and for ROS, it's just like what have been implented in current ansible script. So I would say that use nvidia runtime docker image as base would be a good and neat approach. I will modify the ansible srcipts (add a variant that only installs runtime dependencies) and the Docker file (use mutlti stage build) of course, so that the `prebuilt` image could reduce it's size in later build.

kasperornmeck, 2022-09-19:
Hi @Sharrrrk 
I was installing the NVIDIA driver using the runfile (local) method. You might already be aware of this but you can install only the driver, or any of the other software by selecting/deselecting, e.g., all the runtime dependencies.

```
lqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqk
x CUDA Installer                                                               x
x - [X] Driver                                                                 x
x      [X] 515.65.01                                                           x
x - [ ] CUDA Toolkit 11.7                                                      x
x    - [ ] CUDA Libraries 11.7                                                 x
x       + [ ] CUDA Development 11.7                                            x
x       - [ ] CUDA Runtime 11.7                                                x
x            [ ] cuda-cudart                                                   x
x            [ ] cuda-nvrtc                                                    x
x            [ ] libcublas11                                                   x
x            [ ] libcufft                                                      x
x            [ ] libcurand                                                     x
x            [ ] libcusolver                                                   x
x            [ ] libcusparse                                                   x
x            [ ] libnpp                                                        x
x            [ ] libnvjpeg                                                     x
x    - [ ] CUDA Tools 11.7                                                     x
x       + [ ] CUDA Command Line Tools 11.7                                     x
x       + [ ] CUDA Visual Tools 11.7                                           x
x    - [ ] CUDA Compiler 11.7                                                  x
x         [ ] cuda-cuobjdump                                                   x
x                                                                              x
x Up/Down: Move | Left/Right: Expand | 'Enter': Select | 'A': Advanced options x
```

Sharrrrk, 2022-09-20:
Hi @kasperornmeck  thanks for the information, I will compare the docker size between manually install and cuda-runtime.

Sharrrrk, 2022-09-20:
Currently I have started to change Dockerfile and ansible scripts, the main ideas:
-     Use cuda runtime images as base
-     Only runtime packages of TensorRT are installed
-     Only minimal set of ROS packages are installed
-     Multistage image build, copy install folder

BonoloAWF, 2023-01-18:
@Sharrrrk have you created a PR for these changes? If so, please link it to the Issue.

xmfcx, 2023-04-28:
I've created this discussion on re-layering of the dockerfile/image:

https://github.com/orgs/autowarefoundation/discussions/3469

kaspermeck-arm, 2023-07-06:
Merge this issue with https://github.com/autowarefoundation/autoware.universe/issues/4111.

xmfcx, 2023-11-16:
Closing this issue since it will be addressed by:
- https://github.com/autowarefoundation/autoware/issues/3999

closed 2023-11-16 by xmfcx