autowarefoundation/autoware #2863: Ray ground filter is not actually removing ground points

jfqureshi, 2022-09-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hello, I am using Autoware.ai + Carla Simulator. The issue I have is that my `ray_ground_filter` is actually not removing the ground points. So, my vehicle is unable to move because it detects the ground points. Any help would be appreciated.

### Expected behavior

The output points topic `points_no_ground` should be without ground points.

### Actual behavior

The ground points are not removed from the output points topic `points_no_ground` .

### Steps to reproduce

Use `ray_ground_filter` node with `points_raw `as input topic.

### Versions

Ubuntu 18.04
ROS Melodic
Autoware.ai 1.14

### Possible causes

Bug in the code

### Additional context

Please find the visualization of the no ground topic. i,e `points_no_ground`
https://drive.google.com/drive/u/0/my-drive

mitsudome-r, 2022-09-15:
@jfqureshi We cannot determine if this is a bug or parameter issue just from your information.
I think your shared link to google drive isn't valid. If you want to share the file, you have to right click your file and get the link through "Share" option. 

Have you tried playing around with the parameters in ray_ground_filter node?
[Relevant code](https://github.com/autowarefoundation/autoware_ai_perception/blob/411330c5bb0eb5a963393f869c19d7815bb2dafc/points_preprocessor/nodes/ray_ground_filter/ray_ground_filter.cpp#L435)
Here are some of the parameters that you might want to checkout.
* min_height_threshold: points lower than this value will be removed
* local_max_slope: angle of slope that the node use to determine if the point is ground or not. Roughly speaking, if the angle between two points has larger than this value, it will be considered as an obstacle.
* min_point_distance: Minimum valid distance, points closer than this will be removed.

Also, you might want to make sure that you have tf published between your sensor frame and base_link frame.

jfqureshi, 2022-09-15:
@mitsudome-r Thanks for the answer and sorry for the wrong link. Here is the correct link [https://drive.google.com/file/d/1X26bm_zTzDOzGxVh6XLew3RIIbcYLUAy/view?usp=sharing](url)
I think the issue is with the parameters(quite convinced). I tried to play with them but I still couldn't get to a point where all the ground points are removed. I have my Lidar at (x,y,z = 0,0,2 m) from the base_link.

Also I wanted to ask the `min_height_threshold` is with respect to Lidar frame, right?
Here are my ray_ground parameters:

- `clipping_height` = 2.0
- `local_max_slope` = 5.0
- `min_point_distance` = 1.85

closed 2022-10-18 by jfqureshi