autowarefoundation/autoware #2883: Autoware humble doesn't compile without CUDA

xmfcx, 2022-09-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Is there a flag to disable CUDA? I couldn't find one in the documentation.

I've also gone through the https://github.com/autowarefoundation/autoware.universe/blob/main/common/trtexec_vendor/CMakeLists.txt file too but couldn't find one.

### Expected behavior

[trtexec_vendor](https://github.com/autowarefoundation/autoware.universe/tree/main/common/trtexec_vendor) compiles.

### Actual behavior

[trtexec_vendor](https://github.com/autowarefoundation/autoware.universe/tree/main/common/trtexec_vendor) doesn't compile.

```bash
--- stderr: trtexec_vendor                                                                     
CUDA_TOOLKIT_ROOT_DIR not found or specified
CMake Error at CMakeLists.txt:15 (find_package):
  By not providing "Findcudnn_cmake_module.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "cudnn_cmake_module", but CMake did not find one.

  Could not find a package configuration file provided by
  "cudnn_cmake_module" with any of the following names:

    cudnn_cmake_moduleConfig.cmake
    cudnn_cmake_module-config.cmake

  Add the installation prefix of "cudnn_cmake_module" to CMAKE_PREFIX_PATH or
  set "cudnn_cmake_module_DIR" to a directory containing one of the above
  files.  If "cudnn_cmake_module" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< trtexec_vendor [1.22s, exited with code 1]
```

### Steps to reproduce

1. Compile autoware without having cuda installed.

### Versions

- OS: Ubuntu 22.04
- ROS: ROS2 Humble

kenji-miyake, 2022-09-21:
@xmfcx Could you rerun `rosdep install`? It seems new dependencies were added.

kenji-miyake, 2022-09-21:
The no-CUDA build passes in CI.
https://github.com/autowarefoundation/autoware.universe/actions/workflows/build-and-test.yaml

xmfcx, 2022-09-21:
Hmm it installed
```
executing command [sudo -H apt-get install -y ros-humble-cudnn-cmake-module]
executing command [sudo -H apt-get install -y ros-humble-tensorrt-cmake-module]
```

Now it seems to be working. Thanks for the help.

closed 2022-09-21 by xmfcx

Omar-Mabrouk-9, 2024-11-11:
I am trying to use autoware without cuda, but still have problems. @xmfcx can you guide on how you did it

xmfcx, 2024-11-11:
@Omar-Mabrouk-9 please post a question in https://github.com/orgs/autowarefoundation/discussions/categories/q-a for this.

Omar-Mabrouk-9, 2024-11-11:
thanks, here is the link for the Q-A
https://github.com/orgs/autowarefoundation/discussions/5422