autowarefoundation/autoware #3069: Rviz2 drops to 1-5fps when enabling pointcloud2 topic in planning simulation tutorial

TZECHIN6, 2022-11-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I have created my own dockerfile (base image: nvidia/cuda:11.6.2-cudnn8-devel-ubuntu20.04), basically I follow the manual install step and implement back to a dockerfile. I run the docker with below:
```
docker run --name autoware-devel -dt --restart unless-stopped --gpus all --network host \
    -e DISPLAY=$DISPLAY \
    -v /tmp/.X11-unix/:/tmp/.X11-unix \
    -v $USER_VOLUME_1 \
    -v $USER_VOLUME_2 \
    $CONTAINER_IMAGE $USER_COMMAND
```
The autoware project can be build without error and the planning simulation demo can also be run as usual.

However, the fps only achieve ~5fps, which is not usual, as other team member who had a worst PC spec than mine (i5-10400, 3090GPU, 32GB RAM) still run it as 30fps. When only starting rviz2, it is at 30fps.

[image]
Above image shows the cuda driver should be installed correctly.

I can provide the full Dockerfile if needed. Please let me know if this is the case.

### Expected behavior

Run planning simulation demo smoothly (30fps or more depends on global config)

### Actual behavior

Planning simulation demo only got ~5fps, not only this demo, I have also tried different .pcd file, same result low fps.

### Steps to reproduce

1. Docker build with nvidia cuda base image
2. Install required dependencies in the Dockerfile.
3. Enter the docker environment with specific `docker run` command
4. clone the project where the docker is mounted
5. rosdep install
6. colcon build
7. -----build success-----
8. source autoware/install/setup.bash
9. ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit

Pay attention to the fps

### Versions

- OS: Ubuntu 20.04.5
- ROS2: Galactic
- Autoware: universe
- Docker environment

### Possible causes

Google it a bit, not much topic talking about this.

Most of the thread is talking about nvidia driver issue, but in my contaitner, I think its working as normal.. at least it show the correct message when verifying the installation.

BonoloAWF, 2022-11-29:
@TZECHIN6 please post your question here https://github.com/orgs/autowarefoundation/discussions/categories/q-a for faster assistance

TZECHIN6, 2022-11-30:
@BonoloAWF Thanks for the advice, will do it now.

TZECHIN6, 2022-12-09:
DDS Tuning fix the problem.

closed 2022-12-09 by TZECHIN6