autowarefoundation/autoware #3096: Remove tier4_ad_api_adaptor

isamu-takagi, 2022-12-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, [tier4_ad_api_adaptor](https://github.com/tier4/tier4_ad_api_adaptor.git) is imported as a sample API. However, it is now possible to run the vehicle even with the newly introduced AD API. It's confusing to have two API examples, so I plan to remove [tier4_ad_api_adaptor](https://github.com/tier4/tier4_ad_api_adaptor.git).

### Purpose

Unify API samples to [default_ad_api](https://github.com/autowarefoundation/autoware.universe/tree/main/system/default_ad_api).

### Possible approaches

1. Replace code that uses [tier4_ad_api_adaptor](https://github.com/tier4/tier4_ad_api_adaptor.git) with [default_ad_api](https://github.com/autowarefoundation/autoware.universe/tree/main/system/default_ad_api).
2. Remove [tier4_ad_api_adaptor](https://github.com/tier4/tier4_ad_api_adaptor.git) from autoware.repos.

### Definition of done

Remove [tier4_ad_api_adaptor](https://github.com/tier4/tier4_ad_api_adaptor.git) from autoware.repos

xmfcx, 2023-11-16:
Hello @isamu-takagi -san,

https://github.com/autowarefoundation/autoware/blob/c9c4546705bc5a3ef4ebaff0cd151e32b1a5b16b/autoware.repos#L28-L31
Since this comment is present, I guess this issue is still relevant.

Is there a timeline for this issue to be completed? (Is this low priority?)

Could you provide a list of packages to be modified/refactored to achieve this task?

isamu-takagi, 2023-11-17:
@xmfcx [RTC has not been deleted yet.](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_autoware_api_launch/launch/autoware_api.launch.xml#L23-L29)

This will be completed by migrating to this API. I'm currently implementing this API.
https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-interfaces/ad-api/features/cooperation/

xmfcx, 2023-11-17:
@isamu-takagi thanks for the information.

I couldn't find Cooperation API in the [default_ad_api](https://github.com/autowarefoundation/autoware.universe/tree/main/system/default_ad_api/src).

> I'm currently implementing this API.

Is it being implemented into the `default_ad_api` package?

Are there ongoing PRs for this task?

isamu-takagi, 2023-11-17:
There are no PRs for AD API yet. I'm currently changing the implementation to match the API.
https://github.com/autowarefoundation/autoware.universe/pull/5590

xmfcx, 2025-02-17:
@isamu-takagi -san
I've searched the current Autoware workspace for the usages of this package but couldn't find any. Can we remove `tier4_ad_api_adaptor` from Autoware completely now?

`sudo apt install ripgrep -y`

```console
mfc@bunny:~/projects/autoware$ rg "tier4_ad_api_adaptor" --hidden --no-ignore --glob '!*.rst' --glob '!build/**' --glob '!install/**' --glob '!log/**' --glob '!**/.git/**' --glob '!**/.idea/**' .
./autoware-nightly.repos
22:  universe/external/tier4_ad_api_adaptor: # TODO(TIER IV): Migrate to AD API and remove this repository entry.
24:    url: https://github.com/tier4/tier4_ad_api_adaptor.git

./src/universe/external/tier4_ad_api_adaptor/.github/workflows/sync-micro-bus.yaml
25:          sync-target-repository: https://github.com/tier4/tier4_ad_api_adaptor

./autoware.repos
37:  universe/external/tier4_ad_api_adaptor: # TODO(TIER IV): Migrate to AD API and remove this repository entry.
39:    url: https://github.com/tier4/tier4_ad_api_adaptor.git
```

isamu-takagi, 2025-02-17:
@xmfcx The following packages in tier4_ad_api_adaptor are used when `launch_deprecated_api=true`. So, the option is fine if it is set to false (default value), but setting it to true will cause an error. 

- autoware_iv_external_api_adaptor
- autoware_iv_internal_api_adaptor
- awapi_awiv_adapter

Releted files:
- https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_autoware_api_launch/package.xml#L16-L19
- https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_autoware_api_launch/launch/deprecated_api.launch.xml

isamu-takagi, 2025-04-08:
We are working on separating `tier4_ad_api_adaptor` from autowarefoundation repository this year. First, change the package in the repository as follows, then remove it.

[image]

closed 2026-02-16 by xmfcx