autowarefoundation/autoware #3119: Source installation failed: ndt_omp

cjffly, 2022-12-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

After Follow the source installation doc, when I am trying to build the workspace using 
`colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release`
it reports:
```
--- stderr: ndt_omp                                                           
make[2]: *** No rule to make target '/usr/lib/gcc/x86_64-linux-gnu/9/libgomp.so', needed by 'libndt_omp.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/ndt_omp.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
```

### Expected behavior

Success build

### Actual behavior

Build failed.

### Steps to reproduce

`colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release`

### Versions

- OS: Pop!_OS 22.04 LTS (Ubuntu 22.04)
- ROS2: Humble
- Autoware: main [14e1486](https://github.com/autowarefoundation/autoware/commit/14e1486fe469f8becb1a91b05a09c45160cd6d6d)

### Possible causes

gcc version issue

### Additional context

On my system, the gcc version is 
```
~/autoware$ gcc --version
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
```
The libndt_omp.so seems to asking gcc version 9

kenji-miyake, 2022-12-09:
@cjffly Was your Ubuntu 22.04 clean-installed? Or was upgraded from 20.04? I guess there are some issues with your environment.
My gcc version is also `gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0`, but can successfully build ndt_omp.

cjffly, 2022-12-09:
> @cjffly Was your Ubuntu 22.04 clean-installed? Or was upgraded from 20.04? I guess there are some issues with your environment. My gcc version is also `gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0`, but can successfully build ndt_omp.

You are right, mine is upgraded from 20.04. Not sure if there is quick fix or I have to reinstall 22.04 from scratch.

cjffly, 2022-12-10:
> I guess there is some cache remaining, or your environment variable is wrong. Please check them.
> @cjffly If it's not a bug, can I close this issue? Or can I convert this issue to a discussion thread?

Maybe concert into issue would work. I will try to fix my environment and update here later