autowarefoundation/autoware #3165: No package mactching ros-galactic-desktop available  for AWSIM-stable branch in Ubuntu 22.04

anilbommareddy, 2022-12-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/

git clone https://github.com/autowarefoundation/autoware.git  /*humble branch-Ubuntu 22.04*/
cd autoware
git checkout awsim-stable
./setup-dev-env.sh   */enter Y and install cuda-Y.
got an error No package matching ros-galactic-desktop.

And continued with below steps got an error in PCL lib's.

mkdir src
vcs import src < autoware.repos
vcs import src < simulator.repos.
rosdep update
source /opt/ros/humble/setup.bash
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
Build-workspace:
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

### Expected behavior

ros-humble-desktop. download  

### Actual behavior

got an error No package matching ros-galactic-desktop.

### Steps to reproduce

steps are update in the description.

kenji-miyake, 2022-12-27:
@shmpwk Could you tell me the right branch we should use?

shmpwk, 2022-12-27:
@anilbommareddy 
Currently, the official support for AWSIM is only ROS2 Galactic, Ubuntu20.04.

For Ubuntu 22.04, try work in progress version which includes our individual repos but will be officially supported soon:
Autoware: https://github.com/shmpwk/autoware/tree/humble-awsim-stable
AWSIM: https://github.com/tier4/AWSIM/tree/humble

Try the following command
```
git clone https://github.com/shmpwk/autoware/
cd autoware
git checkout humble-awsim-stable
./setup-dev-env.sh */enter Y and install cuda-Y.
mkdir src
vcs import src < autoware.repos
source /opt/ros/humble/setup.bash
rosdep update
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w"
source install/setup.bash
ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=<your mapfile location>
```

patelabhay-12, 2023-01-04:
hii @shmpwk 
Shall we install autoware for AWSIM in ubuntu 22.04 by your above steps.

shmpwk, 2023-01-04:
@patelabhay-12 
Still the above step is not official but now should work correctly in Ubuntu 22.04 from the success of my several co-workers  attempt.

patelabhay-12, 2023-01-04:
[image]
okay, But I got the error when giving the below command-
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w"

shmpwk, 2023-01-04:
Let me check several points
- Did you succeeded following without any error:  `./setup-dev-env.sh`, `vcs import src < autoware.repos`, `rosdep update`, `rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO` ?
- Have you built the other Autoware branch under your `~/autoware` folder?

shmpwk, 2023-01-04:
I firstly doubt [this issue](https://github.com/autowarefoundation/autoware/issues/3119) but it seems you have different error.

patelabhay-12, 2023-01-04:
> @patelabhay-12 Let me check several points
> 
>     * Did you succeeded following without any error:  `./setup-dev-env.sh`, `vcs import src < autoware.repos`, `rosdep update`, `rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO` ?
> 
>     * Have you built the other Autoware branch under your `~/autoware` folder?

yes I succeeded in following without any error.

Yes, I built the other Autoware branch under your ~/autoware folder

shmpwk, 2023-01-04:
> Yes, I built the other Autoware branch under your ~/autoware folder

In that case you might mixed the previous build cache. Could you `rm build install log` under `~/autoware` ? and colcon build again?

patelabhay-12, 2023-01-04:
> > Yes, I built the other Autoware branch under your ~/autoware folder
> 
> In that case you might mixed the previous build cache. Could you once `rm build install log` under `~/autoware` ? and colcon build again?

Did the same, & getting error again
[image]

shmpwk, 2023-01-04:
@patelabhay-12 
According to [this issue](https://github.com/koide3/ndt_omp/issues/47), your PCL version is much newer than we expected?

patelabhay-12, 2023-01-04:
[image]

shmpwk, 2023-01-04:
We don't use `pcl-tools` and `python3-pcl`. Could you remove these pakage by `sudo apt remove pcl-tools` and `python3-pcl` ?

shmpwk, 2023-01-04:
FYI, this is our environment. 
[image]

(We might also need `ros-humble-perception-pcl` which should be installed by `rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO`)

patelabhay-12, 2023-01-04:
I did the same for installation of ros-humble-perception-pcl , but its not showing ??
[image]

shmpwk, 2023-01-04:
Hmm... at least `ros-humble-perception-pcl ` is not related to your error for now. (Perhaps it might cause the other perception pakage error later on.)

Could you try this under `~/autoware/`?
```
rm build/ndt_omp/ install/share/ndt_omp/ install/include/ndp_omp -rf 
```
and then colcon build again?

patelabhay-12, 2023-01-04:
did the same & also getting same errors.
[image]
[image]

shmpwk, 2023-01-04:
Could you try manually  `sudo apt install ros-humble-perception-pcl` ? (This is not good way but not harmful to your environment.)

patelabhay-12, 2023-01-04:
[image]
[image]

[image]

ros-humble-perception-pcl is installed successfully, after that i try same and getting same errors

patelabhay-12, 2023-01-04:
same error

[image]

shmpwk, 2023-01-04:
@patelabhay-12 

In your error log, I found your build depends `/usr/local/include/pcl-1.13 ......` which is newer than we expected.
So you should remove it. Now I investigate how to remove it .......

patelabhay-12, 2023-01-04:
I removed it by sudo rm /usr/local/include/pcl-1.13/ -rf
then try colcon build, now its showing that fatal error: pcl/point_types.h: No such file or directory

/home/mytsp01065/Pictures/Screenshots/error1.png
/home/mytsp01065/Pictures/Screenshots/error2.png

shmpwk, 2023-01-04:
OK, in that situation, you have to execute `rm build install log  -rf` under ~/autoware

patelabhay-12, 2023-01-04:
rm build install log  -rf 
after this, Shall I give colcon build

shmpwk, 2023-01-04:
YES!

patelabhay-12, 2023-01-04:
[image]

now iot is showing error in Cmake list

shmpwk, 2023-01-04:
You need ` sudo rm  /usr/share/pcl-1.13/  -rf` , rm build install log  -rf, colcon build

patelabhay-12, 2023-01-04:
after giving  sudo rm  /usr/share/pcl-1.13/  -rf
and rm build install log  -rf

and then i  locate pcl by locate libpcl_  |grep 1.13
it is showing PCL 1.13 , i think these are not removed so we got errors

[image]

[image]

shmpwk, 2023-01-04:
Your pcl under home directory is basically not related and should not remove them.

shmpwk, 2023-01-04:
But you have to remove system area pcl 1.13

patelabhay-12, 2023-01-04:
I removed all the folders, and i will try from scratch, if I face any issue will let you know.

thanks

patelabhay-12, 2023-01-05:
I started from scratch today, and i clone the code by your link only
i am facing the error in changing the branch to humble , due to that lock in autoware and i am not able to change that permissions

[image]

shmpwk, 2023-01-05:
I am new to the error. Could you investigate the permission error around git like https://www.google.com/search?q=git+config+global+add+safe.directory+%22permission+denied%22&ei=gXG2Y-a4BI6FoAT-wrWwAg&ved=0ahUKEwim_b7r6a_8AhWOAogKHX5hDSYQ4dUDCA8&uact=5&oq=git+config+global+add+safe.directory+%22permission+denied%22&gs_lcp=Cgxnd3Mtd2l6LXNlcnAQAzIFCAAQogQyBQgAEKIEOgoIABBHENYEELADOgUIIRCgAUoECEEYAEoECEYYAFCDB1iIF2C1GGgBcAF4AIABpgGIAYQCkgEDMS4xmAEAoAEByAEKwAEB&sclient=gws-wiz-serp ?

patelabhay-12, 2023-01-05:
resolved, thanks

patelabhay-12, 2023-01-05:
after colcon build, I got this
i think that build is completed

[image]

shmpwk, 2023-01-05:
Yes, build succeed!!! :+1:

shmpwk, 2023-01-05:
@anilbommareddy How's it going?

patelabhay-12, 2023-01-05:
Thank you @shmpwk 

for launching autoware, that last command asking for map path location, from where to get that map file??

awsim_sensor_kit map_path:=<your mapfile location??

shmpwk, 2023-01-05:
@patelabhay-12 

map is given on [AWSIM tutorial](https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/).

For more question about AWSIM, please post [AWSIM issue](https://github.com/tier4/AWSIM/issues).

patelabhay-12, 2023-01-06:
after launching autoware, it is showing like this in RViz

[image]

shmpwk, 2023-01-06:
@patelabhay-12 
It seems RViz does not work correctly. 
Could you try [planning simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/) and [rosbag replay simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/) first?

shmpwk, 2023-01-06:
It seems the way to load map or the map itself is wrong.
Typical mistake is the map path is wrong. You have to correctly locate the map and specify the **absolute path**, not relative path.

patelabhay-12, 2023-01-06:
how to give absolute path??
i am not getting

shmpwk, 2023-01-06:
`/home/mytsp01065/......`, for example.
Could you share your launching command?

patelabhay-12, 2023-01-06:
ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=/home/mytsp01065/Downloads

shmpwk, 2023-01-06:
If your directory is like this, `map_path:=/home/mytsp01065/Downloads/nishishinjuku_autoware_map`
- Downloads
  -   nishishinjuku_autoware_map
    -  lanelet2_map.osm
    - pointcloud+map.pcd

patelabhay-12, 2023-01-06:
Thank you @shmpwk

shmpwk, 2023-01-06:
Could you try the [tutorial](https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/)?

patelabhay-12, 2023-01-06:
not yet, but wil try

anilbommareddy, 2023-01-12:
@shmpwk  with your provided steps able to build successful on AWSIM-Ubuntu 22.04.
I would like to cross check is AWSIM provide interface  data like  lidar/radar(sensor) data to other application or Hardware.
For an example I have an application(non ROS)  which calculate the position of a tag and i need the get this positions of data(x,y,z) from AWSIM and the calculated position tag  need to display on AWSIM GUI.

shmpwk, 2023-01-12:
@anilbommareddy Sounds great! 

I close the issue since the original problem as stated in the title seems to have been resolved.
If you find a new problem, please post a new issue.

closed 2023-01-12 by shmpwk

patelabhay-12, 2023-01-12:
thank you @shmpwk 

could you plz tell me how to generate the sensor data ??

shmpwk, 2023-01-12:
@patelabhay-12 
The sensor data is published from AWSIM simulator. You can find it when you launch AWSIM binary, as introduced in the tutorial.