autowarefoundation/autoware #3173: rviz2 failed/crash with additional monitor

cjffly, 2023-01-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When launching rviz2 with the default config file in autoware/src/launcher/autoware_launch/autoware_launch/rivz/autoware.rviz, the rviz2 window will appear for a second and then close. The console will reports:
```
[ERROR] [rviz2-46]: process has died [pid 220579, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2
 -d /home/thing1/autoware/install/autoware_launch/share/autoware_launch/rviz/autoware.rviz 
 -s /home/thing1/autoware/install/autoware_launch/share/autoware_launch/rviz/image/autoware.png --ros-args -r __node:=rviz2 -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
```

This only happens when my laptop is connected to a extended monitor. (if no extended monitor was connected, it will work fine)

### Expected behavior

rviz2 launch nomally

### Actual behavior

rviz2 window appear for a second and then closed.

### Steps to reproduce

1. Have a laptop connected to a monitor
2. Start rviz2 with `rviz2 -d ~/autoware/install/autoware_launch/share/autoware_launch/rviz/autoware.rviz`

### Versions

- OS: PopOs 22.04
- ROS2: Humble
- Autoware: latest from source

### Possible causes

Solution: Not sure the cause of this bug, but I found the solution:

Open the .rviz config file with an editor, find and delete the following 2 part:
* First:
```
  - Class: AutowareDateTimePanel
    Name: AutowareDateTimePanel
```
* Second:
 The number after `QMainWindow State:`

I am not clear what these two part do for rviz, but deleting them solved my problem.

kyoichi-sugahara, 2023-01-12:
@cjffly 
Thank you for the report!!
I had same issue and solved problem thanks to your suggested proposal.
I don't know the cause though....

taikitanaka3, 2023-01-20:
@isamu-takagi 

Do you have any idea about this issue?

isamu-takagi, 2023-01-20:
@taikitanaka3 Thank you for the notification. I've never had this problem before, so I'll try it in an environment with an extra monitor.

isouf, 2023-01-22:
Thank for reporting this issue. 

I have observed the same behaviour (`rviz2 window appear for a second and then closed.`) but on a single laptop screen (Dell XPS 15 9520 RTX 3050 Ti). 

I have exactly the same versions of OS, ROS and Autoware but I am running it via Docker ([`humble-latest-cuda`](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/63596980?tag=humble-latest-cuda)). My NVIDIA Driver and CUDA versions inside the Docker container: `Driver Version: 525.60.13    CUDA Version: 12.0`.

The problem appears when I set the resolution of my screen to 3456 x 2160 (16:10) with 200 % scale factor. I can open `rviz2` if I change the scale factor to 100 %, but I cannot set the initial pose (see relevant Discord thread [here](https://discord.com/channels/953808765935816715/1060241579291451473/1060901801982955560)). After trying multiple screen resolutions and scale factors I found that 1680 x 1050 (16:10) with 100 % scale factor is the one that works me (I can now open `rviz2` and set the initial/goal poses).

isamu-takagi, 2023-01-26:
I can't reproduce it, but I found the suspicious code. Can you check if this PR will fix it?
https://github.com/autowarefoundation/autoware.universe/pull/2753

kyoichi-sugahara, 2023-01-27:
I tried this [PR](https://github.com/autowarefoundation/autoware.universe/pull/2753)
But the problem is not fixed...

AhmadAmine998, 2023-07-04:
Hello, I was facing this same issue with both a single laptop and a dual-monitor setup. It happens to me when the screen resolution of either is set to 3840x2160 (4K). 

A solution to this is: 
1. Set your display to 1920x1080 or 100% scaling.
2. Start rviz2 with `rviz2 -d ~/autoware/install/autoware_launch/share/autoware_launch/rviz/autoware.rviz`
3. Uncheck the PointCloadMap visualization (Map > PointCloudMap). 
[image]
6. Save the autoware.rviz config.
7. Set the display back to your native resolution and desired scaling.
8. Start rviz2 again with `rviz2 -d ~/autoware/install/autoware_launch/share/autoware_launch/rviz/autoware.rviz`

This way the PointCloudMap topic is not visualized on startup since PointCloudMap consumes a lot of memory. I was able to enable the PointCloudMap again after startup. It seems that rviz2 consumes a lot of memory on launch when rendering on a 4K screen and unchecking PointCloudMap helps reduce the load. Hope this helps.

isamu-takagi, 2023-07-12:
I feel like rviz crashes if config file contains many plugins. The same behavior occurs with plugins other than AutowareDateTimePanel, and the following exception is sometimes thrown.
```
[ERROR] [1685087505.357431157] [rviz2]: InternalErrorException: Vertex Buffer: Out of memory in GLHardwareVertexBuffer::lock at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GL/src/OgreGLHardwareVertexBuffer.cpp (line 125)
terminate called after throwing an instance of 'Ogre::InternalErrorException'
  what():  InternalErrorException: Vertex Buffer: Out of memory in GLHardwareVertexBuffer::lock at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GL/src/OgreGLHardwareVertexBuffer.cpp (line 125)
```

isamu-takagi, 2023-07-21:
Related discussions:
https://github.com/orgs/autowarefoundation/discussions/3474
https://github.com/orgs/autowarefoundation/discussions/3121