autowarefoundation/autoware #3222: Generate Debian packages for the ROS packages in Autoware.core and Autoware.universe

esteve, 2023-01-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Given that Ubuntu is the platform of choice for developing and running Autoware, providing binary Debian packages will help users deploying Autoware

### Purpose

Provide Debian packages for our users and developers

### Possible approaches

Integrate https://github.com/jspricke/ros-deb-builder-action into our GitHub actions workflow

### Definition of done

Debian packages are generated for all ROS packages and can be downloaded from our repositories

isamu-takagi, 2023-01-24:
I made some changes to actions to support a differential build. I would like to decide a release operation of Autoware and propose features to upstream to support it.

https://github.com/tier4/autoware-apt-testing/tree/main
https://github.com/isamu-takagi/ros-deb-builder-action/tree/develop (contains a lot of temporary code)

The features I think we need are:
- support differential build
- support multiple suite/components/architectures 
- use version in package.xml

isamu-takagi, 2023-01-24:
My idea is that each package will only build when the version tag in package.xml is changed, and the repository keeps all previously built versions.

esteve, 2023-01-24:
@isamu-takagi thanks for listing all the features, really nice.

I prefer to follow what MoveIt is doing:

https://github.com/moveit/moveit2_packages/blob/main/.github/workflows/build.yaml#L6

it runs once every day (in MoveIt's case, at 1:17am), so there's no need for differential builds, because it doesn't runs when a new branch is merged. But I think using the version from package.xml would be a useful alternative to using tags, though it's strongly encouraged for ROS packages/repositories to use tags for every new release.

esteve, 2023-01-24:
@isamu-takagi I tried adding you as an assignee to this issue, but for some reason your username does not show up on the list. @kenji-miyake do you know why I can't add @isamu-takagi ?

isamu-takagi, 2023-01-24:
> it runs once every day (in MoveIt's case, at 1:17am), so there's no need for differential builds, because it doesn't runs when a new branch is merged.

My concern is that the build takes a long time. If the dependent packages have not changed, it is efficient to use the released ones.

> But I think using the version from package.xml would be a useful alternative to using tags, though it's strongly encouraged for ROS packages/repositories to use tags for every new release.

If the package has not changed, the same binary will be generated as a different version.

kenji-miyake, 2023-01-24:
> @kenji-miyake do you know why I can't add @isamu-takagi ?

@esteve Hmm, sorry I'm not sure. I believe you have Maintain access to this repository.
For now, I'll add him.

jspricke, 2023-01-24:
> My concern is that the build takes a long time. If the dependent packages have not changed, it is efficient to use the released ones.

I measured about a minute for a package that is already in ccache on the free Github VMs. We can speed it up by using a ram disk if we have more powerful VMs. Also we think about parallelizing the build.
Note that it is not enough to compare the diff in the repo itself as upstream dependencies could have changed and require a rebuild (ABI changes, message changes..).

> > But I think using the version from package.xml would be a useful alternative to using tags, though it's strongly encouraged for ROS packages/repositories to use tags for every new release.
> 
> I think there is a request to use the old package version. in this case, If the package has not changed, the same binary will be generated as a different version.

Currently the package includes the date as a Debian revision to make sure new version are recognized as an update by apt. This is the same for the OSRF buildfarm. Generating a Debian revision only in case of a change is not trivial but we can discuss that.

Can you please send relevant patches upstream so we can discuss them?

esteve, 2023-01-25:
> Can you please send relevant patches upstream so we can discuss them?

We will when we feel like we have something that can be upstreamed, until then we'll continue with our experiments here.

isamu-takagi, 2023-01-25:
> Can you please send relevant patches upstream so we can discuss them?

https://github.com/isamu-takagi/ros-deb-builder-action/pull/1
This is the current diff, but not enough discussion. I would like to continue the discussion here.

>Currently the package includes the date as a Debian revision to make sure new version are recognized as an update by apt. This is the same for the OSRF buildfarm. Generating a Debian revision only in case of a change is not trivial but we can discuss that.

The autoware.universe repository contains a lot of packages, so versioning with tags is not suitable. Multiple versions of deb packages for each tag is generated even if the code of stable package hasn't changed. So in current repository structure, we need versioning for each package.

If versioning by tag, the autoware.universe repository should be divided into multiple small repositories. This allows the repository/package maintainers to set the version at their release timing. If using this method, It's need to deploy deb packages to another repository for distribution.

jspricke, 2023-01-25:
> [isamu-takagi/ros-deb-builder-action#1](https://github.com/isamu-takagi/ros-deb-builder-action/pull/1) This is the current diff, but not enough discussion. I would like to continue the discussion here.

Fine with me. I'm also open to a video call if you are interested.

> The autoware.universe repository contains a lot of packages, so versioning with tags is not suitable. Multiple versions of deb packages for each tag is generated even if the code of stable package [...]

I think that is a good idea in general (though unrelated to packaging) as monorepos do not work well with git.

> ~If using this method, It's need to deploy deb packages to another repository for distribution.~ I noticed `sources.repos` resolve this.

Not sure I got you here but it is also possible to pass extra apt repositories to sbuild via SBUILD_CONF.

isamu-takagi, 2023-01-25:
>If versioning by tag, the autoware.universe repository should be divided into multiple small repositories. This allows the repository/package maintainers to set the version at their release timing.

What do you think about this? @mitsudome-r @yukkysaito @kenji-miyake

esteve, 2023-01-25:
> Fine with me. I'm also open to a video call if you are interested.

We can do that, @isamu-takagi is based in Japan and I'm in Spain, I'll send you an email.

yukkysaito, 2023-01-25:
> What do you think about this? @mitsudome-r @yukkysaito @kenji-miyake @xmfcx

@isamu-takagi 
I am thinking now...
On the negative side, some packages have dependencies between packages, and it is a pain to manage PR across packages when modifying functionality.

Are you considering a repository split for each package? Or per component?

kenji-miyake, 2023-01-25:
>> If versioning by tag, the autoware.universe repository should be divided into multiple small repositories. This allows the repository/package maintainers to set the version at their release timing.
>
> What do you think about this? @mitsudome-r @yukkysaito @kenji-miyake @xmfcx

@isamu-takagi I think flexible ways are better ways. I mean, reading the version in `package.xml` is suitable at least for Autoware as of today. If someone want to use Git tags, then they can update the version in `package.xml` when they create a tag.

jspricke, 2023-01-25:
> @isamu-takagi I think flexible ways are better ways. I mean, reading the version in `package.xml` is suitable at least for Autoware as of today. If someone want to use Git tags, then they can update the version in `package.xml` when they create a tag.

Note that the `package.xml` version does not accommodate changes in git so new commits will have the same version number. That's why we use `git describe --tags`.

isamu-takagi, 2023-01-26:
> Note that the package.xml version does not accommodate changes in git so new commits will have the same version number. That's why we use git describe --tags.

What is the difference between tag and package.xml? New commits will have the same version unless a new tag is created. I think it can be replaced by just updating the package.xml instead of creating the tag. Did I miss something?

jspricke, 2023-01-26:
> What is the difference between tag and package.xml? New commits will have the same version unless a new tag is created. I think it can be replaced by just updating the package.xml instead of [...]

Using `git describe --tags` will generate a version number specific to the used commit. It has the form `<tag>-<number of commits on top>-g<sha1 of commit>`. This makes sure that new commits have a higher version number (due to the counter) and that one can identify the commit belonging to the package (by the sha1).

jspricke, 2023-01-26:
Note that this is mostly relevant when creating packages for non tagged versions such as nightly builds. For tagged release builds `git describe --tags` will resolve to the tag so it should be the same as what is in the package.xml

isamu-takagi, 2023-01-26:
I understood. At least, if it has a build time suffix as debian version, either tag or package.xml is fine. But using `git describe --tags` will give a commit hash.

isamu-takagi, 2023-01-26:
So my understanding is that there is not much difference there. If using the package.xml version, It will be replaced like this: `$(package.xml varsion)-$(time)-$(rev-parse HEAD)`.

jspricke, 2023-01-26:
> So my understanding is that there is not much difference there. If using the package.xml version, It will be replaced like this: `$(package.xml version)-$(time)-$(rev-parse HEAD)`.

That could result in updating to older commits if they are rebuild as it is missing the number of commits on top. Also we currently don't add the hash at the end.

isamu-takagi, 2023-01-26:
It might be better to consider nightly build and release build separately. I'll think it again.

isamu-takagi, 2023-01-26:
> That could result in updating to older commits if they are rebuild as it is missing the number of commits on top. Also we currently don't add the hash at the end.

That’s right. It is the advantage of `git describe`.

isamu-takagi, 2023-01-26:
In conclusion, I found that there is no problem with version management by tags for a while.

After thinking about it, using tags is fine with for latest releases and nightly builds. For these, simply build the latest main branch or a specific tag. It's a bit unnatural for the version to be updated even if the package hasn't changed, but it doesn't matter.

My concern is for stable release. The reason I want to avoid using tags is that the Autoware repository is too large. Since the version of the repository is reflected in all packages, it is not possible to control the release timing for each package. This will the package is released with the latest code even if the package is not stable and maintainer want to provide the previous version as stable.

However, there is no problem with tag version at this stage. This is because the current Autoware is under development and it is difficult to provide a stable version. So as a first step, creating a tag for the entire autoware.universe and providing all packages with the same version of that tag is a reasonable choice.

In the future we need to support individual releases of packages. At this time, the version in package.xml is required, or split the repository. As @yukkysaito said, splitting the repository has the disadvantage of managing PR, but I hope that the code will be stable and easy to handle at this time.

kenji-miyake, 2023-01-26:
@jspricke @isamu-takagi Thank you for the additional information and summaries.

There was one thing I misunderstood.
When we say "use tags", I think there are two meanings:
1. Update packages only when a new tag is released.
2. Use tag-based information (such as `git describe --tags`) to determine the package versions, but they are built regularly.

I imagined the No. 1, but it seems the important point is mainly the No. 2, right?
In that case, I think it'd be okay for now.

In the future, perhaps we can consider other options like "updating packages only when package.xml is updated, and keeping all the versions (not only the latest) of deb packages.", as I mentioned above.
But I know there might be a problem then; the repository size would be extremely large.

Please correct me if I'm saying something wrong.

kenji-miyake, 2023-01-26:
Regarding stable releases, I think we can resolve it by preparing multiple repositories (channels).
For example, prepare the following 3 channels/repositories:
1. Stable/autoware-deb-packages-stable
2. Beta (Latest release)/autoware-deb-packages-beta
3. Nightly/autoware-deb-packages-nightly

It's very similar to Rust's way.
https://doc.rust-lang.org/book/appendix-07-nightly-rust.html

And set the different timings of updating the packages.
- Stable is updated when a maintainer manually updated the tag to use. It's assumed to be done after passing some tests such as scenario tests.
- Beta is updated when a new tag is pushed.
- Nightly is updated daily.

Then, users select the repository to use by changing the URL of add-apt-repository. (Or we can provide some commands to switch the channels.)

What do you think about it?

kenji-miyake, 2023-01-26:
> That could result in updating to older commits if they are rebuild

@jspricke By the way, regarding the above part in https://github.com/autowarefoundation/autoware/issues/3222#issuecomment-1404688816, when will they be rebuilt with older commits? I'm not sure when it happens. Is that when the maintainer reruns the workflow?
The question is just out of curiosity.

jspricke, 2023-01-26:
> @jspricke By the way, regarding the above part in [#3222 (comment)](https://github.com/autowarefoundation/autoware/issues/3222#issuecomment-1404688816), when will they be rebuilt with older commits? I'm not sure when it happens. Is that when the maintainer reruns the workflow? The question is just out of curiosity.

For reruns, yes. For example when you rebuild everything do to an ABI change in an upstream library.

isamu-takagi, 2023-01-27:
@esteve @jspricke @kenji-miyake @yukkysaito @mitsudome-r 
Thank you for the meeting. From what we talked about, it would be nice to provide a nightly build as a first step. So how about creating a new repository something like `autoware-deb-packages-nightly`?

esteve, 2023-01-27:
@isamu-takagi 

> Thank you for the meeting. From what we talked about, it would be nice to provide a nightly build as a first step. So how about creating a new repository something like autoware-deb-packages-nightly or autoware-apt-repository-nightly?

That's one option, the other option is to keep them all in one repository and use branches for nightly, stable, etc. That's what the [moveit2_packages](https://github.com/moveit/moveit2_packages) repository does for different Ubuntu/ROS distros. We could store packages in separate nightly branches named after the date they were built (e.g. `jammy-humble-nightly-20230201`) and always have a tag that points to the latest nightly (e.g. `jammy-humble-nightly-latest`)

jspricke, 2023-01-27:
Or just not set `SQUASH_HISTORY` and have a branch with all the version and the HEAD pointing to the latest ;).

esteve, 2023-01-27:
> Or just not set SQUASH_HISTORY and have a branch with all the version and the HEAD pointing to the latest ;).

I believe it's easier to address each set of nightly builds by branch or tag, otherwise it can get messy.

jspricke, 2023-01-27:
I'm pretty sure it is the opposite but feel free.

esteve, 2023-01-27:
A tag or a branch is more descriptive that a commit id, and each set of nightly builds will be separate from the others

jspricke, 2023-01-27:
That's already the case with the current implementation.

esteve, 2023-01-27:
In that case, that's good, is that with `SQUASH_HISTORY` set to false?

jspricke, 2023-01-27:
which is the default, yes

isamu-takagi, 2023-01-27:
> That's one option, the other option is to keep them all in one repository and use branches for nightly, stable, etc. 

This seems to be good because the settings can be collectively managed in one repository. So let's create `autoware-deb-packages` or `autoware-apt-repository`. What name is intuitive?

> SQUASH_HISTORY

Depends on package size. It is useful to set it to false to create branches or tags at old commit. But if the data is large, it have to be set to true.

jspricke, 2023-01-27:
> > SQUASH_HISTORY
> 
> Depends on package size. It is useful to set it to false to create branches or tags at old commit. But if the data is large, it have to be set to true.

From our experience it does not matter.

esteve, 2023-01-27:
> > 
> 
> This seems to be good because the settings can be collectively managed in one repository. So let's create `autoware-deb-packages` or `autoware-apt-repository`. What name is intuitive?

I had `autoware_packages` in my repo, named after the `moveit2_packages` repo, but I like your options more. `autoware-deb-packages` seems slightly more descriptive to me, but both are fine.

isamu-takagi, 2023-01-27:
@mitsudome-r Could you create a repository named `autoware-deb-packages`?

mitsudome-r, 2023-01-30:
@esteve @isamu-takagi I created the repository.
https://github.com/autowarefoundation/autoware-deb-packages

isamu-takagi, 2023-01-30:
@mitsudome-r @esteve 
Thank you. I'm going to port my test code. It builds autoware_common.

esteve, 2023-01-31:
@isamu-takagi thanks, I'm building all the packages locally with the latest changes merged in, it should hopefully build, will report with the results

esteve, 2023-02-01:
@isamu-takagi build failed on GitHub (see https://github.com/esteve/autoware_packages/actions/runs/4056342022/jobs/6980712591), but it does not seem to be related to Autoware or ros-deb-builder-action. `ndt_omp` needs its dependencies fixed (https://github.com/tier4/ndt_omp/pull/20 is waiting to be reviewed and merged).

I left the build.sh script running on my laptop last night, but there was an unrelated error. I triggered the script again and hopefully it'll build this time.

esteve, 2023-02-01:
@isamu-takagi I got to https://github.com/tier4/tamagawa_imu_driver locally, but the build failed because the latest tag available does not follow the Debian versioning convention. I've filed a ticket at https://github.com/jspricke/ros-deb-builder-action/issues/5

esteve, 2023-02-07:
@knzo25 @yukke42 I've managed to build all the packages up to [lidar_centerpoint](https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_centerpoint), however one of its dependencies is `python3-open3d-pip`, which is only available via PIP. The current version on PyPI for open3d is 0.16, whereas the version available via a Debian package on Ubuntu Jammy is 0.14 Would the Debian package work as well? Otherwise we'd have to backport it from Ubuntu Lunar (which has 0.16)

yukke42, 2023-02-13:
> @knzo25 @yukke42 I've managed to build all the packages up to [lidar_centerpoint](https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_centerpoint), however one of its [...]

@esteve 
Sorry for the late reply. I have checked it works with both open3d ==0.16 and 0.14.1.

knzo25, 2023-02-13:
@esteve 
Could you paste the PR in question (when available) to avoid duplicate work?
:pray: 

https://github.com/autowarefoundation/autoware.universe/pull/2869

esteve, 2023-02-13:
> @esteve Could you paste the PR in question (when available) to avoid duplicate work? pray
> 
> [autowarefoundation/autoware.universe#2869](https://github.com/autowarefoundation/autoware.universe/pull/2869)

I added a link to this PR to the description of this issue so that all the linked PRs are in the same place, otherwise they're linked randomly from the comments which makes it less readable.

esteve, 2023-02-13:
@yukke42 

> @esteve
> Sorry for the late reply. I have checked it works with both open3d ==0.16 and 0.14.1.

Thank you so much for testing it, with this change we should be able to build Autoware using Debian packages only

kenji-miyake, 2023-03-17:
@esteve Great work! :smile: Thank you for sharing the progress.

First, I'll write some additional information.

Regarding velodyne packages, the version is a hack that was intentionally changed because we wanted to force users to use our customized fork.
If the version is the same as or higher than the original one, and if the package is missing for some reason (for example, failed to run "vcs import"), the upstream version could be mistakenly used. That is an unintended behavior.

However, I think it's okay to redesign the way now.

---

Next, questions and my opinions.

> For the first set of packages (the ones depending on CUDA), they could be moved to a separate repository and then have two .repos files (one with the CUDA-dependent packages and one without them)

I think separating the repository is a good idea also considering the CI efficiency. Currently, we run builds of both no-CUDA and with-CUDA, but CUDA isn't related to most of the packages.
The separated repository name would be, `autoware_perception.universe`, `autoware_cuda.universe`, or something like that? :thinking:

And there might be other solutions.
1. Create Debian packages with a CUDA-installed machine.
  - I think it might be possible since CUDA packages are also Debian-packaged, but I'm not sure.
2. Remove CUDA dependencies, for example by replacing CUDA with TVM.

Regarding option 1, I have a question.
Is that possible? I'm not familiar with the `sbuild` and other tools yet.

Regarding option 2, first I'd like to ask @mitsudome-r and @angry-crub about it because they were working on the related tasks.

> For the second set of packages (Velodyne)

I think there are some options.
- As you propose, separate the `.repos` file.
  - However, I guess this has a disadvantage. If we mistakenly skip the `vcs import velodyne.repos`, the upstream version will be used unless we don't have some version-check feature in the dependent packages.
- Submit pull requests to upstream so that it has the same features as TIER IV's customized ones.
- Have the packages in other names like `tier4_velodyne_*`.
- Replace the packages with some alternative packages.

> For the third set of packages (pacmod)

I think adding it to `.repos` can be tentatively allowed.
However, I'd like to find and fix the core cause if possible.
Would it be possible for you to consult it with @jspricke -san, or have you already done that?

---

Regarding the next actions, how about having an online meeting about this? Others might also have questions/opinions, and it would be faster to discuss them verbally.

esteve, 2023-03-17:
> @esteve Great work! smile Thank you for sharing the progress.

Thanks for taking the time to discuss the next steps :slightly_smiling_face: 

> I think separating the repository is a good idea also considering the CI efficiency. Currently, we run builds of both no-CUDA and with-CUDA, but CUDA isn't related to most of the packages. The [...]

I think `autoware_cuda.universe` is fine, if there are any other packages that depend on CUDA we can put them there, whether they are perception or not.

> And there might be other solutions.
> 
>     1. Create Debian packages with a CUDA-installed machine.
> 
> 
>     * I think it might be possible since CUDA packages are also Debian-packaged, but I'm not sure.

The problem with this one is that CUDA is not in the ROS index, so we can't declare it as a dependency in `package.xml`. We could submit a change to the index, but I'm unsure it'll get accepted since the packages for CUDA are not in the official Ubuntu/Debian repositories.

>     2. Remove CUDA dependencies, for example by replacing CUDA with TVM.

This, to me, sounds the best approach. TVM is open source and there are no proprietary dependencies, and we'd allow users to use different backends (e.g. ROCm for users who have an AMD GPU).

> Regarding option 1, I have a question. Is that possible? I'm not familiar with the `sbuild` and other tools yet.

Theoretically yes, but this defeats the use of the GitHub action as part of our CI. It'd be best if we can just use the GitHub infrastructure.

> I think there are some options.
>
>     * As you propose, separate the `.repos` file.
>
>       * However, I guess this has a disadvantage. If we mistakenly skip the `vcs import velodyne.repos`, the [...]

That'd be my preferred option, and any further improvements be submitted to the upstream package to avoid forks.

>     * Have the packages in other names like `tier4_velodyne_*`.

If we can't get our modifications upstreamed, this would work.

> > For the third set of packages (pacmod)
> 
> I think adding it to `.repos` can be tentatively allowed. However, I'd like to find and fix the core cause if possible. Would it be possible for you to consult it with @jspricke -san, or have you already done that?

There's still one option I'm working on, but the core issue here is that AutonomouStuff should submit the packages to the ROS buildfarm, perhaps we could convince them.

Which we should as well. `ros-deb-builder-action` is not a replacement for submitting packages to the ROS buildfarm. From PickNick's post https://picknik.ai/ros/debian/packaging/2023/02/27/packaging-ros-with-github-actions.html : "Note that this is not meant to replace the ROS build farm but as an addition."

Moreover, we could plan on tagging the repos and making sure that version numbers are in sync, which is always a good idea for any project, perhaps we should do that after the next milestone.

> Regarding the next actions, how about having an online meeting about this? Others might also have questions/opinions, and it would be faster to discuss them verbally.

Sounds good! @xmfcx we could discuss this at the next ASWG meeting, would that work?

jspricke, 2023-03-17:
> The problem with this one is that CUDA is not in the ROS index, so we can't declare it as a dependency in `package.xml`. We could submit a change to the index, but I'm unsure it'll get accepted since the packages for CUDA are not in the official Ubuntu/Debian repositories.

At least for the Action you can simply specify them in as an additional `ROSDEP_SOURCE`.

angry-crab, 2023-03-20:
The problem with TVM is that to support different backends, the package need to be recompiled(For example, to enable CUDA, we have to recompile it with CUDA support). That means we have to prebuild combinatoric number of TVM packages which is difficult.

xmfcx, 2023-03-20:
Hi, @esteve!

> The following package depends on `pacmod3_msgs`, which is packaged in https://github.com/astuff/pacmod3#installation. I was able to add AutonomouStuff's APT repository, however it's not signed and I couldn't find a way to tell `sbuild` to trust it anyway (I tried passing the repos as [trusted=yes], but didn't work)

For this I've looked up into [sbuild man page](https://manpages.ubuntu.com/manpages/focal/en/man1/sbuild.1.html):
```md
--extra-repository=spec
       Add a repository to the  list  of  apt  sources  during  the  package  build.   The
       repository  specification  is  a line suitable for an apt sources.list(5) file. For
       instance,  you  might  use   --extra-repository="deb   http://deb.debian.org/debian
       experimental  main"  to  allow packages in the experimental distribution to fulfill
       build-dependencies. Note that the build chroot must already trust the key  of  this
       repository  or  a  key must be given with the --extra-repository-key flag (see apt-
       secure(8)).   This  command  line  option   appends   to   the   EXTRA_REPOSITORIES
       configuration variable. See sbuild.conf(5) for more information.

--extra-repository-key=file.asc
       Add  file.asc  to  the list of trusted keys inside the chroot. The key is read from
       the filename given, and added to the trusted keys. For more information,  see  apt-
       secure(8).  This flag is particularly useful if the target in --extra-repository is
       not signed with a key that's trusted by the base chroot.  This command line  option
       appends to the EXTRA_REPOSITORY_KEYS configuration variable. See sbuild.conf(5) for
       more information.
```

So you've tried following:
`sbuild --extra-repository"deb [trusted=yes] https://s3.amazonaws.com/autonomoustuff-repo/ $(lsb_release -sc) main"`

And it still didn't trust?

esteve, 2023-03-20:
## Unsigned pacmod3 repository

> So you've tried following: `sbuild --extra-repository"deb [trusted=yes] https://s3.amazonaws.com/autonomoustuff-repo/ $(lsb_release -sc) main"`

That's not how the action works, but basically that's what I did (see my comment "I tried passing the repos as `[trusted=yes]`, but didn't work"). You can pass extra arguments to the `build` script that will get passed to `sbuild`. However it doesn't seem that `[trusted=yes` is enough, but what I'm doing now is adding `--extra-package=` with a local copy of the `pacmod3_msgs` package as a test, though we shouldn't rely on that for releasing packages.

> And it still didn't trust?

Nope.

> Is there an easy way for us to replicate this behavior on our machines?

The action runs the `prepare` and `build` scripts, so what I've been doing to replicate the action locally is to run those two scripts. The first one creates a `chroot` with the base dependencies and other stuff, and the `build` script is the one that calls `sbuild` (among other things) to generate the packages.

## CUDA / TVM dependent packages

> I think, we can keep them in their place. Then, add and use a freshly forked custom rosdep database with CUDA repositories that we use. It seems [custom_rosdistro.rst](https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/custom_rosdistro.rst#use-the-forked-rosdep-database-or-keep-using-the-official-one) from this documentation, ros_buildfarm also uses this method to allow custom declarations.
> 
> http://docs.ros.org/en/independent/api/rosdep/html/contributing_rules.html shows how to contribute but we can clone, modify and use it every time we want to build it. What do you think? @kenji-miyake @esteve @jspricke

It's much simpler than that, we can add extra `rosdep` repositories via `ROSDEP_SOURCE`, no need to fully clone the ROS index and keep it synced, `rosdep` allows you to add multiple sources. In any case, the ROS buildfarm and the action are not the same software, they have different documentation, although some things apply to both, so I wouldn't use the ROS buildfarm documentation as reference for the action, and viceversa.

I still think it's cleaner to move the CUDA-related packages to a separate repository so that we have a set of packages that we know can be built with the dependencies currently available in ROS and that don't have any license issues.

jspricke, 2023-03-20:
> ## Unsigned pacmod3 repository
>
> That's not how the action works, but basically that's what I did (see my comment "I tried passing the repos as `[trusted=yes]`, but didn't work"). You can pass extra [...]

Can you share  a log of that?

jspricke, 2023-03-20:
Just tried it myself, you are probably missing the ca-certificates package inside the chroot. You can add that with:
```
SBUILD_CONF: |
  $external_commands = {
    "chroot-setup-commands" => [ [ 'apt-get', 'update'], ['apt-get', 'install', '-y', 'ca-certificates' ],
  };
  $extra_repositories = ["deb [trusted=yes] https://s3.amazonaws.com/autonomoustuff-repo/ focal main"];
```

(replace focal with the target distribution).

esteve, 2023-03-20:
> Just tried it myself, you are probably missing the ca-certificates package inside the chroot. You can add that with:

Thanks, that seems to be the case. Your solution seems fine, but I'm trying with `$core_depends` instead as it's cleaner (see https://manpages.debian.org/bullseye/sbuild/sbuild.conf.5.en.html#CORE_DEPENDS)

[20 more replies]