autowarefoundation/autoware #3225: Create api for visualization

isamu-takagi, 2023-01-25:


tkhmy, 2023-01-31:
## Object Recognition
- The message will be get from `/perception/object_recognition/objects`
- The original message type include some unneeded data, https://github.com/tier4/autoware_auto_msgs/blob/tier4/main/autoware_auto_perception_msgs/msg/PredictedObject.idl
- Therefore, I suggest to summarized the data into following
- Objects
	- UUID object_id
	- float existence_probability
	- [ObjectClassification](https://github.com/tier4/autoware_auto_msgs/blob/tier4/main/autoware_auto_perception_msgs/msg/ObjectClassification.idl) classification
		- uint8 label
		- float probability
	- [PoseWithCovariance](https://docs.ros2.org/foxy/api/geometry_msgs/msg/PoseWithCovariance.html) initial_pose_with_covariance
	- [TwistWithCovariance](https://docs.ros2.org/foxy/api/geometry_msgs/msg/TwistWithCovariance.html) initial_twist_with_covariance
	- [PredictedPath](https://github.com/tier4/autoware_auto_msgs/blob/tier4/main/autoware_auto_perception_msgs/msg/PredictedPath.idl)[100] predicted_paths 
	- [Shape](https://github.com/tier4/autoware_auto_msgs/blob/tier4/main/autoware_auto_perception_msgs/msg/Shape.idl) shape
		- uint8 type
		- Polygon footprint
		- Vector3 dimensionsl

tkhmy, 2023-01-31:
## Detection Area
- Show the area of the autoware that currently looking at.
- Most of the topic publish is in [MarkerArray](https://docs.ros2.org/foxy/api/visualization_msgs/msg/MarkerArray.html), I suggest we convert to [Shape](https://github.com/tier4/autoware_auto_msgs/blob/tier4/main/autoware_auto_perception_msgs/msg/Shape.idl) when in api

isamu-takagi, 2023-02-13:
@tkhmy https://github.com/autowarefoundation/autoware_adapi_msgs/pull/26
I created a general shape message. I think it can be used for both visualization and detection areas.

tkhmy, 2023-02-24:
https://github.com/autowarefoundation/autoware/issues/3232#issuecomment-1439450387

I think we need to have an api that publish current max velocity. And put in under this topic
`/api/planning/max_velocity`

closed 2023-08-31 by isamu-takagi