autowarefoundation/autoware #3232: Create api for vehicle status

tkhmy, 2023-01-26:


isamu-takagi, 2023-01-27:
For `target_velocity` and `target_acceleration`, It might be better to separate these as vehicle command message.
For `geo_point`, this cannot be provided if the map only has local coordinates.

tkhmy, 2023-01-31:
> For `target_velocity` and `target_acceleration`, It might be better to separate these as vehicle command message. For `geo_point`, this cannot be provided if the map only has local coordinates.

Sorry for late reply, I miss the message.
I agree that we should move the `target_velocity` and `target_acceleration` to vehicle_command.
We should just put in the current vehicle status data

For the `geo_point` I think we might able to convert from MGRS to the global longitude and latitude

tkhmy, 2023-02-01:
I created the document for vehicle status
https://autowarefoundation.github.io/autoware-documentation/pr-312/design/autoware-interfaces/ad-api/list/api/vehicle/

isamu-takagi, 2023-02-01:
> For the geo_point I think we might able to convert from MGRS to the global longitude and latitude

According to [the lanelet2_extension documentation](https://github.com/autowarefoundation/autoware_common/tree/main/tmp/lanelet2_extension#io), Autoware works in a local coordinate system independent of MGRS by setting `local_x` and `local_y`.

tkhmy, 2023-02-22:
I'm thinking we also can add the contraint for this vehicle status.
It will be a new api named, `/api/vehicle/moving_constraint`

which will contains following
```
float32 max_velocity
float32 min_acceleration
float32 max_jerk
float32 min_jerk
```

isamu-takagi, 2023-06-26:
I forgot the API to control doors. I have three suggestions.

1. array command
```yaml
message:
  - NONE
  - CLOSE
  - CLOSE
  - NONE
```

2. index command
```yaml
message:
  - { index: 1, command: CLOSE }
  - { index: 2, command: CLOSE }
```

3. role command
```yaml
message:
  - { role: GET_ON, command: CLOSE }
```

closed 2023-07-14 by isamu-takagi