autowarefoundation/autoware #3238: Docker build script fails with error `invalid not-modified ETag`

VRichardJP, 2023-01-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

From a clean install, the docker build script `./docker/build.sh` fails with the following error message:
```
$ ./docker/build.sh
+ docker buildx bake --load --progress=plain -f /home/vrichard/autoware.github/docker/autoware-universe/docker-bake.hcl --set '*.context=/home/vrichard/autoware.github/docker/../' --set '*.ssh=default' --set '*.platform=linux/amd64' --set '*.args.ROS_DISTRO=humble' --set '*.args.BASE_IMAGE=ubuntu:22.04' --set '*.args.SETUP_ARGS=--no-cuda-drivers' --set devel.tags=ghcr.io/autowarefoundation/autoware-universe:humble-latest-cuda --set prebuilt.tags=ghcr.io/autowarefoundation/autoware-universe:humble-latest-prebuilt-cuda
#1 [prebuilt internal] load build definition from Dockerfile
#1 transferring dockerfile: 32B done
#1 DONE 0.8s

#2 [devel internal] load .dockerignore
#2 transferring context: 2B done
#2 ...

#3 [devel internal] load build definition from Dockerfile
#3 transferring dockerfile: 32B done
#3 DONE 1.1s

#4 [prebuilt internal] load .dockerignore
#4 transferring context: 2B done
#4 DONE 1.2s

#2 [devel internal] load .dockerignore
#2 DONE 1.4s

#5 [prebuilt internal] load metadata for docker.io/library/ubuntu:22.04
#5 DONE 1.0s

#6 [devel internal] load build context
#6 DONE 0.0s

#7 [prebuilt internal] load build context
#7 DONE 0.0s

#8 [prebuilt] https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json
#8 ERROR: invalid not-modified ETag: "983d456a492957dff61261d2e42a15b3"

#9 [devel] https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json
#9 CANCELED

#10 [prebuilt devel  1/15] FROM docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763
#10 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763
#10 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763 0.5s done
#10 DONE 0.5s
------
 > [prebuilt] https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json:
------
ERROR: failed to solve: failed to load cache key: invalid not-modified ETag: "983d456a492957dff61261d2e42a15b3"
```

With the `--no-cuda` option, I get a similar error:
```
$ ./docker/build.sh --no-cuda
+ docker buildx bake --load --progress=plain -f /home/vrichard/autoware.github/docker/autoware-universe/docker-bake.hcl --set '*.context=/home/vrichard/autoware.github/docker/../' --set '*.ssh=default' --set '*.platform=linux/amd64' --set '*.args.ROS_DISTRO=humble' --set '*.args.BASE_IMAGE=ubuntu:22.04' --set '*.args.SETUP_ARGS=--no-nvidia' --set devel.tags=ghcr.io/autowarefoundation/autoware-universe:humble-latest --set prebuilt.tags=ghcr.io/autowarefoundation/autoware-universe:humble-latest-prebuilt
#1 [prebuilt internal] load build definition from Dockerfile
#1 transferring dockerfile: 32B done
#1 DONE 0.9s

#2 [prebuilt internal] load .dockerignore
#2 transferring context: 2B done
#2 ...

#3 [devel internal] load build definition from Dockerfile
#3 transferring dockerfile: 32B done
#3 DONE 1.1s

#2 [prebuilt internal] load .dockerignore
#2 DONE 1.3s

#4 [devel internal] load .dockerignore
#4 transferring context: 2B done
#4 DONE 1.5s

#5 [prebuilt internal] load metadata for docker.io/library/ubuntu:22.04
#5 DONE 1.0s

#6 [devel internal] load build context
#6 DONE 0.0s

#7 [prebuilt internal] load build context
#7 DONE 0.0s

#8 [devel devel  1/15] FROM docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763
#8 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763
#8 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763 0.4s done
#8 DONE 0.4s

#9 [prebuilt] https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json
#9 ERROR: invalid not-modified ETag: "1b109a6d7f3a4dd479154980405a2272"

#10 [devel] https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json
#10 CANCELED
------
 > [prebuilt] https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json:
------
ERROR: failed to solve: failed to load cache key: invalid not-modified ETag: "1b109a6d7f3a4dd479154980405a2272"
```

### Expected behavior

No error

### Actual behavior

Build fails

### Steps to reproduce

`$ ./docker/build.sh`

### Versions

- Ubuntu 22.04.1 
- latest main

kenji-miyake, 2023-01-31:
@ambroise-arm Do you have anything?

Considering that the link https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json is available, should we try using `wget` or something instead of the `ADD` command?
https://github.com/autowarefoundation/autoware/blob/d34355c8d5ce26405546c183405f6fa0f76e001e/docker/autoware-universe/Dockerfile#L46

kenji-miyake, 2023-01-31:
Although I haven't read it yet, is this probably related? https://github.com/moby/buildkit/issues/905

VRichardJP, 2023-01-31:
@kenji-miyake Yes, it might be the root cause. 

`wget`/`curl` could be a workaround, however it means the file would be download once, cached and never updated (unless the `--no-cache` is used). Apparently, `docker system prune` does the trick, I will test.

VRichardJP, 2023-01-31:
`docker system prune` only works once:

- first `./docker/build.sh` after `docker system prune`: OK
```
+ docker buildx bake --load --progress=plain -f /home/vrichard/autoware.sig/docker/autoware-universe/docker-bake.hcl --set '*.context=/home/vrichard/autoware.sig/docker/../' --set '*.ssh=default' --set '*.platform=linux/amd64' --set '*.args.ROS_DISTRO=humble' --set '*.args.BASE_IMAGE=ubuntu:22.04' --set '*.args.SETUP_ARGS=--no-cuda-drivers' --set devel.tags=ghcr.io/autowarefoundation/autoware-universe:humble-latest-cuda --set prebuilt.tags=ghcr.io/autowarefoundation/autoware-universe:humble-latest-prebuilt-cuda
#1 [prebuilt internal] load build definition from Dockerfile
#1 ...

#2 [devel internal] load build definition from Dockerfile
#2 transferring dockerfile: 2.46kB done
#2 DONE 1.2s

#1 [prebuilt internal] load build definition from Dockerfile
#1 transferring dockerfile: 2.46kB done
#1 DONE 1.9s

#3 [prebuilt internal] load .dockerignore
#3 transferring context: 2B done
#3 ...

#4 [devel internal] load .dockerignore
#4 transferring context: 2B done
#4 DONE 2.1s

#3 [prebuilt internal] load .dockerignore
#3 DONE 3.1s

#5 [devel internal] load metadata for docker.io/library/ubuntu:22.04
#5 DONE 2.8s

#6 [prebuilt internal] load build context
#6 DONE 0.0s

#7 [devel internal] load build context
#7 DONE 0.0s

#8 [prebuilt devel  1/15] FROM docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763
#8 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763
#8 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763 1.0s done
#8 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763 1.0s done
#8 DONE 1.0s

#9 [prebuilt] https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json
#9 DONE 1.3s

#6 [prebuilt internal] load build context
#6 ...

#10 [prebuilt] https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json
#10 DONE 1.7s

#6 [prebuilt internal] load build context
#6 transferring context: 51.62kB 0.0s done
#6 DONE 1.0s

#7 [devel internal] load build context
#7 transferring context: 51.62kB 0.0s done
#7 DONE 1.2s

#11 [prebuilt devel  4/15] COPY ansible/ /autoware/ansible/
#11 CACHED

#12 [prebuilt devel  6/15] RUN ls /autoware
#12 CACHED

#13 [prebuilt devel 12/15] ADD https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json /etc/glvnd/egl_vendor.d/10_nvidia.json
#13 CACHED

#14 [prebuilt devel  5/15] WORKDIR /autoware
#14 CACHED
[... 82 lines cut ...]

#6 [devel internal] load build context
#6 DONE 0.0s

#7 [prebuilt internal] load build context
#7 DONE 0.0s

#8 [devel] https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json
#8 ERROR: invalid not-modified ETag: "1b109a6d7f3a4dd479154980405a2272"

#9 [devel] https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json
#9 ERROR: invalid not-modified ETag: "983d456a492957dff61261d2e42a15b3"

#10 [prebuilt devel  1/15] FROM docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763
#10 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763
#10 resolve docker.io/library/ubuntu:22.04@sha256:9dc05cf19a5745c33b9327dba850480dae80310972dea9b05052162e7c7f2763 0.6s done
#10 DONE 0.6s
------
 > [devel] https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json:
------
------
 > [devel] https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json:
------
ERROR: failed to solve: failed to load cache key: invalid not-modified ETag: "1b109a6d7f3a4dd479154980405a2272"
```

VRichardJP, 2023-01-31:
Seems to be fix with:
```diff
diff --git a/docker/autoware-universe/Dockerfile b/docker/autoware-universe/Dockerfile
index 6b75784d..94ccba7b 100644
--- a/docker/autoware-universe/Dockerfile
+++ b/docker/autoware-universe/Dockerfile
@@ -43,10 +43,8 @@ RUN rm -rf \
   /etc/apt/sources.list.d/nvidia-docker.list
 
 ## Register Vulkan GPU vendors
-ADD "https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json" /etc/vulkan/icd.d/nvidia_icd.json
-RUN chmod 644 /etc/vulkan/icd.d/nvidia_icd.json
-ADD "https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json" /etc/glvnd/egl_vendor.d/10_nvidia.json
-RUN chmod 644 /etc/glvnd/egl_vendor.d/10_nvidia.json
+RUN curl https://gitlab.com/nvidia/container-images/vulkan/raw/dc389b0445c788901fda1d85be96fd1cb9410164/nvidia_icd.json -o /etc/vulkan/icd.d/nvidia_icd.json && chmod 644 /etc/vulkan/icd.d/nvidia_icd.json
+RUN curl https://gitlab.com/nvidia/container-images/opengl/raw/5191cf205d3e4bb1150091f9464499b076104354/glvnd/runtime/10_nvidia.json -o /etc/glvnd/egl_vendor.d/10_nvidia.json && chmod 644 /etc/glvnd/egl_vendor.d/10_nvidia.json
 
 ## Register OpenCL GPU vendors
 RUN mkdir -p /etc/OpenCL/vendors \
```

ambroise-arm, 2023-01-31:
It is probably even better with `RUN curl` than with `ADD` because as is pointed out in https://github.com/moby/buildkit/issues/905#issuecomment-874610691 this will cache the file, which is good because we are targeting a specific commit, so the file is expected to stay the same.

kenji-miyake, 2023-02-03:
@VRichardJP Can this be closed now?

closed 2023-02-03 by VRichardJP