autowarefoundation/autoware #3357: PluginlibFactory: The plugin for class 'autoware_rviz_debug/DecisionMakerPanel' failed to load.

lymcs, 2023-03-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

PluginlibFactory: The plugin for class 'autoware_rviz_debug/DecisionMakerPanel' failed to load.  Error: According to the loaded plugin descriptions the class autoware_rviz_debug/DecisionMakerPanel with base class type rviz::Panel does not exist. Declared types are  jsk_rviz_plugin/CancelAction jsk_rviz_plugin/EmptyServiceCallInterfaceAction jsk_rviz_plugin/ObjectFitOperatorAction jsk_rviz_plugin/PublishTopic jsk_rviz_plugin/RecordAction jsk_rviz_plugin/RobotCommandInterfaceAction jsk_rviz_plugin/SelectPointCloudPublishAction jsk_rviz_plugin/TabletControllerPanel jsk_rviz_plugin/YesNoButton rviz_plugin_tutorials/Teleop

### Expected behavior

PluginlibFactory: The plugin for class 'autoware_rviz_debug/DecisionMakerPanel' failed to load.  Error: According to the loaded plugin descriptions the class autoware_rviz_debug/DecisionMakerPanel with base class type rviz::Panel does not exist. Declared types are  jsk_rviz_plugin/CancelAction jsk_rviz_plugin/EmptyServiceCallInterfaceAction jsk_rviz_plugin/ObjectFitOperatorAction jsk_rviz_plugin/PublishTopic jsk_rviz_plugin/RecordAction jsk_rviz_plugin/RobotCommandInterfaceAction jsk_rviz_plugin/SelectPointCloudPublishAction jsk_rviz_plugin/TabletControllerPanel jsk_rviz_plugin/YesNoButton rviz_plugin_tutorials/Teleop

### Actual behavior

PluginlibFactory: The plugin for class 'autoware_rviz_debug/DecisionMakerPanel' failed to load.  Error: According to the loaded plugin descriptions the class autoware_rviz_debug/DecisionMakerPanel with base class type rviz::Panel does not exist. Declared types are  jsk_rviz_plugin/CancelAction jsk_rviz_plugin/EmptyServiceCallInterfaceAction jsk_rviz_plugin/ObjectFitOperatorAction jsk_rviz_plugin/PublishTopic jsk_rviz_plugin/RecordAction jsk_rviz_plugin/RobotCommandInterfaceAction jsk_rviz_plugin/SelectPointCloudPublishAction jsk_rviz_plugin/TabletControllerPanel jsk_rviz_plugin/YesNoButton rviz_plugin_tutorials/Teleop

### Steps to reproduce

PluginlibFactory: The plugin for class 'autoware_rviz_debug/DecisionMakerPanel' failed to load.  Error: According to the loaded plugin descriptions the class autoware_rviz_debug/DecisionMakerPanel with base class type rviz::Panel does not exist. Declared types are  jsk_rviz_plugin/CancelAction jsk_rviz_plugin/EmptyServiceCallInterfaceAction jsk_rviz_plugin/ObjectFitOperatorAction jsk_rviz_plugin/PublishTopic jsk_rviz_plugin/RecordAction jsk_rviz_plugin/RobotCommandInterfaceAction jsk_rviz_plugin/SelectPointCloudPublishAction jsk_rviz_plugin/TabletControllerPanel jsk_rviz_plugin/YesNoButton rviz_plugin_tutorials/Teleop

### Versions

PluginlibFactory: The plugin for class 'autoware_rviz_debug/DecisionMakerPanel' failed to load.  Error: According to the loaded plugin descriptions the class autoware_rviz_debug/DecisionMakerPanel with base class type rviz::Panel does not exist. Declared types are  jsk_rviz_plugin/CancelAction jsk_rviz_plugin/EmptyServiceCallInterfaceAction jsk_rviz_plugin/ObjectFitOperatorAction jsk_rviz_plugin/PublishTopic jsk_rviz_plugin/RecordAction jsk_rviz_plugin/RobotCommandInterfaceAction jsk_rviz_plugin/SelectPointCloudPublishAction jsk_rviz_plugin/TabletControllerPanel jsk_rviz_plugin/YesNoButton rviz_plugin_tutorials/Teleop

### Possible causes

PluginlibFactory: The plugin for class 'autoware_rviz_debug/DecisionMakerPanel' failed to load.  Error: According to the loaded plugin descriptions the class autoware_rviz_debug/DecisionMakerPanel with base class type rviz::Panel does not exist. Declared types are  jsk_rviz_plugin/CancelAction jsk_rviz_plugin/EmptyServiceCallInterfaceAction jsk_rviz_plugin/ObjectFitOperatorAction jsk_rviz_plugin/PublishTopic jsk_rviz_plugin/RecordAction jsk_rviz_plugin/RobotCommandInterfaceAction jsk_rviz_plugin/SelectPointCloudPublishAction jsk_rviz_plugin/TabletControllerPanel jsk_rviz_plugin/YesNoButton rviz_plugin_tutorials/Teleop

### Additional context

PluginlibFactory: The plugin for class 'autoware_rviz_debug/DecisionMakerPanel' failed to load.  Error: According to the loaded plugin descriptions the class autoware_rviz_debug/DecisionMakerPanel with base class type rviz::Panel does not exist. Declared types are  jsk_rviz_plugin/CancelAction jsk_rviz_plugin/EmptyServiceCallInterfaceAction jsk_rviz_plugin/ObjectFitOperatorAction jsk_rviz_plugin/PublishTopic jsk_rviz_plugin/RecordAction jsk_rviz_plugin/RobotCommandInterfaceAction jsk_rviz_plugin/SelectPointCloudPublishAction jsk_rviz_plugin/TabletControllerPanel jsk_rviz_plugin/YesNoButton rviz_plugin_tutorials/Teleop

lymcs, 2023-03-21:
/home/mcs/图片/2022-10-20 15-30-40屏幕截图.png

tl230419, 2023-08-07:
Just source deve/setup.bash in your package directory. 
Reference: https://answers.ros.org/question/215487/could-not-load-panel-in-rviz-pluginlibfactory-the-plugin-for-class/

xmfcx, 2023-11-16:
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/#1-launch-autoware

As it was stated above, please source the Autoware workspace before running RViz2:
`source ~/autoware/install/setup.bash` (default location)

If the issue still persists, please reopen this issue.

closed 2023-11-16 by xmfcx