autowarefoundation/autoware #3360: Release Autoware packages into the ROS buildfarm

esteve, 2023-03-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

In addition to generating our own Debian packages nightly (see https://github.com/autowarefoundation/autoware/issues/3222), submitting our packages to the ROS buildfarm will benefit our users because they'll be able to download Autoware from the standard ROS distribution.

### Purpose

Provide users with Debian packages they can install from the ROS servers

### Possible approaches

Follow the steps in https://docs.ros.org/en/humble/How-To-Guides/Releasing/First-Time-Release.html

### Definition of done

A user can install Autoware from the ROS servers

mitsudome-r, 2023-03-23:
Thanks for creating the issue.
Although this would help a lot of ROS users, I would like to think carefully about the timing and the target package.
 
I propose to do this after we do a clean up of messages (replacing autoware_auto_msgs with autoware_msgs). This is because change in interface could be quite a disruptive change and we would like to have more organized interface before making it ROS "official" release.
Also, I think we should only release into the ROS buildfarm for the packages in Core considering the fact that we defined Core as AWF maintained package and Universe to be community maintained. Universe packages can be registered to ROS buildfarm by the owner of the package at their decision. As Core is currently empty, we should start with porting Universe packages into Core first before doing this task.

esteve, 2023-03-23:
> Although this would help a lot of ROS users, I would like to think carefully about the timing and the target package.

Absolutely! This issue is just meant to start the discussion and find out what steps we'd need if we ever want to release Autoware into the ROS buildfarm, but I don't think we're ready now to do it.

> I propose to do this after we do a clean up of messages (replacing autoware_auto_msgs with autoware_msgs). This is because change in interface could be quite a disruptive change and we would like to have more organized interface before making it ROS "official" release.

Sounds good. I found https://github.com/autowarefoundation/autoware_msgs/issues/2 and I can work on it, if nobody else is working.

> Also, I think we should only release into the ROS buildfarm for the packages in Core considering the fact that we defined Core as AWF maintained package and Universe to be community maintained. [...]

A priori I don't see the problem with releasing Autoware.Universe as well, but I understand that Autoware.Core is meant to be of higher quality, so it might make sense releasing it first.

However, the way package releases work with the ROS buildfarm is that [bloom](https://docs.ros.org/en/humble/How-To-Guides/Releasing/First-Time-Release.html#bloom-release) releases all the packages in a given repository (that's why it's required that all packages have the same version in their `package.xml`). So if we want to let community maintainers to release their packages on their own, they'll have to keep the packages in separate repositories and not all in Autoware.Universe. If anyone runs `bloom` on the Autoware.Universe repository, they'll release all the packages there.

xmfcx, 2024-01-18:
Related discussion: [Prefix packages with autoware_ (4097)](https://github.com/orgs/autowarefoundation/discussions/4097)

mitsudome-r, 2026-04-17:
autoware_core is now released from ROS Buildfarm so I'm closing the issue.