autowarefoundation/autoware #3406: Add routing API for goal modification

isamu-takagi, 2023-04-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Depending on traffic conditions, it may not be possible to stop at the goal position. To solve this, add the following features to the routing API.
- Add goal modification option. This specifies whether the goal should be changed in order to pull over or avoid obstacles.
- Add route change API. This can manually change the goal while driving. It keeps the nearby route.

These features are associated with MRM (https://github.com/autowarefoundation/autoware.universe/issues/3221).

### Purpose

Support goal modification in the API.

### Possible approaches

Add `change_route` API.
Add goal modification option to the `set_route` and `change_route` API.

### Definition of done

- [ ] Add goal modification option to the message
- [ ] Add `change_route` API

mitsudome-r, 2023-04-10:
Were there any reasons for providing different interfaces for "modify goal" and "Route Change API"?

isamu-takagi, 2023-04-10:
>Were there any reasons for providing different interfaces for "modify goal" and "Route Change API"?

@mitsu
Mission planner manages normal route and emergency route. The "modify goal" is the internal interface of planning component for obstacle avoidance, etc, and it modifies goal of normal and emergency routes. The "route change API" and "emergency goal" switches route mode. When "emergency goal" is called, mission planner will be emergency mode and ignore "route change API".

closed 2023-06-26 by isamu-takagi