autowarefoundation/autoware #3427: Add option to lanelet2_map_loader for local coordinates

isamu-takagi, 2023-04-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

I'm developing [API that provides the global vehicle position (latitude and longitude)](https://github.com/autowarefoundation/autoware/issues/3232). For that, I tried to convert from map position to global position, but the [autoware osm parser](https://github.com/autowarefoundation/autoware_common/tree/main/tmp/lanelet2_extension#autoware-osm-parser) in the `lanelet2_extension` package overwrites position if `local_x` or `local_y` is specified in the Lanelet2 map. This works regardless of the projection type settings in the map loader, the transformation may not match the map projection type. So add an explicit option to the map loader to use local coordinates.

### Purpose

To support reverse projection of coordinates.

### Possible approaches

Change the X options as follows.
- Add a flag to enable or disable overwriting for local coordinates.
- Add a projection type that does nothing for local coordinates (optional).

The reason for the separate flags is for the use case of specifying only some nodes in local coordinates for experimentation.
- If the flag is enabled, the type for reverse projection will be treated as local.
- The combination of pojection type local and disable overwriting is an error.

### Definition of done

T.B.D.

tkhmy, 2023-04-21:
I removed the local parser for lanelet_extension
https://github.com/autowarefoundation/autoware_common/pull/174

the local parser will be in map_loader
https://github.com/autowarefoundation/autoware.universe/pull/3200

xmfcx, 2023-11-16:
@isamu-takagi -san,

The task: Add dummy projector to lanelet2_extension
Is this task important, will it be implemented?

isamu-takagi, 2023-11-17:
> The task: Add dummy projector to lanelet2_extension

This task is not important. Currently, [if a local type is used, it is overwritten with local coordinates after being projected with MGRS](https://github.com/autowarefoundation/autoware.universe/blob/b255111b7b11df532a11817134c33ca36c92b498/map/map_loader/src/lanelet2_map_loader/lanelet2_map_loader_node.cpp#L102-L115), so there is a bit of processing waste.

Also, there may be cases where only some positions are overwritten with local coordinates. In this case, it is better to be able to specify a sub-projection type when using local projections.

isamu-takagi, 2023-12-14:
>Also, there may be cases where only some positions are overwritten with local coordinates. In this case, it is better to be able to specify a sub-projection type when using local projections.

Changed to use dummy projector. I will consider adding the above cases if necessary.
https://github.com/autowarefoundation/autoware.universe/pull/5866

closed 2023-12-20 by isamu-takagi

xmfcx, 2023-12-21:
Thank you @isamu-takagi -san!