autowarefoundation/autoware #3466: Replace binary images with lighter SVG counterparts

xmfcx, 2023-04-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

I've sorted the files in the `autoware/src` directory by the file sizes.

Used the following command to list them by the size:

```bash
find . -type f -printf '%s %p\n' | grep -v '/\.git/' | sort -nr | awk '{ if($1>1024) print "| "$2" | "$1/1024/1024"MB |"; else print "| "$2" | "$1"B |"}' | head -n 30
```

| File Path | Size |
| -- | -- |
| ./universe/autoware.universe/planning/planning_test_utils/test_map/pointcloud_map.pcd | 26.8229MB |
| ./vehicle/sample_vehicle_launch/sample_vehicle_description/mesh/lexus.dae | 14.3274MB |
| ./universe/autoware.universe/perception/detected_object_validation/image/obstacle_pointcloud_based_validator/debug_image.gif | 8.59634MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/avoidance/avoidance_design.fig1.drawio.svg | 6.75412MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/avoidance/how_to_decide_path_shape_one_object.drawio.svg | 5.06166MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/avoidance/obstacle_to_road_shoulder_distance.drawio.svg | 5.04273MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/avoidance/intersection_problem.drawio.svg | 5.03856MB |
| ./universe/autoware.universe/perception/probabilistic_occupancy_grid_map/image/pointcloud_based_occupancy_grid_map_sample_image.gif | 3.84837MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/shift_length_parameters.drawio.svg | 3.37762MB |
| ./universe/external/muSSP/input_MOT_seq07_followme.txt | 2.18657MB |
| ./universe/autoware.universe/planning/behavior_velocity_planner/docs/run_out/collision_detection_for_shape.svg | 2.15132MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/lane_change/lane_change-enable_collision_check_at_prepare_phase.png | 2.1327MB |
| ./universe/autoware.universe/perception/traffic_light_visualization/images/roi-visualization.png | 2.01213MB |
| ./universe/autoware.universe/planning/planning_debug_tools/image/closest-velocity-checker.png | 1.99427MB |
| ./universe/autoware.universe/perception/occupancy_grid_map_outlier_filter/image/occupancy_grid_map_outlier_filter.drawio.svg | 1.98519MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/lane_change/lane_change-collision_check.png | 1.87883MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/avoidance/avoidance_module-shift_point_and_its_constraints.drawio.png | 1.75513MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/avoidance/how_to_decide_path_shape_multi_object_both_direction.drawio.svg | 1.68741MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/avoidance/how_to_decide_path_shape_multi_object_one_direction.drawio.svg | 1.6855MB |
| ./universe/autoware.universe/perception/traffic_light_map_based_detector/docs/traffic_light_map_based_detector_result.svg | 1.52102MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/limitations/limitation-chattering_shifts.png | 1.35751MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/lane_change/lane_change-debug.png | 1.14853MB |
| ./universe/external/ndt_omp/data/251371071.pcd | 1.06512MB |
| ./universe/external/ndt_omp/data/251370668.pcd | 1.05438MB |
| ./universe/external/muSSP/muSSP/cmake-build-debug/muSSP | 0.989799MB |
| ./universe/autoware.universe/planning/behavior_velocity_planner/docs/run_out/create_dynamic_obstacle.svg | 0.98032MB |
| ./universe/autoware.universe/planning/behavior_velocity_planner/docs/run_out/collision_points.svg | 0.962627MB |
| ./universe/autoware.universe/system/bluetooth_monitor/docs/block_diagram.drawio.svg | 0.866027MB |
| ./universe/external/muSSP/img/mcf_efficiency.png | 0.792622MB |
| ./universe/autoware.universe/planning/behavior_path_planner/image/avoidance/avoidance-debug-marker.png | 0.784833MB |

### Purpose

We can replace the png files by their svg counterparts to reduce the overall file sizes. And help them make more git friendly.

Also the large svg files contain binary images, they should also be replaced by vector image counterparts.

Otherwise it will keep inflating the git history unnecessarily.

### Possible approaches

Trace them with tools like Adobe Illustrator and replace them with the light vector images.

### Definition of done

At least these top 20 items are reduced to vector images. (Not all items in the list are images but this is a start)

xmfcx, 2023-04-27:
@kenji-miyake I think we should add a rule in documentation to not allow binary image contributions.

(I don't know if it's practical to add a CI check for this.)

Maybe we can use this https://github.com/ppremk/lfs-warning to prevent them from being contributed.

xmfcx, 2023-04-27:
For example,
Following file:

https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/image/avoidance/avoidance_design.fig1.drawio.svg

Has no reason to be this large.

But it has many binary images inside which can be replaced with vector images.

VRichardJP, 2023-04-28:
I don't know a thing about graphic software, but from a quick google it seems Inkscape has quick auto-vectorization tool:

https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/image/lane_change/lane_change-enable_collision_check_at_prepare_phase.png

- Path > Trace Bitmap...
- Tab: Multicolor
- Detection Mode: Color
- Remove Background

Took me 30 seconds (sorry for dark mode users):
[image]

Size before: 7MB
Size after: 650KB

I don't know if it is a very clean SVG, but it looks quick and effective.

kenji-miyake, 2023-04-28:
> https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/image/avoidance/avoidance_design.fig1.drawio.svg
> Has no reason to be this large.

I'm not sure, but it might have some meta data for re-editing in `draw.io`.

kenji-miyake, 2023-04-28:
> @kenji-miyake I think we should add a rule in documentation to not allow binary image contributions.

@xmfcx I remember you said we can consider utilizing `git-lfs` for images.
Did you find any problem with `git-lfs` and change your mind?

> (I don't know if it's practical to add a CI check for this.)

If we really disallow image files, I think checking on CI is good.

xmfcx, 2023-04-28:
@VRichardJP thanks, I didn't know Inkscape could do it too.

I could get following in Illustrator following https://helpx.adobe.com/illustrator/using/image-trace.html

[image]

> I'm not sure, but it might have some meta data for re-editing in draw.io

@kenji-miyake it's likely. Because when I opened the file with Illustrator and just saved it back without doing anything, it got reduced to 25KB.

[image]

We can still utilize git-lfs, but just for light vector files, it might not be necessary.

kenji-miyake, 2023-05-01:
@xmfcx By the way, how critical is this issue? Do you have any specific problems caused by the repository size?

xmfcx, 2023-05-01:
> By the way, how critical is this issue? Do you have any specific problems caused by the repository size?

I'd say it has normal priority. Keeping repo small in general is a good practice. And earlier we do this, better it is. Because these binary files keep making the .git folder larger and larger with no return.

I've realized this while investigating why the autoware folder was so large. Then realized these images and it'd be a good thing to do if anyone would like to spare some time.

It'd be the best if people who contributed these files put here the lighter versions instead, since we might lose some quality otherwise.

xmfcx, 2023-06-08:
> @xmfcx I remember you said we can consider utilizing `git-lfs` for images.
> Did you find any problem with git-lfs and change your mind?

We had a discussion with the Autoware Planning & Control team in the [WG meeting](https://github.com/orgs/autowarefoundation/discussions/3558).

@kenji-miyake 

The common agreement was that the draw.io metadata information (the binary blob) in the files are important for the team. We should keep them.

And to allow this, I had 2 solutions:
- Add `git-lfs` support for the image and binary files. (my preference)
- Replace the `.svg` files with smaller versions. But collect and host the original `.drawio.svg` files in some google drive, have links to them in the documentation. (too complicated and hard to maintain)

@esteve do you think adding `git-lfs` tracking to the generic image files in the middle of repository timeline might have any unknown side effects?

I would like to go with this option but we also need to update the documentation, maybe ansible scripts, maybe for certain CI actions (like mk-docs generation) to have this support.

esteve, 2023-06-08:
> @esteve do you think adding `git-lfs` tracking to the generic image files in the middle of repository timeline might have any unknown side effects?

I don't know of any side effects, but I agree with using `git-lfs` for storing the `.svg` files.

mehmetdogru, 2024-02-15:
@xmfcx agrees on closing the issue.

closed 2024-02-15 by mehmetdogru