autowarefoundation/autoware #3468: Clean unused dependencies to improve build time

VRichardJP, 2023-04-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Building Autoware takes time, but it is not just because there are a lot of files to compile. Building time also increases when packages have more dependencies than they should, because:
- it may create bottleneck in the build pipeline: some packages are waiting for some others while they could be built in parallel.
- it inflates the numbers of include/link arguments passed down to the compiler, linker and recursively to downstream packages. This is most impactful on "deep" packages. See for example: https://github.com/ament/ament_cmake/issues/442#issue-1670725910

Every now and then, I think it is important to clean the `packages.xml` files and remove unused dependencies.

### Purpose

Improve build time.

### Possible approaches

This is a very tedious and annoying job. I wrote a simple script that try to list seemingly unused packages:
```bash
#!/bin/bash
pkgs=$(find src -name "package.xml")
for pkg in $pkgs; do
  echo "--- Checking $pkg ---"
  # get all packages marked <depend>
  deps=$(grep -oP '<depend>\K[^<]+' $pkg)
  dir=$(dirname $pkg)
  for dep in $deps; do
    # look by default for C/C++ includes
    regex="#include [<\"]$dep/.*[>\"]"

    # a few exceptions
    [[ "$dep" =~ libpcl-.*$ ]] && regex="#include [<\"]pcl/.*[>\"]"
    [[ "$dep" =~ libboost-.*$ ]] && regex="#include [<\"]boost/.*[>\"]"
    [[ "$dep" =~ libqt5-.*$ ]] && regex="#include [<\"]Q.*[>\"]"
    [[ "$dep" == "qtbase5-dev" ]] && regex="#include [<\"]Q.*[>\"]"
    [[ "$dep" == "libopencv-dev" ]] && regex="#include [<\"]opencv2/.*[>\"]"
    [[ "$dep" == "pugixml-dev" ]] && regex="#include [<\"]pugixml.hpp[>\"]"
    [[ "$dep" == "yaml_cpp_vendor" ]] && regex="#include [<\"]yaml-cpp/yaml.h[>\"]"
    [[ "$dep" == "nlohmann-json-dev" ]] && regex="#include [<\"]nlohmann/json.hpp[>\"]"
    [[ "$dep" == "range-v3" ]] && regex="#include [<\"]range/v3/.*[>\"]"
    [[ "$dep" == "libcpprest-dev" ]] && regex="#include [<\"]cpprest/.*[>\"]"
    [[ "$dep" == "eigen" ]] && regex="#include [<\"]Eigen/.*[>\"]"
    [[ "$dep" == "libnl-3-dev" ]] && regex="#include [<\"]netlink/.*[>\"]"
    # add more...
    
    # rosidl_default_generators is for the rosidl_generate_interfaces() cmake macro
    [[ "$dep" == "rosidl_default_generators" ]] \
      && [[ $(grep -wr rosidl_generate_interfaces "$dir" --include CMakeLists.txt | wc -l) -gt 0 ]] \
      && continue
    
    # msg packages may be re-used in other messages
    if [[ "$dep" =~ .*_msgs$ ]] || [[ "$dep" == "builtin_interfaces" ]]; then
      grep -wIPrq "$dep" "$dir" --include \*.msg --include \*.srv --include \*.idl
      [[ $? -eq 0 ]] && continue
    fi

    # finally check if the dependency is used anywhere in the source files
    grep -wIPrq "$regex" "$dir" \
      --include \*.c --include \*.cc --include \*.cpp --include \*.h --include \*.hpp 
    [[ $? -eq 0 ]] && continue

    echo "$dep seems not to be used"
  done
  echo ""
done

``` 
[... 851 lines cut ...]

--- Checking src/universe/external/rtklib_ros_bridge/rtklib_bridge/package.xml ---

--- Checking src/universe/external/rtklib_ros_bridge/rtklib_msgs/package.xml ---
builtin_interfaces seems not to be used

--- Checking src/universe/external/ndt_omp/package.xml ---

--- Checking src/universe/external/pointcloud_to_laserscan/package.xml ---
libpcl-all-dev seems not to be used
tf2 seems not to be used

--- Checking src/universe/external/llh_converter/package.xml ---

--- Checking src/vehicle/sample_vehicle_launch/sample_vehicle_launch/package.xml ---

--- Checking src/vehicle/sample_vehicle_launch/sample_vehicle_description/package.xml ---

--- Checking src/vehicle/external/pacmod_interface/pacmod_interface/package.xml ---

```

### Definition of done

No more unused dependencies listed in `package.xml` files.

TakaHoribe, 2023-04-28:
So great…!

simon-t4, 2023-04-28:
This is a great initiative! Thanks!

In the exceptions section of the script, is it easy to add a list of verified dependencies per package? So that if people check a particular "unused dependency" and verify it is in fact used, that it can be excepted from the search in the future?

VRichardJP, 2023-04-28:
@simon-t4 This is just a PoC script. By default, I just look for any `#include <dep_name/.*>` in the C/C++ source files. But as you can see, there are quite a few listed "exceptions" (and I may have missed a few). Most exceptions are system libraries (e.g. `libboost-dev` -> `#include <boost/.*>`), but there are also "special" dependencies that won't be used in the source code (e.g. `rosidl_default_generators` brings cmake macros). It is not difficult to add more cases/exceptions to avoid false positives.

VRichardJP, 2023-04-29:
I have updated the script above a bit and quickly fixed all reported dependencies. Then checked release build time before/after on my machine:

```bash
$ colcon build --symlink-install --event-handlers=console_cohesion+ --cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=1 -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=False -DCMAKE_C_FLAGS='-fdiagnostics-color' -DCMAKE_CXX_FLAGS='-fdiagnostics-color' -GNinja
```

Before:
```
real	42m25.554s
user	261m18.837s
sys	25m59.806s
```

After:
```
real	42m17.404s
user	244m15.675s
sys	16m57.653s
```

The `real` time is not impacted much (not exactly sure why), but you can see `user` and `sys` times are reduced by a decent amount: 261m12 seconds of "cpu time" (user+sys) after compared to 287m17s before. That is +10% faster, just by removing unused stuff...!

There are a lot of `package.xml` to clean, any help to create PRs is welcomed :)

Detailed diff below:
```diff
=== src/core/autoware.core (git) ===
=== src/core/autoware_adapi_msgs (git) ===
=== src/core/autoware_common (git) ===
diff --git a/autoware_utils/package.xml b/autoware_utils/package.xml
index 4355dc8..e6b7fdf 100644
--- a/autoware_utils/package.xml
+++ b/autoware_utils/package.xml
@@ -12,7 +12,6 @@
 
   <build_depend>autoware_cmake</build_depend>
 
-  <depend>builtin_interfaces</depend>
   <depend>rclcpp</depend>
 
   <test_depend>ament_cmake_ros</test_depend>
diff --git a/tmp/lanelet2_extension/package.xml b/tmp/lanelet2_extension/package.xml
index 2a117b2..7dab023 100644
--- a/tmp/lanelet2_extension/package.xml
+++ b/tmp/lanelet2_extension/package.xml
@@ -12,18 +12,14 @@
   <build_depend>autoware_cmake</build_depend>
 
   <depend>autoware_auto_mapping_msgs</depend>
-  <depend>autoware_auto_planning_msgs</depend>
   <depend>autoware_planning_msgs</depend>
   <depend>autoware_utils</depend>
-  <depend>geographiclib</depend>
   <depend>geometry_msgs</depend>
   <depend>lanelet2_core</depend>
   <depend>lanelet2_io</depend>
-  <depend>lanelet2_maps</depend>
   <depend>lanelet2_projection</depend>
   <depend>lanelet2_routing</depend>
   <depend>lanelet2_traffic_rules</depend>
-  <depend>lanelet2_validation</depend>
   <depend>pugixml-dev</depend>
   <depend>range-v3</depend>
   <depend>rclcpp</depend>
=== src/core/autoware_msgs (git) ===
diff --git a/autoware_common_msgs/package.xml b/autoware_common_msgs/package.xml
index 834e667..54e3fe3 100644
--- a/autoware_common_msgs/package.xml
+++ b/autoware_common_msgs/package.xml
@@ -8,8 +8,7 @@
[... 2471 lines cut ...]
   "srv/UpdateAccelBrakeMap.srv"
-  DEPENDENCIES std_msgs geometry_msgs tier4_control_msgs
+  DEPENDENCIES std_msgs #geometry_msgs tier4_control_msgs
 )
 
 if(BUILD_TESTING)
diff --git a/tier4_vehicle_msgs/package.xml b/tier4_vehicle_msgs/package.xml
index 40b1d2c..a943c22 100644
--- a/tier4_vehicle_msgs/package.xml
+++ b/tier4_vehicle_msgs/package.xml
@@ -11,9 +11,9 @@
 
   <build_depend>rosidl_default_generators</build_depend>
 
-  <depend>geometry_msgs</depend>
+  <!-- <depend>geometry_msgs</depend> -->
   <depend>std_msgs</depend>
-  <depend>tier4_control_msgs</depend>
+  <!-- <depend>tier4_control_msgs</depend> -->
 
   <exec_depend>rosidl_default_runtime</exec_depend>
=== src/vehicle/external/pacmod_interface (git) ===
=== src/vehicle/sample_vehicle_launch (git) ===

```

mitsudome-r, 2023-05-08:
@VRichardJP Good work!
I haven't looked through all the items in the list, but at least for lanelet2_extension, it should be using `geographiclib`. Likewise, there could be some non-ROS package (what you call "system dependencies" in your script) that shouldn't be removed from the dependencies.

VRichardJP, 2023-05-09:
@mitsudome-r 
Thank you for pointing out `geographiclib`, indeed the library is not unused.

Btw, ["system dependencies"](https://index.ros.org/deps/) is how ROS call them ^^

mdabir1203, 2023-08-20:
I would like to take on this and how may I start. A little example would be helpful

a-maumau, 2024-07-18:
Recent result of `unused_deps.sh` run on `main` branch

763 unused packages.

<details>
<summary>results (non showing package is filtered)</summary>

```Text
--- Checking src/universe/autoware.universe/evaluator/autoware_control_evaluator/package.xml ---
autoware_evaluator_utils seems not to be used
autoware_motion_utils seems not to be used
autoware_route_handler seems not to be used
autoware_universe_utils seems not to be used
pluginlib seems not to be used

--- Checking src/universe/autoware.universe/evaluator/autoware_planning_evaluator/package.xml ---
autoware_evaluator_utils seems not to be used
autoware_motion_utils seems not to be used
autoware_route_handler seems not to be used
autoware_universe_utils seems not to be used
eigen should be rather be marked as <buildtool_depend>, not <depend>
pluginlib seems not to be used
tf2 seems not to be used

--- Checking src/universe/autoware.universe/evaluator/autoware_evaluator_utils/package.xml ---
rclcpp_components seems not to be used

--- Checking src/universe/autoware.universe/evaluator/diagnostic_converter/package.xml ---
pluginlib seems not to be used

--- Checking src/universe/autoware.universe/evaluator/kinematic_evaluator/package.xml ---
autoware_planning_msgs seems not to be used
autoware_universe_utils seems not to be used
eigen should be rather be marked as <buildtool_depend>, not <depend>
tf2 seems not to be used

--- Checking src/universe/autoware.universe/evaluator/perception_online_evaluator/package.xml ---
autoware_lanelet2_extension seems not to be used
autoware_motion_utils seems not to be used
autoware_universe_utils seems not to be used
eigen should be rather be marked as <buildtool_depend>, not <depend>
pluginlib seems not to be used
tf2 seems not to be used

--- Checking src/universe/autoware.universe/evaluator/localization_evaluator/package.xml ---
autoware_planning_msgs seems not to be used
autoware_universe_utils seems not to be used
eigen should be rather be marked as <buildtool_depend>, not <depend>
tf2 seems not to be used

--- Checking src/universe/autoware.universe/vehicle/autoware_external_cmd_converter/package.xml ---
geometry_msgs seems not to be used
std_msgs seems not to be used
tier4_system_msgs seems not to be used

--- Checking src/universe/autoware.universe/vehicle/autoware_accel_brake_map_calibrator/package.xml ---
autoware_motion_utils seems not to be used
autoware_universe_utils seems not to be used
std_srvs seems not to be used
tf2_ros seems not to be used

--- Checking src/universe/autoware.universe/vehicle/autoware_steer_offset_estimator/package.xml ---
autoware_universe_utils seems not to be used
std_msgs seems not to be used

--- Checking src/universe/autoware.universe/vehicle/autoware_vehicle_info_utils/package.xml ---
autoware_universe_utils seems not to be used

--- Checking src/universe/autoware.universe/vehicle/autoware_raw_vehicle_cmd_converter/package.xml ---

[... 1482 lines cut ...]

--- Checking src/core/external/autoware_auto_msgs/autoware_auto_msgs/package.xml ---

--- Checking src/core/external/autoware_auto_msgs/autoware_auto_perception_msgs/package.xml ---

--- Checking src/core/external/autoware_auto_msgs/autoware_auto_control_msgs/package.xml ---
std_msgs seems not to be used

--- Checking src/core/autoware_utils/autoware_utils/package.xml ---
builtin_interfaces seems not to be used

--- Checking src/core/autoware_common/tmp/lanelet2_extension_python/package.xml ---
geometry_msgs seems not to be used
lanelet2_projection seems not to be used
lanelet2_routing seems not to be used
lanelet2_traffic_rules seems not to be used
lanelet2_validation seems not to be used
libboost-python-dev should be rather be marked as either <build_depend> or <test_depend>, not <depend>

--- Checking src/core/autoware_common/tmp/lanelet2_extension/package.xml ---
lanelet2_maps seems not to be used
lanelet2_validation seems not to be used

```
</details>

a-maumau, 2024-07-22:
I updated some of the recent including style of autoware sources.

Basically, the rule is like `autoware_motion_utils` -> `autoware/motion_utils`, so it may possible to automatically generate the regular expressions.

... well, some of then seems still with `autoware_*`.

<details>
<summary>unused_deps.sh</summary>

```
#!/bin/bash
# This script find recursively all package.xml in the src/ subdirectory
# Then check if all package marked <depend> are effectly used somewhere in the package.
# Packages marked <depend> normally either provide shared libraries or custom messages

# Packages that should always be <buildtool_depend>, not <depend>
KNOWN_BUILDTOOLS="rosidl_default_generators;autoware_cmake;eigen3_cmake_module"

pkgs=$(find src -name "package.xml")
for pkg in $pkgs; do
  echo "--- Checking $pkg ---"
  # Get all packages marked <depend>. For example:
  #   <depend>dep_name</depend>
  deps=$(grep -oP '^\s*<depend>\K[^<]+' $pkg)
  dir=$(dirname $pkg)

  if [[ ! -f "$dir/CMakeLists.txt" ]]; then
    echo "Skipping package with no CMakeLists.txt (is python package?)"
    echo ""
    continue
  fi

  for dep in $deps; do
    # filter out buildtools (should not use <depend>)
    if [[ $(echo "$KNOWN_BUILDTOOLS" | grep "$dep" | wc -l) -ne 0 ]]; then
      echo "$dep should be rather be marked as <buildtool_depend>, not <depend>"
      continue
    fi

    # filter out ament stuff (should not use <depend>)
    if [[ "$dep" =~ ament_.*$ ]]; then
      echo "$dep should be rather be marked as either <build_depend> or <test_depend>, not <depend>"
      continue
    fi

    # filter out python stuff (should not use <depend>)
    if [[ "$dep" =~ python-.*$ ]]; then
      echo "$dep should be rather be marked as either <build_depend> or <test_depend>, not <depend>"
      continue
    fi

    # By convention, dependency headers must be prefixed with the package name. For Example:
    # tier4_autoware_utils/geometry/geometry.hpp
    header_regex="$dep/.*"

    # System dependencies don't follow this rule. They all have custom names.
    [[ "$dep" =~ libpcl-.*$ ]] && header_regex="pcl/.*"
    [[ "$dep" =~ libboost-.*$ ]] && header_regex="boost/.*"
    [[ "$dep" =~ libqt5-.*$ ]] && header_regex="Q.*"
    [[ "$dep" == "qtbase5-dev" ]] && header_regex="Q.*"
    [[ "$dep" == "libopencv-dev" ]] && header_regex="opencv2/.*"
    [[ "$dep" == "pugixml-dev" ]] && header_regex="pugixml.hpp"
    [[ "$dep" == "yaml_cpp_vendor" ]] && header_regex="yaml-cpp/yaml.h"
    [[ "$dep" == "nlohmann-json-dev" ]] && header_regex="nlohmann/json.hpp"
    [[ "$dep" == "range-v3" ]] && header_regex="range/v3/.*"
    [[ "$dep" == "libcpprest-dev" ]] && header_regex="cpprest/.*"
    [[ "$dep" == "eigen" ]] && header_regex="Eigen/.*"
    [[ "$dep" == "libnl-3-dev" ]] && header_regex="netlink/.*"
    [[ "$dep" == "libpcap" ]] && header_regex="pcap.h"
    [[ "$dep" == "cgal" ]] && header_regex="CGAL/.*"
[... 31 lines cut ...]
    include_regex="^#include [<\"]$header_regex[>\"]"

    # Dependencies defining custom messages may also be re-used in other messages or services
    # By convention, these packages must be named "*_msgs", excepted for "builtin_interfaces"
    if [[ "$dep" =~ .*_msgs$ ]] || [[ "$dep" == "builtin_interfaces" ]]; then
      grep -wIPrq "$dep" "$dir" --include \*.msg --include \*.srv --include \*.idl
      [[ $? -eq 0 ]] && continue  # found!
    fi

    # Check if the dependency is included anywhere in the source files
    grep -wIPrq "$include_regex" "$dir" \
      --include \*.c --include \*.cc --include \*.cpp --include \*.h --include \*.hpp 
    [[ $? -eq 0 ]] && continue  # found!

    echo "$dep seems not to be used"

    # /!\ WARNING - DANGEROUS /!\
    # Uncomment to delete unused dependency!
    # sed -i "/<depend>$dep<\/depend>/d" $pkg
  done
  echo ""
done

```
</details>

VRichardJP, 2024-07-22:
Would it make sense to have this script somewhere like https://github.com/autowarefoundation/autoware_tools?

YamatoAndo, 2024-07-30:
> Would it make sense to have this script somewhere like https://github.com/autowarefoundation/autoware_tools?

I think it is a good idea!

@youtalk @mitsudome-r What do you think?

youtalk, 2024-07-30:
@YamatoAndo Please make the PR. I would like to review it.

YamatoAndo, 2024-07-31:
@youtalk Sure! I will collaborate with @a-maumau  to make the PR.

a-maumau, 2024-08-02:
@youtalk @VRichardJP 
I made a PR for the script in autoware_tools: https://github.com/autowarefoundation/autoware_tools/pull/89

sorry, I don't have the authority to assign the assigners so if possible, please assign youself.