autowarefoundation/autoware #3487: Create API to request manual operation

isamu-takagi, 2023-05-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

This issue is from https://github.com/tier4/tier4_ad_api_adaptor/pull/95#issuecomment-1543635600. Create an API that requests switching to manual driving when the vehicle becomes unable to drive.

Related to:
- [Fail-safe API](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-interfaces/ad-api/list/api/fail_safe/)
- [Operation mode API](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-interfaces/ad-api/list/api/operation_mode/)

### Purpose

Resolve vehicle stuck.

### Possible approaches

Create an API that requests switching of operation mode.

### Definition of done

- [ ] message design
- [ ] implementation

isamu-takagi, 2023-05-11:
@mitsudome-r @yukkysaito cc: @tkhmy, @yabuta for UI, @mkuri for MRM
I think that the message type should have at least a bool variable. Any other information you need (e.g. text message)?

yukkysaito, 2023-05-11:
Can we use an existing RTC Interface?

isamu-takagi, 2023-05-11:
> Can we use an existing RTC(Request to corporate) Interface?

@yukkysaito RTC is a function that the  operator supports Autoware decisions during autonomous driving, so it is assumed that autonomous driving can continue. On the other hand, since this is a function that requests mode switching when autonomous driving cannot continue, we concluded that it would be better to use a different API in [this PR](https://github.com/tier4/tier4_ad_api_adaptor/pull/95) discussion.

yabuta, 2023-06-04:
@isamu-takagi In RTI, there are situations such as OR for a stuck vehicle or an emergency vehicle passing by.
In such cases, I think it would be better to notify the operator of what caused the request for OR, so I think it would be better to have a section that describes what happened.

isamu-takagi, 2023-06-05:
@yabuta Thank you for comment. Add message field. Also, I feel that  a stuck vehicle or an emergency vehicle passing have different priorities when supporting multiple vehicles remotely, so I changed from bool to uint to support that.

xmfcx, 2025-02-17:
@isamu-takagi -san, are there any updates on this task?

isamu-takagi, 2025-02-19:
@xmfcx Since it may include notifications such as RTC as well as MRM, I will change it to a more generic message.

isamu-takagi, 2026-08-20:
@xmfcx Thank you for checking the implementation status. Contrary to expectations, it turns out that a fail-safe state is sufficient in most cases. Therefore, I would like to remove it from the documentation while keeping its status as not released.