autowarefoundation/autoware #3540: Lanelet misplaced in Autoware

Tester2009, 2023-06-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I have draw a simple lanelet using Tier IV Vector Map Builder. But when I load it to Autoware (using 713dfa1b8d118557226473f5f596877ce137c9e1), the lanelet is wrongly located.

### Expected behavior

Based on how I draw it on Tier IV Vector Map Builder, it should look like this;

[image]

### Actual behavior

Below is how it is located.

[image]

Not only that, the lanelet we draw is only for a short distance, but the shown looks like have longer distance.

### Steps to reproduce

1. Assume Autoware has been configured.
2. Run autoware command: `ros2 launch autoware_launch autoware.launch.xml vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit map_path:=$HOME/ros2_ws/map-one`
3. Wait until Autoware full load (PCD and lanelet).
4. Issue reproduced.

### Versions

- OS: Ubuntu 22.04.2 LTS
- ROS2: Humble
- Autoware: 713dfa1b8d118557226473f5f596877ce137c9e1

### Possible causes

We assume that newer version of Autoware have the ability to know if the pointcloud is global shifted or not. Because we never had this issue on previous version of Autoware.

mitsudome-r, 2023-06-01:
Try modifying the projection type in lanelet2_map_loader to `local` from the default `MGRS`
https://github.com/autowarefoundation/autoware_launch/blob/c6312c7dcaa8fdcf17b54c11953b2c23861cd577/autoware_launch/config/map/lanelet2_map_loader.param.yaml#L3

We have made change to the projector in https://github.com/autowarefoundation/autoware.universe/pull/3492, and user must choose the local projection type explicitly if the map is using local coordinates from pointcloud.

Tester2009, 2023-06-01:
Thank you @mitsudome-r for your response. I have made the changes from the default `MGRS` to `local` as you suggest.

I will now close this issue.