autowarefoundation/autoware #3549: Compilation of `vesc_ackermann` failed

RonaldEnsing, 2023-06-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am following the steps 'without ADE' from here: https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/building.html and getting a compilation error in the `vesc_ackermann` package about no matching function calls.

I have also seen the issue here: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/1497
However, since the project is migrated to github, I am filing it here.

### Expected behavior

I expect all autoware modules to compile successfully.

### Actual behavior

Executing `colcon build` results in:

```
AutowareAuto/src/external/vesc/vesc_ackermann/src/ackermann_to_vesc.cpp:51:71: error: no matching function for call to ‘vesc_ackermann::AckermannToVesc::declare_parameter<double>(const char [19])’
   51 |   speed_to_erpm_gain_ = declare_parameter<double>("speed_to_erpm_gain");
      |                                                                       ^
In file included from /opt/ros/foxy/include/rclcpp/node.hpp:1224,
                 from /opt/ros/foxy/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:22,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
........
make: *** [Makefile:141: all] Error 2
---
Failed   <<< vesc_ackermann [8.96s, exited with code 2]
Aborted  <<< tvm_utility [5.61s]                                                                                                                                                                            
Aborted  <<< lonely_world_prediction [6.36s]                                                                                                                                                                 
Aborted  <<< autoware_auto_create_pkg [2.83s]                                                                                                                                                               
Aborted  <<< autoware_auto_algorithm [12.5s]
Aborted  <<< autoware_auto_common [14.3s]                                                                                                                                                            
Aborted  <<< razor_imu_ros2 [6.33s]                                                                                                                                                                      
Aborted  <<< grid_map_octomap [14.2s]                                                                                                                                                                      
Aborted  <<< lgsvl_laser_adapter_nodes [19.4s]                                                                                                                                                                  
Aborted  <<< grid_map_cv [26.2s]                                                                                                                                  
Aborted  <<< autoware_auto_vehicle_msgs [32.1s]                                                                                   
Aborted  <<< laser_filters [34.5s]                                     

Summary: 26 packages finished [44.1s]
  1 package failed: vesc_ackermann
  11 packages aborted: autoware_auto_algorithm autoware_auto_common autoware_auto_create_pkg autoware_auto_vehicle_msgs grid_map_cv grid_map_octomap laser_filters lgsvl_laser_adapter_nodes lonely_world_prediction razor_imu_ros2 tvm_utility
  3 packages had stderr output: neural_networks tvm_utility vesc_ackermann
  123 packages not processed
```

### Steps to reproduce

1. Install ROS2 Foxy
2. `git clone https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto.git`
3. `cd AutowareAuto`
4. `vcs import < autoware.auto.$ROS_DISTRO.repos`
5. `git lfs pull --exclude="" --include="*"`
6. `echo "export COLCON_DEFAULTS_FILE=/usr/local/etc/colcon-defaults.yaml" >> ~/.bashrc`
7. `colcon build`

### Versions

- Ubuntu 20.04
- ROS Foxy
- Autoware 1.0.0 or `master` branch (commit `15d2e91a2b3c8cd61b20b7e6e79da32919c2cbcd`)

jyh3690, 2023-06-06:
i have this issue also ,are you resolve it?

maxime-clem, 2023-06-06:
I opened a merged request that should fix the issue: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/merge_requests/1388
I cannot test it myself so please let me know whether it solves the issue or not.

RonaldEnsing, 2023-06-06:
Thanks!

Yes, this solves the compilation of `vesc_ackermann`.

Compilation of another package (`mpc_controller`) fails, but I guess I should file a separate issue for that.

guchangjun, 2023-07-01:
I solve the problem by changing the function as:
speed_to_erpm_gain_ = declare_parameter<double>("speed_to_erpm_gain",speed_to_erpm_gain_);

RonaldEnsing, 2023-09-01:
The main problem is that the instructions on gitlab seem to be outdated and that the instructions on github should be taken.

closed 2023-09-01 by RonaldEnsing