autowarefoundation/autoware #3686: 1 package failed: trtexec_vendor

wo-ai-xue-xi, 2023-07-23:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Summary: 209 packages finished [3min 4s]
  1 package failed: trtexec_vendor
  7 packages aborted: autoware_auto_perception_rviz_plugin bag_time_manager_rviz_plugin behavior_velocity_walkway_module lidar_centerpoint morai_msgs tensorrt_yolox tier4_hmi_msgs
  2 packages had stderr output: tensorrt_yolox trtexec_vendor
  78 packages not processed

### Expected behavior

`trtexec_vendor` compile

### Actual behavior

`trtexec_vendor` 
```
autoware/build/trtexec_vendor/_deps/tensorrt-src/samples/common/sampleEngines.cpp:319:60: error: ‘OnnxParserFlag’ has not been declared
  319 |             auto parserflags = 1U << static_cast<uint32_t>(OnnxParserFlag::kNATIVE_INSTANCENORM);

autoware/build/trtexec_vendor/_deps/tensorrt-src/samples/common/sampleEngines.cpp:320:32: error: ‘class nvonnxparser::IParser’ has no member named ‘setFlags’
  320 |             parser.onnxParser->setFlags(parserflags);
```

### Steps to reproduce

1.colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

### Versions

rosdistro=humble
rmw_implementation=rmw_cyclonedds_cpp
base_image=ubuntu:22.04
cuda_base_image=nvidia/cuda:12.0.0-devel-ubuntu22.04
prebuilt_base_image=ubuntu:22.04
cuda_version=12.0
cudnn_version=8.8.0.121-1+cuda12.0
tensorrt_version=8.6.0.12-1+cuda12.0

### Possible causes

The cuda version is too high

maxime-clem, 2023-07-24:
@angry-crab do you have some idea about this issue ?

cho-jang-hyun, 2023-07-25:
I have this issue also ,are you resolve it?

angry-crab, 2023-07-25:
I guess you are using a different version of TensorRT. Maybe you could try to upgrade it.

wo-ai-xue-xi, 2023-07-25:
> 我猜你使用的是不同版本的 TensorRT。也许你可以尝试升级它。

I installed the tensorrt environment according to this configuration https://github.com/autowarefoundation/autoware/pull/3084#discussion_r1160428947

> ❯ find-cuda-versions --cuda 12.0 --os ubuntu2204
> [cudnn x86_64]
> 8.8.0.121-1+cuda12.0
> 8.8.1.3-1+cuda12.0
> 
> [cudnn sbsa]
> 8.8.0.121-1+cuda12.0
> 8.8.1.3-1+cuda12.0
> 
> [nvinfer x86_64]
> 8.6.0.12-1+cuda12.0
> 
> [nvinfer sbsa]
> 
> [tensorrt x86_64]
> 8.6.0.12-1+cuda12.0
> 
> [tensorrt sbsa]

wo-ai-xue-xi, 2023-07-25:
> I have this issue also ,did you resolve it?

I switched the cuda environment ，everything is fine now

> 1. cuda_version=11.8
> 2. cudnn_version=8.7.0.84-1+cuda11.8
> 3. tensorrt_version=8.6.1.6-1+cuda11.8

If you want to do like me, you can execute the following command
```
sudo apt-get -y install cuda-11-8
sudo apt-get install libcudnn8=8.7.0.84-1+cuda11.8 libcudnn8-dev=8.7.0.84-1+cuda11.8
sudo apt-mark hold libcudnn8 libcudnn8-dev

sudo apt-get install libnvinfer-headers-dev=8.6.1.6-1+cuda11.8 libnvinfer-headers-plugin-dev=8.6.1.6-1+cuda11.8 libnvinfer8=8.6.1.6-1+cuda11.8 libnvonnxparsers8=8.6.1.6-1+cuda11.8 libnvparsers8=8.6.1.6-1+cuda11.8 libnvinfer-plugin8=8.6.1.6-1+cuda11.8 libnvinfer-dev=8.6.1.6-1+cuda11.8 libnvonnxparsers-dev=8.6.1.6-1+cuda11.8 libnvparsers-dev=8.6.1.6-1+cuda11.8 libnvinfer-plugin-dev=8.6.1.6-1+cuda11.8

sudo apt-mark hold libnvinfer-headers-dev libnvinfer-headers-plugin-dev libnvinfer8 libnvonnxparsers8 libnvparsers8 libnvinfer-plugin8 libnvinfer-dev libnvonnxparsers-dev libnvparsers-dev libnvinfer-plugin-dev
```

closed 2023-07-27 by J-xinyu