autowarefoundation/autoware #3943: Upgrade to CUDA 12

xmfcx, 2023-10-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

In the Autoware, currently we are using a CUDA version from Ubuntu 20.04 with [a workaround](https://github.com/autowarefoundation/autoware/tree/b92953c3ded0bcfb0d7f56b686011888547ed43c/ansible/roles/cuda#manual-installation).

Related PR:
- https://github.com/autowarefoundation/autoware/pull/3084

We haven't updated the version for a while because NVIDIA didn't release the TensorRT for ARM until [2023 September](https://github.com/autowarefoundation/autoware/pull/3084#issuecomment-1706847144).

Now that it is released, we can test our CUDA dependent packages and finally update.

### Purpose

To use recommended, latest NVIDIA packages.

### Possible approaches

We can run each CUDA dependent package on both x86_64 and ARM64 and if they work, we can update the ansible script here to install the latest CUDA, TensorRT and cuDNN:
- https://github.com/autowarefoundation/autoware/tree/b92953c3ded0bcfb0d7f56b686011888547ed43c/ansible/roles/cuda
- https://github.com/autowarefoundation/autoware/tree/b92953c3ded0bcfb0d7f56b686011888547ed43c/ansible/roles/tensorrt (contains both cudnn and tensorrt)

### Definition of done

- CUDA dependent packages are listed
- CUDA dependent packages are tested
- Ansible scripts are updated

xmfcx, 2023-10-30:
I followed:
- https://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html#network-repo-installation-for-ubuntu

```bash
wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2204/x86_64/cuda-keyring_1.1-1_all.deb
sudo dpkg -i cuda-keyring_1.1-1_all.deb

sudo apt update

sudo apt install cuda tensorrt-dev libcudnn8-dev
```
to install latest CUDA on my machine and I was able to compile Autoware without errors.

I will test some of the packages (starting from the ones required for AWSIM testing) and report my findings.

xmfcx, 2023-10-30:
[image]

It works!

centerpoint and traffic_light_classifier works.

xmfcx, 2023-10-30:
@mitsudome-r do you think we should also test this in ARM or can we start upgrading the Ansible scripts right away?

xmfcx, 2023-10-31:
@oguzkaganozt have you tested Autoware on AVA developer kit?
If so, could you test it on it too?
It's not urgent, we can use CI to test if it builds and just update the ansible scripts too.

esteve, 2023-10-31:
@xmfcx I'm targeting CUDA 12 for the Debian packages, the process for installing the previous version was a bit clunky and hopefully by the time Autoware works with CUDA, the old steps will be obsolete.

oguzkaganozt, 2023-11-02:
> @oguzkaganozt have you tested Autoware on AVA developer kit? If so, could you test it on it too? It's not urgent, we can use CI to test if it builds and just update the ansible scripts too.

I have tested Autoware on AVA and it works. I think this update should also work on it too. But give me some time to validate.

esteve, 2023-11-02:
@xmfcx I've updated the Ansible scripts for CUDA in #3956 , I don't have a GPU powerful enough to test it, though.

esteve, 2023-11-06:
I've updated the ticket with the list of packages that depend on CUDA, I'd appreciate if someone can test #3956 with a GPU on x86_64 and ARM, thanks.

oguzkaganozt, 2023-11-09:
I have tested on both arm64(AADP-AVA) and amd64 platforms using rosbag replay simulator with GeForce RTX 3070. It is working fine, I didn't notice any difference between older versions of CUDA and TensorRT.

[image]
[image]
[image]

closed 2023-11-13 by esteve

xmfcx, 2024-07-02:
CUDA 12.2 downgrade discussion:
- https://github.com/autowarefoundation/autoware/pull/3956#issuecomment-1845866638

PR #3956 (2023-11-13): build: update to CUDA 12.3
## Description

This PR updates the ansible scripts CUDA to 12.3.

Fixes #3943 

## Related links

## Tests performed

## Notes for reviewers

## Interface changes

## Effects on system behavior

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [ ] The PR follows the [pull request g [... cut]