autowarefoundation/autoware #3944: colcon build error for autoware_auto_tf2

felixf4xu, 2023-10-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

`colcon build --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-select autoware_auto_tf2`

There are 2 compiling errors:
```
-- stderr: autoware_auto_tf2                                                                  
In file included from /home/ccccc/github/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/test/test_tf2_autoware_auto_msgs.cpp:17:
/home/ccccc/github/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/include/autoware_auto_tf2/tf2_autoware_auto_msgs.hpp:60:13: error: redefinition of ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Point32_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’
   60 | inline void doTransform(
      |             ^~~~~~~~~~~
In file included from /home/ccccc/github/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/include/autoware_auto_tf2/tf2_autoware_auto_msgs.hpp:36,
                 from /home/ccccc/github/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/test/test_tf2_autoware_auto_msgs.cpp:17:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp:260:6: note: ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Point32_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’ previously declared here
  260 | void doTransform(
      |      ^~~~~~~~~~~
```
and
```
In file included from /home/ccccc/github/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/test/test_tf2_autoware_auto_msgs.cpp:17:
/home/ccccc/github/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/include/autoware_auto_tf2/tf2_autoware_auto_msgs.hpp:81:13: error: redefinition of ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Polygon_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’
   81 | inline void doTransform(
      |             ^~~~~~~~~~~
In file included from /home/ccccc/github/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/include/autoware_auto_tf2/tf2_autoware_auto_msgs.hpp:36,
                 from /home/ccccc/github/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/test/test_tf2_autoware_auto_msgs.cpp:17:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp:460:6: note: ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Polygon_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’ previously declared here
  460 | void doTransform(
      |      ^~~~~~~~~~~

```

### Expected behavior

no error

### Actual behavior

error

### Steps to reproduce

`colcon build --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-select autoware_auto_tf2`

### Versions

Ubuntu 22
humble
autoware code from git

### Possible causes

duplicated function, one quick fix is to delete those 2 functions.

SakodaShintaro, 2023-10-30:
Thanks for the report.

It is probably the same problem as below.

https://github.com/orgs/autowarefoundation/discussions/3877

I hope this information helps you.

felixf4xu, 2023-10-31:
yes, updating to latest git version fixed my problem.

closed 2023-10-31 by felixf4xu