autowarefoundation/autoware #3984: Modernize the RViz2 overlay interface

xmfcx, 2023-11-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The current RViz2 interface in Autoware is outdated, visually unappealing, and hard to read, particularly the cyan text on a white background. This not only makes it difficult for users to extract crucial information but also creates a negative first impression for new users. Improvements are needed, starting with the 2D overlay.

The existing 2D overlay shows:
- Vehicle speed
- Steering angle
- Current speed limit
- Left/right signal indicators

Proposed additions:
- Current gear display
- Detected traffic light state/color

A redesign of the 2D RViz2 overlay is proposed to address these issues.

Current Autoware RViz2:
[image]

Proposed Design:
[image]

Figma link:
- https://www.figma.com/file/xjrQXpUX0LbRNa14zKiy98/autoware-hud?type=design&node-id=17%3A263&mode=design&t=nGswozNNo5Sg345d-1

### Purpose

The purpose of this redesign is to modernize the RViz2 interface, making it more user-friendly, informative, and aesthetically pleasing. The updated interface will provide clearer and more comprehensive data at a glance, improving the overall user experience and utility of Autoware. This redesign aims to bring Autoware's interface in line with industry standards and user expectations, based on examples from other autonomous vehicle companies.

### Possible approaches

- **Interface Overhaul:** Redesigning the entire layout and visual elements of the RViz2 overlay.
- **Color Scheme Revision:** Updating the color palette to improve readability, especially for critical information.
- **Data Display Expansion:** Incorporating additional relevant data such as gear status and traffic light recognition.
- **User Feedback Incorporation:** Gathering and implementing feedback from current users to guide design changes.
- **Investigating Similar Designs:** Analyzing interface designs from companies like Tesla, AutoX, Nvidia, Pony.ai, Toyota Teammate, and Mobileye to incorporate best practices.

### Definition of done

The project will be considered complete when:
- [ ] The new design is fully implemented and functional in the RViz2 overlay.
- [ ] All proposed data points (vehicle speed, steering angle, speed limit, signals, gear status, traffic light state) are clearly and accurately displayed.
- [ ] The interface has been tested for readability and ease of use under various conditions.
- [ ] User feedback on the new design is overwhelmingly positive.
- [ ] The updated interface aligns with or surpasses current industry standards in terms of design and functionality.

xmfcx, 2023-11-13:
To achieve this, we can use the following package as a base in the Universe repository:
- https://github.com/autowarefoundation/autoware.universe/tree/main/common/tier4_vehicle_rviz_plugin

And create a new package called:
- `autoware_hud_plugin` ([HUD](https://en.wikipedia.org/wiki/Head-up_display))

yukkysaito, 2023-11-15:
Very good!
However, Rviz has its limitations, so if we are aiming more beautiful visualization, we may want to consider Game Engine as well.

https://github.com/yukkysaito/godot_rviz2

mitsudome-r, 2023-11-15:
FYI: It seems like some developers were considering using color palettes for designing visualization markers. We might want to design the color palettes for overlays and other markers to have unified look.

xmfcx, 2023-11-15:
@yukkysaito I know about your project in Godot. And I agree that we should create a visualization tool in Unity or Godot.

But reality is that we don't have many developers who would work on getting these done, it'd be a big project.

Also every developer opens up rviz2 many times while developing Autoware. And even small improvements like these can make everyone's experiences much better.

I'm not aware of all the limitations of rviz2 just yet. I know that it uses [Ogre](https://www.ogre3d.org/) for rendering and [QtWidgets 5](https://doc.qt.io/qt-5/qtwidgets-index.html) for window management and overall gui. I think there is still some potential there.

@mitsudome-r thanks for letting me know about them. We plan to make it easy for people to define their own theme colors.

xmfcx, 2024-02-02:
Closing this issue with both PRs are merged, thanks @KhalilSelyan !

closed 2024-02-02 by xmfcx