autowarefoundation/autoware #4021: rosdep cannot locate rosdep definition for "pacmod3_msgs"

AmirInt, 2023-11-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When installing the dependent ROS packages for Autoware based on the [installation guide](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/), the command `rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO` errors out complaining that it cannot find the package **pacmod3_msgs**. Here is the full log:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
pacmod_interface: Cannot locate rosdep definition for [pacmod3_msgs]
```

### Expected behavior

rosdep must start downloading the ROS and system dependencies without errors.

### Actual behavior

But it doesn't!

### Steps to reproduce

Perform a clean and fresh installation of Autoware based on the mentioned installation guide.

### Versions

OS: Ubuntu 22.04
ROS: ROS2 Humble

### Possible causes

It can be an outdated `autoware.repos` file. Probably adding the package index for [pacmod3_msgs](https://github.com/astuff/pacmod3_msgs) to the repos file would fix the issue. I actually got it from the [pacmod_interface build_depends.repos file](https://github.com/tier4/pacmod_interface/blob/main/build_depends.repos).

### Additional context

I've checked the term about duplicate issues, but actually, there is a duplicate issue and it's [here](https://github.com/autowarefoundation/autoware/issues/3941). I wonder why the bug still exists.

xmfcx, 2023-11-28:
If you follow the [source installation guide](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/), on the list, you should see
- [Install pacmod](https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/pacmod#manual-installation)

item on the list if you don't follow the ansible route.

The steps there will help the `rosdep` find the message you need.

AmirInt, 2023-11-28:
I actually took the ansible route and it went well without any errors. After that rosdep failed.

xmfcx, 2023-11-28:
Interesting, at least in the Autoware dockerfile generation, it seems to be installing fine:
- https://github.com/autowarefoundation/autoware/actions/runs/7004746973/job/19053135870#step:4:1254

Could you manually run [these steps](https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/pacmod#manual-installation) and tell if it works?

closed 2023-11-28 by xmfcx