autowarefoundation/autoware #4058: Unify source files for sync-file action into one repository

isamu-takagi, 2023-12-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Autoware repositories shares setting files using sync-file, but there is many source repository and it is difficult to know which files should be synchronized from.

For example, [the sync-file.yaml in autoware.universe repository](https://github.com/autowarefoundation/autoware.universe/blob/main/.github/sync-files.yaml) is copying files from autoware, autoware_common, and autoware-documentation repositories.

### Purpose

By unifying source files for sync-file action into one repository, the settings for sync-file action will be easier to understand.

### Possible approaches

Create a new repository to store the sync source files and place sync source files under an arbitrary directory to separate it from the repository own settings.

### Definition of done

- [ ] Create a sync source repository.
- [ ] Move setting files.
- [ ] Update sync-file.yaml of each repository.

isamu-takagi, 2023-12-13:
@mitsudome-r @xmfcx Could you please create a new repository?
Repository name: `sync-file-templates`

mitsudome-r, 2023-12-13:
I have created the repository: https://github.com/autowarefoundation/sync-file-templates

xmfcx, 2024-12-21:
- [ ] Update sync-file.yaml of each [TIER IV](https://github.com/tier4) repository.
  - [ ] [nebula](https://github.com/tier4/nebula/blob/main/.github/sync-files.yaml)
    - https://github.com/tier4/nebula/pull/248
  - [ ] [autoware_individual_params](https://github.com/autowarefoundation/autoware_individual_params/blob/main/.github/sync-files.yaml)
  - [ ] [pacmod_interface](https://github.com/tier4/pacmod_interface/blob/main/.github/sync-files.yaml)
  - [ ] [sensor_component_description](https://github.com/tier4/sensor_component_description/blob/main/.github/sync-files.yaml)
  - [ ] [awsim_sensor_kit_launch](https://github.com/tier4/awsim_sensor_kit_launch/blob/main/.github/sync-files.yaml)
  - [ ] [tier4_autoware_msgs](https://github.com/tier4/tier4_autoware_msgs/blob/main/.github/sync-files.yaml)
  - [ ] [tier4_ad_api_adaptor](https://github.com/tier4/tier4_ad_api_adaptor/blob/main/.github/sync-files.yaml)

I will leave tier4 organization repos here. If someone with proper access is interested, you may track it under a new issue.

closed 2024-12-21 by xmfcx