autowarefoundation/autoware #4070: Packages should put their headers directory in a directory named after the package

esteve, 2023-12-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Following https://autowarefoundation.github.io/autoware-documentation/main/contributing/coding-guidelines/ros-nodes/directory-structure/#include packages should put their headers in a separate directory named like the packagea

See https://github.com/orgs/autowarefoundation/discussions/4066

### Expected behavior

All packages have their headers in a separate directory

### Actual behavior

The following packages have their headers in directories other than a directory named like the package:

 - [ ] autoware_auto_common
 - [ ] autoware_auto_geometry
 - [ ] autoware_auto_perception_rviz_plugin
 - [ ] detected_object_validation
 - [ ] detection_by_tracker
 - [ ] object_merger
 - [ ] planning_test_utils
 - [ ] probabilistic_occupancy_grid_map
 - [ ] radar_fusion_to_detected_object
 - [ ] system_monitor
 - [ ] tier4_planning_rviz_plugin
 - [ ] traffic_light_visualization

### Steps to reproduce

Check the directory hierarchy

mitsudome-r, 2024-01-22:
All tasks are finished, but we will close this after @esteve confirms that the build passes in https://github.com/autowarefoundation/autoware-deb-packages.

esteve, 2024-02-06:
A followup to this ticket, we will rename headers directories either to `autoware_PACKAGE_NAME` or `autoware/PACKAGE_NAME`, see https://github.com/orgs/autowarefoundation/discussions/4097#discussioncomment-8380000 for context