autowarefoundation/autoware #4079: Adoption of Changelog Enforcer GitHub Action in Autoware Repositories

xmfcx, 2023-12-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

To improve the management and tracking of changes across the Autoware repositories, I propose the adoption of the [Changelog Enforcer GitHub Action](https://github.com/marketplace/actions/changelog-enforcer).
This action will ensure that every pull request includes a changelog update, helping us maintain a clear and consistent record of changes.
We aim to start this with the `autoware.universe` repository and, if successful, expand to other repositories like `autoware_launch` and `autoware_msgs`.

### Purpose

The primary goals of implementing this action are:
1. **Ensuring Consistency:** Each PR should contribute to the changelog, making it easier to track changes over time.
2. **Improving Transparency:** Clear documentation of changes enhances transparency for both developers and users.
3. **Simplifying Release Process:** With quarterly releases planned using CalVer, a well-maintained changelog will streamline the release process.

### Possible approaches

1. **Gradual Implementation:** Start with `autoware.universe`, assess the effectiveness, and then proceed to other repositories.
2. **Automate Changelog Entries:** Utilize the Changelog Enforcer to automate the inclusion of changelog updates in PRs.
3. **Quarterly Sync with CalVer:** Align the changelog updates with our quarterly CalVer release cycle, ensuring that each release has a comprehensive changelog.
4. **Centralized Changelog for Main Autoware Folder:** Compile changes from the child repositories' changelogs into the main Autoware folder's changelog for each release.

### Definition of done

- [ ] Agreement on the adoption of the Changelog Enforcer GitHub Action.
- [ ] Successful implementation and testing in the `autoware.universe` repository.
- [ ] Development of a clear guideline for contributors on how to update the changelog.
- [ ] Establishment of a process for syncing the main Autoware folder's changelog with child repositories.
- [ ] Review and adjust the process after the first quarterly release cycle.

xmfcx, 2023-12-21:
There is also an automated way of doing it using [github-changelog-generator](https://github.com/github-changelog-generator/github-changelog-generator#github-changelog-generator-).

It generates [CHANGELOG.md files like this](https://github.com/github-changelog-generator/github-changelog-generator/blob/master/CHANGELOG.md).

There is also [Github Releases](https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases).
- [automatically-generated-release-notes](https://docs.github.com/en/repositories/releasing-projects-on-github/automatically-generated-release-notes#creating-automatically-generated-release-notes-for-a-new-release)

I will investigate these options more and also share my findings. Please don't hesitate to add your inputs.

xmfcx, 2023-12-22:
https://github.com/googleapis/release-please#release-please

> Release Please automates CHANGELOG generation, the creation of GitHub releases, and version bumps for your projects.

https://github.com/google-github-actions/release-please-action

TakaHoribe, 2023-12-22:
@xmfcx I strongly support this activity. Moreover, it should be linked with future releases of Autoware. 

One question: is it required to modify the changelog file in all PRs? Or, the changelog is updated automatically from the PR's description?

xmfcx, 2023-12-25:
@TakaHoribe thank you!

> is it required to modify the changelog file in all PRs? Or, the changelog is updated automatically from the PR's description?

For the https://github.com/marketplace/actions/changelog-enforcer mentioned in the first issue, you must manually edit the changelog.

But if we use https://github.com/googleapis/release-please#release-please then it is completely automated (from Commit Message.

It generates ChangeLogs like this: https://github.com/googleapis/gapic-generator/blob/master/CHANGELOG.md

[release-please](https://github.com/googleapis/release-please) seems very popular and also maintained by Google(?) for now it is the most promising option. And it requires the least additional effort from the contributors.

shmpwk, 2024-01-01:
@xmfcx Interesting investigation!
Actually, we TIER IV has already automated changelog [here](https://github.com/tier4/autoware.universe/releases).
They are created by [generate-changelog CI](https://github.com/autowarefoundation/autoware-github-actions/tree/main/generate-changelog) which wraps [git-cliff](https://github.com/orhun/git-cliff).
I'm curious if changelog-enforcer and other changelog generator are better than our current automated changelog.