autowarefoundation/autoware #4117: C++ code should be in the autoware namespace

esteve, 2024-01-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Current C++ code does not follow a unified convention for namespaces. According to the Google C++ Style Guide (which the ROS Style guide is based upon) code must be under a namespace that matches the name of the project https://google.github.io/styleguide/cppguide.html#Namespaces

### Purpose

- Avoid conflicts with other non-Autoware projects
- Highlight that our packages are part of `autoware`

### Possible approaches

The most straightforward approach is to just add all software to the `autoware` namespace, however, we might want to also add another subnamespace for the component the code is part of (i.e. `perception`, `localization`, etc.)

### Definition of done

All the code is under either the `autoware` namespace or a subnamespace of it

esteve, 2024-02-02:
Related discussion: https://github.com/orgs/autowarefoundation/discussions/4097#discussioncomment-8347308

xmfcx, 2024-02-02:
Over 2 options:
1) `autoware::package_name`
2) `autoware::component_name::package_name`

My vote would be on Option 1.
Reasons:
- Easier to type and reduced complexity
- Easier to move package over component folders (if we want to change its component category) (ease of refactoring)
- Is not necessary to have the additional namespace since we don't expect the packages within the Autoware to clash.

cc. @yukkysaito @mitsudome-r @isamu-takagi @TakaHoribe

esteve, 2024-02-02:
@xmfcx I agree, I prefer to keep hierarchies as flat as possible

esteve, 2024-03-27:
Closing, superseded by #4569

closed 2024-03-27 by esteve