autowarefoundation/autoware #4165: Delete several packages/implementations in perception directory

xmfcx, 2024-02-09:
### Discussed in https://github.com/orgs/autowarefoundation/discussions/4136

<div type='discussions-op-text'>

<sup>Originally posted by **kminoda** January 31, 2024</sup>
# Introduction

I would like to delete several packages/implementations in perception, listed below:

- heatmap_visualizer
- lidar_centerpoint/single_inference
- traffic_light_ssd_fine_detector
- tensorrt_yolo
- front_vehicle_velocity_estimator

# Justification

## heatmap_visualizer
https://github.com/autowarefoundation/autoware.universe/tree/main/perception/heatmap_visualizer

This is a script which is not intended to run on runtime. I have confirmed with the author of this script and agreed that we can delete this package. 

## lidar_centerpoint/single_inference
https://github.com/autowarefoundation/autoware.universe/blob/main/perception/lidar_centerpoint/launch/single_inference_lidar_centerpoint.launch.xml

Single inference script for CenterPoint has been added 2 years ago starting from this issue: https://github.com/orgs/autowarefoundation/discussions/2755. However, the script is not intended to run on runtime but rather at a debug time, which makes it difficult for us to apply CI on this implementation. If there is no one currently using this script, we would like to delete this launch file and some related files as well.

## traffic_light_ssd_fine_detector
https://github.com/autowarefoundation/autoware.universe/tree/main/perception/traffic_light_ssd_fine_detector

This package is primarily implemented for traffic light detection using SSD. However, with the recent development, current Autoware uses `traffic_light_fine_detector` as a default option, outperforming SSD detector in most cases. 
Please let us know if you are currently using this package. Also, let us know if it is difficult for you to make a transition to `traffic_light_fine_detector`. 

## tensorrt_yolo
https://github.com/autowarefoundation/autoware.universe/tree/main/perception/tensorrt_yolo

This package performs 2D detection on camera images to achieve camera-LiDAR fusion. This is based on YOLO, but recently TIER IV has developed YOLOX version (`tensorrt_yolox`) which outperforms this package in most of the cases. 

## front_vehicle_velocity_estimator
https://github.com/autowarefoundation/autoware.universe/tree/main/perception/front_vehicle_velocity_estimator

This package is currently not used in default Autoware implementation. I confirmed with the author of this script and agreed that we can delete this package.

# Process of deleting these files

We will have a survey period for 2 weeks from today until 16th of February. Please comment if you have any oppositions or opinions on either of the listed files.

After this period, we are going to delete all the files listed above as long as there is no disagreements or concerns raised in this Discussion. 

# Related links
- [Optimizing Feature Management in Autoware: A Proposal for Deprecation and Deletion Process](https://github.com/orgs/autowarefoundation/discussions/3906)</div>

kminoda, 2024-02-22:
It's almost done, the only ToDs left is the lidar_centerpoint/single_inference. Will do that when I have time (or also appreciated if anyone could do that instead :+1: )

xmfcx, 2024-07-02:
Closing since the packages in question are all removed.

closed 2024-07-02 by xmfcx