autowarefoundation/autoware #4170: Use PR Dependency Check Action across repositories

xmfcx, 2024-02-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Right now the with so many repositories, everyone has to manage the dependent PRs manually.

### Purpose

To put the PR dependencies in order in a systematic way.

### Possible approaches

Add https://github.com/marketplace/actions/pr-dependency-check to:
- https://github.com/autowarefoundation/autoware
- https://github.com/autowarefoundation/autoware.universe
- https://github.com/autowarefoundation/autoware_launch
- https://github.com/autowarefoundation/autoware_common
- https://github.com/autowarefoundation/autoware_msgs

To begin with.

### Definition of done

It's added to respective places and it works as expected.

xmfcx, 2024-02-12:
Example:
- https://github.com/gregsdennis/dependencies-action/pull/5

xmfcx, 2024-02-12:
Without this action, right now developers are resorting to custom methods to point out the dependencies:

[image]

Source:
- https://github.com/autowarefoundation/autoware_launch/pull/760

xmfcx, 2024-02-12:
Alternatives:
- https://github.com/marketplace/actions/pr-dependency-management

yukkysaito, 2024-02-12:
I can't wait for this one. :+1: