autowarefoundation/autoware #4569: Prefix packages with autoware_ prefix and keep code in autoware namespace

esteve, 2024-03-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

See https://github.com/orgs/autowarefoundation/discussions/4097

### Purpose

Keep Autoware code clearly separate from other projects by adding appropriate namespacing and prefixes

### Possible approaches

Add autoware namespace and autoware_ prefix for package names

### Definition of done

All packages are prefixed and the code is in the autoware namespace

xmfcx, 2024-05-02:
Following up from:
- https://github.com/orgs/autowarefoundation/discussions/4097#discussioncomment-8384169
   - `autoware_package_name/include/autoware/package_name`
- https://github.com/autowarefoundation/autoware.universe/pull/6817#issuecomment-2090617834

We should prefix the package folder names with `autoware_` too.

The package name in the `package.xml` file should match its parent folder.

I might have missed something too, open to discussion.

esteve, 2024-05-02:
@xmfcx I've been waiting for feedback from the TierIV engineers regarding renaming the folders. @mitsudome-r do you have any updates? Thanks.

xmfcx, 2024-05-02:
I would like to change it and see the response in this case. And solve the issues after they arise.

mitsudome-r, 2024-05-07:
I think we can move on with changing the package folder name as well.
However, we did have some concerns about the changing the package names & include folder structure for the packages in autoware_common repository and also the packages with tier4_ prefix.

https://github.com/autowarefoundation/autoware_common

technolojin, 2024-06-21:
@esteve @xmfcx @mitsudome-r 

Hello. This is Taekjin Lee, a representative for the package prefix task in the TIER IV perception team.
We will start to work from July. 

We think the working timing should be aligned to minimize impact between repositories, launchers, and related systems may depends on the package names.

TIER IV perception engineering team will take charge of packages under
- universe/perception
- universe/sensing
and related repos.

technolojin, 2024-06-25:
@esteve @xmfcx @mitsudome-r
We (TIER IV perception engineering team) decided to separate tasks of the namespace fix and the package name fix. 
The related PRs will refer this issue, so that the progress can be tracked.

technolojin, 2024-07-04:
Detail of https://github.com/autowarefoundation/autoware/issues/4569#issuecomment-2188129685

In /perception, the package name will be changed after all the packages are ready. 
The separation is to minimize mistakes on namespace changes from package renaming. If those two works are done in the same time, complexity of the work and the test procedure may increases. 

- the old package remains in your build/install folder
- namespace renaming cannot be tested when the old package is remained
- do clean build every time is inefficient. manual cleaning is needed every time, every working branch changes

When the relevant packages are ready (namespace fix), the package renaming will be done in a concentrated timing.

esteve, 2024-07-04:
@technolojin thanks for elaborating, but I have a question about this:

> The separation is to minimize mistakes on namespace changes from package renaming. If those two works are done in the same time, complexity of the work and the test procedure may increases.

Why is that? We've done that with the planning packages and many others and we haven't had a problem. What makes the perception packages different? I've moved the header files and renamed the packages for some packages in the perception subsystem in https://github.com/autowarefoundation/autoware.universe/pull/7809 https://github.com/autowarefoundation/autoware.universe/pull/7808 and https://github.com/autowarefoundation/autoware.universe/pull/7804 and it doesn't seem to have caused any issues.

technolojin, 2024-07-08:
@esteve
There are big differences between sensing/perception and other components. 

The sensing and some perception components are absorbing hardware differences. Therefore, the launcher of the systems contains names of packages and those are distributed over multiple repositories. 
Some packages you mentioned is after virtualization is done, so that those were fine. 
In TIER IV, we have launchers for each products, launchers for main ECU and embedded devices, launchers and scripts for CI/CD. I believe the situation is similar or same to other autoware users. 
As TIER IV perception team, we conducts tests on each PRs to ensure the system and CI/CD are working. 

This is reason why our team want to lead the work for the package prefix and want to concentrate the name changes. By separating the other required works (namespace, distinguishing shared headers, remove unused dependencies), which can be done per package with minimized changes on launchers, it is possible that the package name change can be done in a short time and shorten overall down time for all relevant system. We need cooperation of code owners to achieve the goal.

I cannot fix the schedule of the package prefix fixes since the namespace work is delayed than my expectation. I will announce here when the schedule is visible.

mitsudome-r, 2024-07-08:
@technolojin 
In order to move things forward, I would like to have the following questions answered:

1. Could you provide the names of the packages that you are especially concerned about?
My understanding is that at least, most launch files are referred by tier4_.*_launch files in autoware.universe repository and abstracted in the autoware_launch repository.

2. If we create the PR first for all the sensing/perception packages so that TIER IV would only have to review, would that make things easier for you? If it makes easier for you to align the merge timing, then we can wait for merging until all the PRs are approved.

technolojin, 2024-07-08:
1. These are repositories and packages linked each other, over the boundary of the autoware_universe. This list is as far as I know at this moment, but I guess there are more. 

@kminoda @YoshiRi Please let us know if I miss repos or packages.

* repositories
  - https://github.com/autowarefoundation/sample_sensor_kit_launch
  - https://github.com/tier4/aip_launcher
  - https://github.com/tier4/edge_auto_launch
  - https://github.com/tier4/edge_auto_jetson_launch
  - TIER IV internal - evaluator
  - TIER IV internal - CI/CD (webauto)

* packages
  - bytetrack
  - image_projection_based_fusion
  - euclidean_cluster
  - compare_map_segmentation
  - lidar_apollo_instance_segmentation
  - lidar_centerpoint
  - tensorrt_yolox
  - pointcloud_preprocessor
  - radar_tracks_msgs_converter
  - simple_object_merger
  - radar_tracks_noise_filter
  - traffic_light_classifier
  - traffic_light_map_based_detector
 
2. If there is ongoing PRs that related to those sensing/perception packages, it would me easier to merge with all the related repositories, in a short time.

esteve, 2024-07-09:
@technolojin as @TakaHoribe said in https://github.com/orgs/autowarefoundation/discussions/4097#discussioncomment-9234315, the Autoware project shouldn't be limited by one company's concerns, we should be free to work on this task regardless of any external requirement. Autoware shouldn't halt its development because of any internal TierIV repository.

> https://github.com/tier4/aip_launcher
> https://github.com/tier4/edge_auto_launch
> https://github.com/tier4/edge_auto_jetson_launch
> TIER IV internal - evaluator
> TIER IV internal - CI/CD (webauto)
> TIER IV internal - ML related(?)

These repositories are not part of the Autoware project, so they are not relevant to this discussion.

> If there is ongoing PRs that related to those sensing/perception packages, it would me easier to merge with all the related repositories, in a short time.

There are already PRs that address this task (see https://github.com/autowarefoundation/autoware.universe/pull/7809 https://github.com/autowarefoundation/autoware.universe/pull/7808 https://github.com/autowarefoundation/autoware.universe/pull/7804), they are not being reviewed or merged because they impact TierIV's internal development, which shouldn't liimit what the Autoware project can do or not. This is what happens when you track the main branch in a project, breaking changes happen all the time. You can always target a tag to make sure the code does not change.

mitsudome-r, 2024-07-10:
@esteve I talked with @technolojin today. He said TIER IV will be responsible for changing the package names and namespace for packages in perception/sensing folder. Could you focus on the other packages for now?

He will post more detailed timeline about expected day to finish all the changes later.

mitsudome-r, 2024-07-10:
Also, another thing that @technolojin would like split the work into two PRs for packages without a library:
* one to change the namespace
* another to change the package name

This is because changing the namespace for non-library packages wouldn't be a breaking change because it would be pretty much self-contained in the package, whereas changing the package name would be a breaking change for all the launch files that refers to the package.

He said that he would like to make sure that all the non-breaking change is merged first, and then do some regression tests for the PRs with breaking changes all together in short period of time.

One of the drawbacks of splitting the package would be that it makes us harder to track the progress properly, but I think it can be solved by creating a sub-issue to just track sensing/perception with two checkboxes for each package. Something like the following (or it could be a spreadsheet):

- [ ] detection_by_tracker
  - [ ] autoware namespace added
  - [ ] autoware_prefix added to package name
- [ ] euclidean_cluster
  - [ ] autoware namespace added
  - [ ] autoware_prefix added to package name
...

I personally don't mind if that makes them easier to review the change and do the tests, but do you have any comments?

technolojin, 2024-07-10:
@esteve @mitsudome-r 

I would like to share perception team schedule for those works.

We keep work on namespace changes that are not breaking change. Also, the package name changes can be done within autoware_universe will be done from now (example: https://github.com/autowarefoundation/autoware.universe/pull/7892).

Current estimation for the breaking changes is as the following period.
**22-25th of July**

esteve, 2024-07-10:
@technolojin timeline sounds alright. I'd  prefer to make the changes both namespace and package name in the same PR, but I think it's a fair compromise if this helps getting the changes merged as soon as possible. Would it be possible to make the changes per package in each PR? It's easier to track what's being in the process and what's finished.

However, I wouldn't wait until the shared library that you mentioned earlier is complete, but just make the changes as soon as possible and then you can do any extra refactorings after the rename.

technolojin, 2024-07-11:
@esteve 
Thank you for your understanding. 

1. Separating PRs for namespace and package name
I see some packages that it is better to change the namespace and package name in the same time because both changes are breaking changes. I would like to respect decision of developers who will work on these.
Also, some packages may worked in the same time to avoid complex dependency and conflict. If the work is on multiple packages, it would be better to indicate which packages are the target.

2. Shared library, directory structure, location of header files
> https://github.com/autowarefoundation/autoware.universe/pull/7804#issuecomment-2205559422
> @technolojin I've submitted this PR as a follow-up to https://github.com/autowarefoundation/autoware.universe/pull/7642 to update the package name and move the headers, let me know if the changes are ok. Thanks.

"move the headers" is out of scope of this issue. 
I understand to refactoring packages in this chance may helpful for overall SW quality improvement. 
However, it is not mandatory and I want to discuss about the directory structure. Until we discuss about this topic, developers may not relocate the headers as mandatory. 

3. For object tracking packages
I mentioned as following.
> https://github.com/autowarefoundation/autoware.universe/pull/7808#pullrequestreview-2157746309
> moving headers
I am considering the association to be shared in the future. this association part is the same as multi_object_tracker, radar_object_tracker, and tracking_object_merger.
Let me do the re-organization later.

I checked differences of the DataAssociation classes on each packages, but there are many differences in variables and configurations. I think it is better to do after the work.

esteve, 2024-07-11:
> "move the headers" is out of scope of this issue.
I understand to refactoring packages in this chance may helpful for overall SW quality improvement.
However, it is not mandatory and I want to discuss about the directory structure. Until we discuss about this topic, developers may not relocate the headers as mandatory.

Headers need to be moved anyway, either to private location or to `include/autoware/object_merger` as part of the refactor task. If you want to keep the headers public, I can move them to `include/autoware/object_merger`, no problem, but they have to be moved no matter what (we've done this with plenty of other packages successfully, either public or private), so I took the opportunity to move them private because there are no downstream packages that depend on `object_merger`

technolojin, 2024-07-12:
> Headers need to be moved anyway, either to private location or to `include/autoware/object_merger` as part of the refactor task. If you want to keep the headers public, I can move them to [...]

Yes, we have to move header `include/autoware/object_merger` or private. 
As code owners of perception packages, readability is important to me. If you can get agreement about proper directory structure for private headers, It can be done in this work. But if it is not, then it need to be done separately.

esteve, 2024-07-12:
@technolojin

> Yes, we have to move header include/autoware/object_merger or private.

Moving the headers is in the scope of this task, directory structure is part of the rules we're implementing for all packages.

> For code owners of perception packages, readability is important.

Readibility is important for all packages, not just perception, so that's why we want all subsystems to follow the same rules, the perception subsystem is not special. Code owners can't override the rules we have established for all the Autoware project, the other subystems are following these rules, the perception subsystem is not different.

> If you can get agreement about proper directory structure for private headers, It can be done in this work. But if it is not, then it need to be done separately.

I don't understand this sentence, can you elaborate? What do you mean by getting agreement? The coding guidelines are clear about the directory structure of headers, and moving headers is part of this task, it's not out of scope. The headers structure must match either the public structure (`include/autoware/object_merger`) or private (`src` or `src/include`), see https://autowarefoundation.github.io/autoware-documentation/main/contributing/coding-guidelines/ros-nodes/directory-structure/#include-and-src

technolojin, 2024-07-16:
@esteve 

If my sentences were not clear to you, I am sorry for that.

> The headers structure must match either the public structure (include/autoware/object_merger) or private (src or src/include)
Yes, I agree in this. I am not trying to overwrite those rules. 

There are some packages already worked, and the headers are in (include/autoware/package_name) even those are not used outside of the package.
https://github.com/autowarefoundation/autoware.universe/tree/36a1d14a90d8f68900a7bc84295c958971fb9c05/planning/autoware_obstacle_cruise_planner/include/autoware/obstacle_cruise_planner
https://github.com/autowarefoundation/autoware.universe/tree/36a1d14a90d8f68900a7bc84295c958971fb9c05/planning/autoware_planning_validator
https://github.com/autowarefoundation/autoware.universe/tree/36a1d14a90d8f68900a7bc84295c958971fb9c05/localization/autoware_landmark_based_localizer/autoware_landmark_manager/include/autoware/landmark_manager

As like the other components, let the developers and the code owners decide whether they put the headers in the public structure or private. That is what the 'agreement' I want to say.

esteve, 2024-07-16:
@technolojin 

> Yes, I agree in this. I am not trying to overwrite those rules.

Glad we reached an agreement :slightly_smiling_face:

technolojin, 2024-07-17:
@esteve 
I understand necessity to move non-exporting headers ouside of `include` folder. 
I have done that work for relatively simple packages.
- https://github.com/autowarefoundation/autoware.universe/pull/7650/files#diff-08a8e2822cf0be2d964ef3246fcde71cc814c558b5eae37633de3e46d5f54670
- https://github.com/autowarefoundation/autoware.universe/pull/7646/files#diff-9539e4df0d0068e51bb181440a325b6576c68761d654bddb1f6606e266ce6ed9

and more.

In some complex packages, code owners need time to think its folder structure. The directory structure in `src` is up to developers, and need agreement of the code owners, who considers its readability. 
Therefore, I am suggesting separate the moving non-experting headers outside of `include` work from the current package renaming work.

technolojin, 2024-07-17:
@esteve 

> I'm moving headers back to public directory structure for the perception subsystem PRs I submitted

Thank you for your understanding. I will fix the header locations when overall rename work is done.
I reviewed your PRs.

esteve, 2024-07-17:
@technolojin 

> Thank you for your understanding. I will fix the header locations when overall rename work is done.

Can you elaborate why you need to do it in separate PRs? I'd like to understand your motivation for doing that. Code owners of other subsystems equally as complex as perception have moved headers and renamed packages at the  same time.

> I reviewed your PRs. Please see my comments.

I've addressed your feedback, can you review my PR again? Thanks.

technolojin, 2024-07-19:
@esteve 

> can you review my PRs again? 

There were missing part in the launchers. I fixed those.
https://github.com/autowarefoundation/autoware.universe/pull/7809
https://github.com/autowarefoundation/autoware.universe/pull/7804

> why you need to do it in separate PRs

1. https://autowarefoundation.github.io/autoware-documentation/main/contributing/pull-request-guidelines/#keep-a-pull-request-small-advisory-non-automated
This is one reason, and I got the given works can be separated. 
2. I wanted to reduce chance of conflict, or reduce impact of conflict if occurred. 
3. As I mentioned before, the package names are used in other repos. But this is TIER IV business.

esteve, 2024-07-19:
@technolojin 

> There were missing part in the launchers. I fixed those.
> https://github.com/autowarefoundation/autoware.universe/pull/7809
> https://github.com/autowarefoundation/autoware.universe/pull/7804

Thanks!

a-maumau, 2024-07-29:
Please be informed that I will start adding the `autoware_` prefix to some localization/mapping packages.

youtalk, 2024-09-03:
@esteve Is it possible to prioritize renaming the packages under `autoware.universe/common`? In relation to https://github.com/autowarefoundation/autoware/issues/5077, we would like to complete the `common` packages first.

esteve, 2024-09-03:
@youtalk yes, makes total sense to prioritize `autoware.universe/common`, I'll refocus on the remaining common packages.

xmfcx, 2024-11-18:
@esteve recently I've came across this file:
https://github.com/autowarefoundation/autoware/blob/0dd08838d0ebc154bf698ee1b8d157a456eb044f/docker/Dockerfile

It is referencing some packages by the names, when prefixing, you might need to create a PR against [autoware](https://github.com/autowarefoundation/autoware) repository as well.

It has recently caused an issue here:
- https://github.com/autowarefoundation/autoware/pull/5451

esteve, 2024-11-18:
@xmfcx thanks, I hadn't seen that file.