autowarefoundation/autoware #4574: nebula_hw_interfaces fails to build

isouf, 2024-03-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The `nebula_hw_interfaces` package fails to build after the latest commit in `nebula` https://github.com/tier4/nebula/commit/cb3c56a1c81fe567f8fa44c8f55b5d354a8c0d2f

### Expected behavior

`nebula_hw_interfaces`  builds without errors.

### Actual behavior

[image]

### Steps to reproduce

1. Build Docker image
2. Launch Docker container
3. Try to build the workspace using `colcon`. 

### Versions

- OS: Ubuntu 22.04
- Autoware: Lastest `main` branch

### Possible causes

The latest `nebular` [commit](https://github.com/tier4/nebula/commit/cb3c56a1c81fe567f8fa44c8f55b5d354a8c0d2f) results in this error. 

I am able to build the workspace if I revert `nebula` to the previous commit https://github.com/tier4/nebula/commit/18bb03a70cd0f40d6be747ff6ddd88193b6df864.

xuyuan-han, 2024-03-28:
I also met with this problem.

SakodaShintaro, 2024-03-29:
This problem may be related to https://github.com/MapIV/transport_drivers/pull/7.
I think it's resolved now that it's been merged.

isouf, 2024-03-29:
I can confirm that the latest PR in https://github.com/MapIV/transport_drivers/pull/7 has resolved the `nebula_hw_interfaces` build errors thus closing this issue. 
[image]

closed 2024-03-29 by isouf