autowarefoundation/autoware #4593: Implement Remaining Mission Distance and ETA Functionality and Visualization

ahmeddesokyebrahim, 2024-04-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The issue entails integrating functionality to provide and visualize the remaining mission distance and estimated time of arrival (ETA) within Autoware.

### Purpose

To enhance Autoware's capabilities by providing users with real-time information regarding the remaining distance to their destination and the estimated time needed to reach it.

As the remaining mission distance and estimated time of arrival (ETA) is currently not provided or visualized in the Autoware system., this feature is crucial for enhancing the user experience by providing real-time information about the progress of the mission and the estimated time left.

### Possible approaches

1. Utilize existing Autoware modules to calculate remaining mission distance and ETA.
2. Implement a new module specifically dedicated to computing and visualizing mission-related information.
3. Integrate with external APIs or libraries for more accurate calculations.

### Definition of done

- [ ] Remaining mission distance is accurately calculated.
- [ ] ETA is calculated based on current vehicle status and route information.
- [ ] Visualization component is implemented to display remaining distance and ETA.
- [ ] Functionality is thoroughly tested and validated for accuracy.

xmfcx, 2024-04-19:
@ahmeddesokyebrahim could you open up the PR for this task?

ahmeddesokyebrahim, 2024-04-23:
The following PRs are ready for review :red_circle: **_merge sequence matters_** :red_circle: 
- https://github.com/autowarefoundation/autoware_msgs/pull/85
- https://github.com/autowarefoundation/autoware.universe/pull/6855
- https://github.com/autowarefoundation/autoware_launch/pull/963

xmfcx, 2024-04-23:
Could you add the message to https://github.com/autowarefoundation/autoware_internal_msgs instead?

https://github.com/autowarefoundation/autoware_msgs is for more mandatory interfaces of the Autoware.

Also I think time estimation can be a bit tricky, maybe we can only start with the remaining distance metric.

ahmeddesokyebrahim, 2024-04-24:
@xmfcx :
Thanks for your review comment. I have updated my [comment](https://github.com/autowarefoundation/autoware/issues/4593#issuecomment-2071388798) above.

I am thinking of keeping the time information in the msg definition for reporting and debugging purposes, similar to suggestion from @mitsudome-r -san in the meeting, so they will stay in the publisher and we may handle some corner cases when the vehicle stops and so on.
But I plan to remove these information for now from the visualization display and keep only the distance metric as you clarified.

Let me know what do you think about that.

xmfcx, 2024-05-13:
Finished as all related PRs are merged and the functionality is incorporated into the Autoware.

- https://github.com/autowarefoundation/autoware_internal_msgs/pull/10
- https://github.com/autowarefoundation/autoware.universe/pull/6855
- https://github.com/autowarefoundation/autoware_launch/pull/963

closed 2024-05-13 by xmfcx