autowarefoundation/autoware #4635: Implementing BEVDet in Autoware

cyn-liu, 2024-04-18:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

BEVDet is a BEV perception algorithm based on panoramic cameras.  It unifies multi-view images into the perspective of BEV for 3D object detection task. It is different from the current 3D perception feature of Autoware.
[BEVDet code repos](https://github.com/HuangJunJie2017/BEVDet)

### Purpose

Integrating BEVDet into Autoware for 3D object detection based on multi-view images, this task related to Sensing& Perception task. 

### Possible approaches

BEVDet is a 3D object detection model trained on NuScenes dataset using 6 surround view camera images. The 6 cameras form a 360 degree field of view with overlapping fields of view.  When mapping from 2D to 3D, some parameters are required, including camera  intrinsic parameters and extrinsic parameters between each camera and ego.
Integrating BEVDet into Autoware involves the placement of 6 cameras and calibration. Convert BEVDet model into ONNX format for deployment in Autoware.

### Definition of done

- [ ] The placement of 6 cameras and calibration
- [ ] Convert BEVDet model into ONNX format
- [ ] BEVDet model deployment  in Autoware

liuXinGangChina, 2024-04-18:
Great，maybe you can make a todo task list first and see what others can take part in

cyn-liu, 2024-05-21:
We refer to this [project](https://github.com/linClubs/BEVDet-ROS-TensorRT)  and successfully ran it on our own machine.

Next, we will modify ROS1 node to ROS2 node based on this project, then, we will use [TIER IV's dataset](https://github.com/autowarefoundation/autoware/issues/4609) for testing, and we hope that this dataset can provide ROS2 bag format.

cyn-liu, 2024-05-29:
Environment: 
CUDA11.3.1
cudnn- linux-x86_64-8.8.1.3_cuda11
TensorRT-8.5.1.7.Linux.x86_64-gnu

liuXinGangChina, 2024-06-18:
Maybe try with AWSIM data

liuXinGangChina, 2024-07-02:
list the cuda env here

cyn-liu, 2024-07-03:
Using the BEVDet model to infer the TIER4 dataset, it was found that the model had poor generalization performance on the TIER4 dataset.
Visualization results on TIER4 data：
(1)
[image]

[image]

(2)
[image]

[image]

Visualization results on NuScenes data：
[image]
[image]

liuXinGangChina, 2024-07-04:
Looks like the original pre-trian（based on nuScenes dataset） model‘s generalization on tire4 dataset is not as well as we expected. Obstacles's direction is almost right but the depth of them ge

we plan to close this task once we have the node tested. And creat a new task of "retrain the model" to see whether the new model’s performance on tire4 dataset increase.

cyn-liu, 2024-07-11:
> Our plan of integrate the BEVDet ROS2 node into Autoware:
> 
> 1. define a `bevdet_node` in Autoware perception module
> 2. organize the 3D boxes results into `autoware_perception_msgs::msg::DetectedObjects`  type
> 3. input the output result of `bevdet_node` into the `object_merger` node and fuse it with the detection results of other models

Considering that running the BEV 3D detection algorithm based on multi-cameras and the Lidar based 3D detection algorithm simultaneously is too heavy a load. we have decided not to merge the results of BEVDet with the output results of Lidar, but to create a new `perception_mode`, when   `perception_mode = camera`,  launch `bevdet_node`.

cyn-liu, 2024-07-22:
@xmfcx The PR related this  issue has been successfully tested in the newer Autoware  docker image.
The environment information of this image:
```
CUDA==12.3
libnvinfer==8.6.1.6
```
Note:  Outside in docker, I must upgrade to my nvidia GPU driver version to ensure that this driver supports a maximum CUDA version >= 12.3.
[image]

RahulSundarMCW, 2025-05-13:
Submitting https://github.com/autowarefoundation/autoware_universe/pull/10441 which is a continuation of https://github.com/autowarefoundation/autoware_universe/pull/7956.

Also raised supporting PR's:

- https://github.com/autowarefoundation/bevdet_vendor/pull/1
- [feat: add tensorrt_bevdet artifacts](https://github.com/autowarefoundation/autoware/pull/6138) (model url and checksum will be included in artifacts after the reply for our query on [discord](https://discord.com/channels/953808765935816715/953846504865533983/1371754007265017950)).

Reviewers can test the functionality of this PR with respect to [nuscenes dataset](https://github.com/Owen-Liuyuxuan/ros2_dataset_bridge/blob/main/docs/nusc_readme.md) using this [test branch](https://github.com/autowarefoundation/autoware_universe/compare/main...RahulSundarMCW:autoware_universe:testing_nuscenes).

xmfcx, 2025-07-07:
Listing all relevant PRs related to this issue:
- https://github.com/autowarefoundation/bevdet_vendor/pull/1
- https://github.com/autowarefoundation/autoware/pull/5456
- https://github.com/autowarefoundation/autoware/pull/6138
- https://github.com/autowarefoundation/autoware_universe/pull/10441