autowarefoundation/autoware #4798: Document and prune the docker packages

xmfcx, 2024-05-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

https://github.com/autowarefoundation/autoware/packages

Packages:
- autoware-universe
- autoware-openadk
- autoware
- autoware-openadk-modular
- openadkit

@oguzkaganozt @youtalk could you please remove (if you don't have permissions, I can remove them too) unused packages from here?

And for the remaining, could you add a documentation to a suitable place where people can see and understand what to use when?

Right now this page is a mess and very confusing for newcomers.

### Purpose

-

### Possible approaches

-

### Definition of done

-

xmfcx, 2024-05-30:
Also we can't keep renaming these packages as people rely on these to be stable. Let's discuss what to keep in the open ad kit wg and move forward that way.

Also when these are renamed, the community should also be alerted through github, discord announcements.

youtalk, 2024-06-10:
I think these packages are no longer needed. @oguzkaganozt What do you think?
- autoware-universe
- autoware-openadk
- autoware-openadk-modular

xmfcx, 2024-07-02:
https://github.com/orgs/autowarefoundation/discussions/4555#discussioncomment-9933131

[image]

I've set the visibility of all other [packages](https://github.com/orgs/autowarefoundation/packages?repo_name=autoware) than `autoware` and `openadkit` to private ✅

closed 2024-07-02 by xmfcx