autowarefoundation/autoware #4836: When the display scale is set to 200% the Rviz window always crashes

kaiyama12345679, 2024-06-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

the display setting of My PC(Ubuntu22.04) is 4K, 200% scale.
when I launch autoware, the Rviz window always crashes.
However, setting the display to FHD and the scale to 100% resolves the error

### Expected behavior

Rviz window launched correctly

### Actual behavior

The Rviz window launches and then crashes immediately.

### Steps to reproduce

1. set display to 4K resolution and scale to 200%
1. launch autoware

### Versions

- OS: ubuntu22.04

YuqiHuai, 2024-06-13:
I also encountered this same issue. I am now working with very small text.

xmfcx, 2025-02-17:
Here is my setup:

```console
mfc@bunny:~/projects/autoware$ xrandr
Screen 0: minimum 8 x 8, current 6400 x 2160, maximum 32767 x 32767
DP-1 connected primary 2560x1440+0+0 (normal left inverted right x axis y axis) 381mm x 214mm
   2560x1440     60.00*+ 240.00  
HDMI-0 connected 3840x2160+2560+0 (normal left inverted right x axis y axis) 697mm x 392mm
   3840x2160     60.00*+  59.94    50.00    30.00    29.97    25.00    23.98 
```

Normally I use `xrandr --dpi 96` and it works. To reproduce this issue, I've tested:

```
xrandr --dpi 192
xrandr --dpi 288
xrandr --dpi 384
```

@kaiyama12345679 @YuqiHuai could you test this on your setup again if it still occurs? I couldn't replicate it on my setup (I use i3-wm so it might be a reason but it's dpi issue after all, so it's better if you could confirm.)

## DPI 96 (default)

[image]

## DPI 192

[image]

## DPI 288

[image]

## DPI 384

[image]

YuqiHuai, 2025-02-17:
@xmfcx I was just about to order some new monitors and this issue disappeared on my end. I am not sure what was happening previously.

xmfcx, 2025-02-17:
@YuqiHuai thanks for confirming. Within the past year, we have made a lot of improvements on the rviz plugins & themes side so it might be related to any of those changes 🤷‍♂️

Closing as solved, please reopen if it occurs again.

closed 2025-02-17 by xmfcx