autowarefoundation/autoware #4896: Missing environment variables in devel Docker container

amadeuszsz, 2024-06-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

After running development container, environment variables set via ansible script doesn't exist.

### Expected behavior

All the environment variables set via ansible script are reflected for user while running Docker image with `--devel` flag.

### Actual behavior

`/home/${USER}/.bashrc` doesn't contain environment variables. The missing variables are:
```bash
export PATH="$PATH:/root/.local/bin"
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export CCACHE_DIR="/var/tmp/ccache"
export CC="/usr/lib/ccache/gcc"
export CXX="/usr/lib/ccache/g++"
export PATH="/usr/local/cuda/bin:$PATH"
export LD_LIBRARY_PATH="/usr/local/cuda/lib64:$LD_LIBRARY_PATH"
```

It causes multiple issues, e.g. running ROS 2 daemon with `rmw_fastrtps_cpp` DDS.

### Steps to reproduce

```bash
./docker/run.sh --devel
cat $HOME/.bashrc
sudo su
cat $HOME/.bashrc
```

### Possible causes

Command `exec /usr/sbin/gosu "$USER_NAME" "$@"` from `docker/etc/ros_entrypoint.sh` run bash as user with default `.bashrc` file. We can fix this two ways:
* Handle `/etc/skel/.bashrc` in ansible scripts. That means push all environment variables to two `.bashrc` files at the same time. In that case we will need to change `export PATH="$PATH:/root/.local/bin"` to `export PATH="$PATH:\$HOME/.local/bin"`. I'm not sure if `\` will keep `$` in scripts as a special sign here. It has to be validated.
* Add variables to `$HOME/.bashrc` via `ros_entrypoint.sh`. Copying `/root/.bashrc` is also possible if we modify `export PATH="$PATH:/root/.local/bin"` as described previously.

oguzkaganozt, 2024-07-29:
Hi @amadeuszsz could you resolve the failed checks on the [PR ](https://github.com/autowarefoundation/autoware/pull/4923)?

amadeuszsz, 2024-07-30:
> Hi @amadeuszsz could you resolve the failed checks on the [PR ](https://github.com/autowarefoundation/autoware/pull/4923)?

Hi @oguzkaganozt 
Done!

closed 2024-08-01 by amadeuszsz