autowarefoundation/autoware #4911: Separation of `autoware_common` to `autoware_cmake`, `autoware_utils` and `autoware_lanelet2_extension`

youtalk, 2024-06-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

This issue manages the task of splitting and migrating from `autoware_common` to `autoware_cmake`, `autoware_utils`, and `autoware_lanelet2_extension`.

### Purpose

https://github.com/orgs/autowarefoundation/discussions/4671

### Possible approaches

- [ ] https://github.com/autowarefoundation/autoware.universe/pull/7640
- [ ] https://github.com/autowarefoundation/autoware/pull/4909
- [ ] Archive https://github.com/autowarefoundation/autoware_common

### Definition of done

The above tasks will be checked

xmfcx, 2024-07-01:
https://github.com/autowarefoundation/autoware_common is archived.

But it contains `lanelet2_extension` and `lanelet2_extension_python` packages and still is referenced by `autoware.repos` file.

What steps should be done to remove it from the `autoware.repos` file?

youtalk, 2024-07-02:
@xmfcx Some of the downstream repositories still need to use the old `autoware_common`.
https://github.com/tier4/scenario_simulator_v2/pull/1300#discussion_r1652188056
That's why I cannot remove the `autoware_common` from `autoware.repos` for now.

SakodaShintaro, 2024-12-18:
Since https://github.com/tier4/scenario_simulator_v2/pull/1300 was merged, is this issue resolved?

youtalk, 2024-12-18:
When https://github.com/autowarefoundation/autoware/pull/5578 is merged, this issue will be automatically closed.