autowarefoundation/autoware #4991: Add lightweight docker build option

VRichardJP, 2024-07-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

[Recent refactor](https://github.com/autowarefoundation/autoware/commit/da22bdc9215edf188cdbf8ef34637b43956d9049)  of `./docker/build.sh` has removed the `--no-prebuilt` option, which could be used to create an image without any source code or build artifacts.

I really liked this option, because it both saved a lot of time (1 hour compilation...) and a lot of space on my system. On top of that, I have no use for the any source code or built artifact in `/autoware` since I share my own "autoware" directory with the container (built artifacts are on host).

Could this option be re-implemented?

### Purpose

Lighter docker image, faster build

### Possible approaches

Add alternative branch to the current `prebuilt` image, in which all build dependencies would be installed but autoware would not be built.

### Definition of done

Can skip building autoware when building the docker image. Yet the image could be used with mounted `src/` directory to build autoware with no extra work.

VRichardJP, 2024-07-17:
Example (hacky) implementation from b57e6e1fe846b56140f73abaa2a9901d9cec5602:

```diff
diff --git a/docker/Dockerfile b/docker/Dockerfile
index a1792c6e..9801e4f1 100644
--- a/docker/Dockerfile
+++ b/docker/Dockerfile
@@ -201,3 +201,63 @@ COPY docker/etc/ros_entrypoint.sh /ros_entrypoint.sh
 RUN chmod +x /ros_entrypoint.sh
 ENTRYPOINT ["/ros_entrypoint.sh"]
 CMD ["bash"]
+
+# Autoware environment with no source code nor build artifact 
+FROM base as no-prebuilt
+SHELL ["/bin/bash", "-o", "pipefail", "-c"]
+ARG ROS_DISTRO
+ARG LIB_DIR
+ARG SETUP_ARGS
+
+# Install rosdep dependencies
+COPY --from=rosdep-depend /rosdep-core-depend-packages.txt /tmp/rosdep-core-depend-packages.txt
+COPY --from=rosdep-depend /rosdep-universe-depend-packages.txt /tmp/rosdep-universe-depend-packages.txt
+# hadolint ignore=SC2002
+RUN --mount=type=cache,target=/var/cache/apt,sharing=locked \
+  --mount=type=cache,target=/var/lib/apt,sharing=locked \
+  apt-get update \
+  && cat /tmp/rosdep-core-depend-packages.txt | xargs apt-get install -y --no-install-recommends \
+  && apt-get autoremove -y && rm -rf "$HOME"/.cache
+# hadolint ignore=SC2002
+RUN --mount=type=ssh \
+  apt-get update \
+  && cat /tmp/rosdep-universe-depend-packages.txt | xargs apt-get install -y --no-install-recommends \
+  && apt-get autoremove -y && apt-get clean -y && rm -rf /var/lib/apt/lists/* "$HOME"/.cache
+
+# Install development tools and artifacts
+RUN --mount=type=ssh \
+  --mount=type=cache,target=/var/cache/apt,sharing=locked \
+  --mount=type=cache,target=/var/lib/apt,sharing=locked \
+  ./setup-dev-env.sh -y --module dev-tools --download-artifacts openadkit \
+  && pip uninstall -y ansible ansible-core \
+  && apt-get autoremove -y && rm -rf "$HOME"/.cache
+
+# Set up runtime environment and artifacts
+COPY --from=rosdep-depend /rosdep-exec-depend-packages.txt /tmp/rosdep-exec-depend-packages.txt
+# hadolint ignore=SC2002
+RUN --mount=type=ssh \
+  --mount=type=cache,target=/var/cache/apt,sharing=locked \
+  --mount=type=cache,target=/var/lib/apt,sharing=locked \
+  ./setup-dev-env.sh -y --module all ${SETUP_ARGS} --download-artifacts --no-cuda-drivers --runtime openadkit \
+  && pip uninstall -y ansible ansible-core \
+  && apt-get update \
+  && cat /tmp/rosdep-exec-depend-packages.txt | xargs apt-get install -y --no-install-recommends \
+  && apt-get autoremove -y && rm -rf "$HOME"/.cache \
+  && find /usr/lib/$LIB_DIR-linux-gnu -name "*.a" -type f -delete \
+  && find / -name "*.o" -type f -delete \
+  && find / -name "*.h" -type f -delete \
+  && find / -name "*.hpp" -type f -delete \
+  && rm -rf /autoware/ansible \
+    /root/.local/pipx /opt/ros/"$ROS_DISTRO"/include /etc/apt/sources.list.d/cuda*.list \
+    /etc/apt/sources.list.d/docker.list /etc/apt/sources.list.d/nvidia-docker.list \
+    /usr/include /usr/share/doc /usr/lib/gcc /usr/lib/jvm /usr/lib/llvm*
+
+# Copy bash aliases
+COPY docker/etc/.bash_aliases /root/.bash_aliases
+RUN echo "source /autoware/install/setup.bash" > /etc/bash.bashrc
+
+# Create entrypoint
+COPY docker/etc/ros_entrypoint.sh /ros_entrypoint.sh
+RUN chmod +x /ros_entrypoint.sh
+ENTRYPOINT ["/ros_entrypoint.sh"]
[... 75 lines cut ...]
 
 target "base" {
   inherits = ["docker-metadata-action-base"]
@@ -38,3 +39,10 @@ target "runtime" {
   dockerfile = "docker/Dockerfile"
   target = "runtime"
 }
+
+
+target "no-prebuilt" {
+  inherits = ["docker-metadata-action-no-prebuilt"]
+  dockerfile = "docker/Dockerfile"
+  target = "no-prebuilt"
+}

```

The `no-prebuilt` image was built in 38min on my machine (vs ~2h before). Image size is roughly half the original `ghcr.io/autowarefoundation/autoware-universe:latest-devel-cuda`:

```bash
docker images
REPOSITORY                                                                                    TAG                           IMAGE ID       CREATED          SIZE
ghrc.io/autowarefoundation/autoware                                                           latest-no-prebuilt-cuda       64eaa27d0057   27 minutes ago   10.8GB
ghcr.io/autowarefoundation/autoware                                                           latest-devel-cuda             20e6657cb540   2 weeks ago      18.6GB
```

xmfcx, 2024-07-18:
Related discussion:
- [Proposal to change Docker image tag naming conventions #4995](https://github.com/orgs/autowarefoundation/discussions/4995)

youtalk, 2024-09-06:
I hope this issue is already resolved by https://github.com/autowarefoundation/autoware/pull/5117.

closed 2024-09-06 by youtalk