autowarefoundation/autoware #5075: Separate the .repos in the sample packages and split them into a separate container

youtalk, 2024-08-13:
https://github.com/autowarefoundation/autoware/blob/main/autoware.repos currently includes many packages meant for sample code, such as `tamagawa_imu_driver` and `sample_sensor_kit_launch`. 
First, manage these packages with a separate .repos file.

Next, create a separate container image that includes only these packages, which will help reduce the size of the main container and improve availability.

xmfcx, 2025-02-05:
- `sample_sensor_kit_launch`
- `sample_vehicle_launch`
- `autoware_individual_params`

These are required for running the [tutorials](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/). If these are separated, we should update the documentation with steps to pull these too.

But my concern is, .repos files are already very fragmented. Developers now have to pull more and more .repos files and I'd prefer if we could maintain a comprehensive .repos file for developers.

I agree with moving `tamagawa_imu_driver` and `pacmod_interface` out as vendor specific packages. But not for the repositories that contain things required for the demos.

oguzkaganozt, 2025-10-23:
I think we can close this issue for now. What do you think ? @youtalk @mitsudome-r @xmfcx