autowarefoundation/autoware #5076: Build only RViz plugin packages and split them into a separate container

youtalk, 2024-08-13:
https://github.com/autowarefoundation/autoware.universe/tree/main/common contains a lot of source code related to RViz plugins. Additionally, when launching the main Autoware launch file, RViz is automatically started as well. 
This results in the installation of many RViz dependency packages, increasing the container image size. It also causes inconvenience when you want to run RViz as a separate process.

To address this, add a build stage that extracts and builds only these RViz plugins, and create a separate container image to run RViz using those plugins.

esteve, 2024-11-20:
For this, I'd move the plug-ins to a separate folder so I can pass the `--paths` option to `colcon` and simplify the process (thanks @xmfcx for the idea)

esteve, 2024-11-20:
I've submitted https://github.com/autowarefoundation/autoware.universe/pull/9417 as a draft to move the `rviz` plugins. It's still a draft because I'm waiting for https://github.com/autowarefoundation/autoware.universe/pull/9305 https://github.com/autowarefoundation/autoware.universe/pull/9303 and https://github.com/autowarefoundation/autoware.universe/pull/9302 to be merged. I will rebase https://github.com/autowarefoundation/autoware.universe/pull/9417 after that.

esteve, 2025-03-10:
Closing since all the related PRs have been merged.

closed 2025-03-10 by esteve

youtalk, 2025-03-10:
Great!
https://github.com/autowarefoundation/autoware.universe/tree/main/visualization